nextcloud/nextcloudpi #1207: Requirements for a new documentation platform

Daniel-Krebs, 2020-11-07:
Continuing the discussion from the ncp wiki team telegram group here.

Looking at the [current documentation](https://docs.nextcloudpi.com/) there is a certain interest, to evaluate other platforms for documentation.

Some requirements from the telegram group:

- better formatting for code / commands, currently commands like "$ docker ps -a" are handled with line breaks or with quotes like “sudo reboot now" which is hard to read and makes it difficult to highlight console output. See also [this issue](https://github.com/nextcloud/nextcloudpi/issues/1206)

- multiple language support / translations

- multiple users and/or groups (probably with different rights) 
 
personally I would add:

- good navigation through different entries (sidebar, topbar or menu etc.)

So, lets discuss :)

ovpc, 2020-11-07:
What are you (or anyone else) suggesting, to do, to meet proposed requirements? 
Apart from the Theme and code formatting, Wordpress already meets all mentioned requirements, it is why it was chosen.  (we are looking for assistance from creator of Buddypress, to resolve the [coding issue](https://github.com/nextcloud/nextcloudpi/issues/1206))
I can only think of one platform that might exceed WP's versatility: Drupal. 
Unfortunately not my cup of tea. I'm an old WP dog, and dont learn new tricks that easily anymore, nor do I aspire to ;-)
I am glad to see new interest in ncp documentation, and willingness to contribute, anything that advances NCP's cause, making using NC easier to set up and maintain. 

But if someone wants to port NCP's docs to another better suited platform, they are familiar with, set it up and manage it, that's very fine by me.

sunjam, 2020-11-07:
I think the coding issue is totally resolvable.

Aside from the current platform, what makes the most sense afaik is to use the actual help forum.  We could create a documentation section on the help forum with write access for private group members of NextcloudPi and make all posts wiki editable by default so they are easy for us to maintain.  Give all over users and guests read access, or just the abillity to comment.

Main thing I like about this possibility is it integrates that much deeper into the Nextcloud community, plus greatly simplifies that amount of work needed to maintain documentation.  Awaiting approval on a post is not my favorite, but people could always write up docs on the forum and then have a moderator move them into NextcloudPi doc group.

ovpc, 2020-11-07:
`it integrates that much deeper into the Nextcloud community, plus greatly simplifies that amount of work needed to maintain documentation`
good point, agree, we'd only lose the (manageability of) translations, although I must say: nothing much is and has been going on in that aspect. Most users manage in English apparently, so it seems.

Daniel-Krebs, 2020-11-08:
Just to give you some impressions of what I like looking at other CMS for purpose of documentation:

An expandable, nested navigation bar
[image]

Predefined formatting for things like warning, info etc. 
[image]
[image]

But I don't want to solve a problem that doesnt exist, so if wordpress is the way to go I'm finde with that :+1:  

Using the nc forum for deeper integration into the community seems reasonable. Personally I never got used to the structure, like using a kind of a tag system instead of a sub forums etc.

theCalcaholic, 2020-11-13:
All of that is of cause solvable in Wordpress as well. I'd rather adjust the current documentation than go through the effort of setting up a new solution and migrating all the content and users.

I think it's better to collect a list of desired features and add them to the wordpress page.

nachoparker, 2020-11-18:
actually our first solution was readthedocs, but it didn't have good language support

Daniel-Krebs, 2020-11-22:
I also like Readthedocs. Concerning language support: looking at the current docs non-english entries are so rare that i would question if that is really a needed feature. Just having the "welcome page" in your language but other entries in English is probably no real benefit for non-english readers

ovpc, 2020-11-22:
Agree with languages part, there is hardly any activity there, nor has there been, so not expecting any, everyone seems fine with English.
I did not like Readthedocs tho, there was no easy way to manage or restructure pages, it take its content straight from Github wiki pages. So not very easy to contribute without good knowledge of markdown.
I am not going to spend much time on it anyway, but anyone who would like to take the job on is most welcome too ;-)

michuvon, 2021-10-15:
@nachoparker could you reopen this issue again? I will not be able to act immediately, but still like to start the discussion again.

ovpc, 2021-10-16:
[image]
current theme from http://buddydev.com/
At the time was selected because only one supporting  de translations plugin [WP MultiLang](https://wordpress.org/plugins/wp-multilang/)

ovpc, 2021-10-16:
MultiLang was main, but not only reason to move to WP. Restructuring, editing and organizing existing pages was beyond my skills in Github docs. 
True: few pages have translations, and contributions have been far from overwhelming in numbers \o/ so I agree, it is valid question. 
If we decide to disable the MultiLang plugin, the website will look very messy tho, I think. 
It also will be time consuming to reorganize. 
It would not be fair to contributors, to simply remove translated pages, imho. Maybe they could be stored them in a sub-folder/page way, for them to still show in search results.

michuvon, 2021-10-17:
I see.
I would under no circumstances abandon WP. I think it's a great basis for NCP docs.

So I suggest we are looking around once more if there are other (free) themes with [WP-MultiLang](https://wordpress.org/plugins/wp-multilang/) compatability.
If not we should study more about the current theme BuddyDocs so that we can make the most out of it.

But honestly, I feel BuddyDocs can easely compete in uglyness with my profile picture.

ovpc, 2021-10-17:
Easy discovery? Yeah that would be great, don't know of any.
To find BuddyDocs I set up a WP on a test server, import content, install themes, enabled WPlang, trial and error.
But then a lot has changed since, so we could give it another shot, if there is enough support and willingness to invest time.

ZendaiOwl, 2021-10-20:
For my part I agree with clean it up and try some themes or plugins is the way to go, I think it could be too much work to move to another platform now when it's possible to work with what we have and explore some options there :)

closed 2021-10-31 by ovpc