nextcloud/nextcloudpi #1208: Fresh new installation of NextCloud on a fresh new Raspberry OS installation is completely broken

kohaerenzstifter, 2020-11-13:
I have a completely fresh new installation of Raspberry OS (image "2020-08-20-raspios-buster-armhf-lite.img") on a "Pi 2 Model B v1.1" and tried to install NextCloud with the curl method:

curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash

It does not work. I do not get past activation. A summary of my issues can be found [here](https://help.nextcloud.com/t/nextcloudplus-not-yet-initialized-trying-again-in-a-few-seconds/31019/58?u=martin_knappe) and [here](https://help.nextcloud.com/t/nextcloudplus-not-yet-initialized-trying-again-in-a-few-seconds/31019/60?u=martin_knappe)

ovpc, 2020-11-13:
In order to investigate, thank you for sharing  output of 

- sudo ncp-report
- sudo cat /var/log/syslog
- sudo dmesg

Please use a pastebin service, output can be very long 

Also
- run fsck on your drive(s) if any errors
- swap sd-card, cables, powersource. (90% problems come from corrupt cards, bad drives, broken cables and insufficient power-sources.)
- Pi2 and older are notoriously slow, especially so  at running mysql, and init should create a clean db.
- If serious about running NC in your home or office, I can recommend a NUC or a (refurbished) desktop computer with minimal i3 and 8Gb Ram.

ovpc, 2020-11-13:
moved to https://help.nextcloud.com/t/cannot-activate-nextcloud/97989/2

Grocel, 2020-11-20:
I also had quite a few problems during install on brand new hardware (Raspberry Pi 4 (4GB RAM), micro SD card, USB stick, 3.1A USB C power supply) which I was able to fix by myself. The installation process of NCP is quite rough compared to pi-hole for instance. It is surely still easier than a manual install, but in my opinion it could use a little bit more polishment in general.

ovpc, 2020-11-20:
I'm closing this, please use [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33) for support and or [Telegram](https://t.me/NextCloudPi) group to talk about ncp.

closed 2020-11-20 by ovpc