nextcloud/nextcloudpi #1210: Nextcloud update deleted data!

shaan7, 2020-11-23:
Hi,

Looks like the rollback process has a bug, it ended up deleting the original nextcloud dir (including the data dir!):

```
Running nc-update-nextcloud
Current   Nextcloud version 18.0.4.2
Available Nextcloud version 18.0.11
Download Nextcloud 18.0.11...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20201123_1606135870.tar.gz generated
Maintenance mode disabled
Install Nextcloud 18.0.11...
mv: cannot move 'nextcloud' to 'nextcloud-old': Directory not empty
Abort
Simple roll back...
mv: cannot stat '/data/nextcloud-old': No such file or directory
Clean up...
Done. Press any key...
```

While it was doing `Simple roll back...` for a long time. While I waited, I realized that maybe nextcloud-old is just some leftover because it was quite small (~200M). So I deleted it.
But `Simple roll back...` was still going on so I got suspicious. I checked htop and it shows `rm -rf /data/nextcloud` running! By the time I realized it, pretty much everything was gone (and my disk usage confirms that it is suddenly empty).

I have a old backup of the disk which I will probably restore and then sync back the newer files, but I am reporting this in hopes that it can be fixed before affecting someone else.

My guess is that the `Simple roll back...` assumes that `nextcloud` -> `nextcloud-old` rename was successful, so it tries to delete `nextcloud` thinking that its the "new" one which failed (to install etc), but that is not true.

nachoparker, 2020-11-24:
Hi, thanks for reporting

> So I deleted it

"Simple rollback" all it does is move your original `nextcloud-old` back to it's original location `nextcloud`. Don't mess with this stuff manually or you risk deleting your data, as you did. The updater tries really hard not to delete anything whatsoever until the process is over.

Now the question would be, why did you have `nextcloud-old` leftover in the first place. The updater doesn't even start if the folder is there, so I think you must have started two upgrades at the same time.

closed 2020-11-24 by nachoparker

shaan7, 2020-11-24:
@nachoparker yes, I should probably not have removed the `nextcloud-old` dir myself. But that is not related to the script deleting the data. *It was happening already* when I deleted the `nextcloud-old` dir (I saw `rm -rf /data/nextcloud` in htop before I deleted `nextcloud-old`), I just realized it late.
Just imagine what happens if this fails: https://github.com/nextcloud/nextcloudpi/blob/21a791db986a864d6e34d9471fbceac466c36f03/bin/ncp-update-nc#L123

L123 fails, the trap is activated (the "real" data is still in `nextcloud`) and then L114 (`rm -rf "$BASEDIR"/nextcloud`) starts deleting it.

nachoparker, 2020-11-25:
As I said, we check for `nextcloud-old` very early

https://github.com/nextcloud/nextcloudpi/blob/21a791db986a864d6e34d9471fbceac466c36f03/bin/ncp-update-nc#L29

I think we can assume that `mv -T nextcloud nextcloud-old` is always going to succeed, it's just renaming a folder without changing the rest of the path.

The script definitely does not delete the data that I can see.

In your case, you should be able to restore the backup that is performed automatically at the beginning of the process.

shaan7, 2020-11-25:
Ah, thank you for linking me that line, I seem to have missed it. That should do the trick.
Pardon the noise. I just wanted to help avoiding this happening to someone else in the future :)