nextcloud/nextcloudpi #1211: cp: cannot stat 'nextcloud-old/apps/previewgenerator'

davidoe, 2020-11-30:
Updating to 20.0.2 stops because 'cp: cannot stat 'nextcloud-old/apps/previewgenerator': No such file or directory'

Full output (of https://XXX:4443/?app=nc-update-nextcloud):
```

[ nc-update-nextcloud ] (Mon Nov 30 17:42:02 GMT 2020)
Current Nextcloud version 19.0.4.2
Available Nextcloud version 20.0.2
Download Nextcloud 20.0.2...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20201130_1606758195.tar.gz generated
Maintenance mode disabled
Install Nextcloud 20.0.2...
cp: cannot stat 'nextcloud-old/apps/previewgenerator': No such file or directory
Abort
Simple roll back...
Clean up...

```

u8gMarvin, 2020-11-30:
I'm having the same problem

u8gMarvin, 2020-11-30:
Just install this app in app store (https://apps.nextcloud.com/apps/previewgenerator) and run the update again. Work for me :)

davidoe, 2020-11-30:
Installing previewgenerator solved this problem, thanks @u8gMarvin

nachoparker, 2020-12-01:
thanks for reporting. I'll push out a quick fix

nachoparker, 2020-12-01:
fixed, thanks

closed 2020-12-01 by nachoparker