nextcloud/nextcloudpi #1212: Feature Request: Upgrade to Php 7.4 possible?

JimmyKater, 2020-12-02:
**Description**  
It would be wonderful if you'd do the upgrade to Php 7.4 sometime in the near future... It worked like a charm with upgrading to Php 7.3 not too long ago which I considered wonderful.

Maybe - if you're afraid that not enough users are running nc 19+ it would be possible to only upgrade those?

Looking forward.

nachoparker, 2020-12-02:
Hi!

I don't think it's worth it honestly. The speed bump is not as nice as with 7.2 or 7.3, it's a decent amount of work, a source of bugs if we consider all platforms we support and also it's not compatible with older versions of NC.

closed 2020-12-02 by nachoparker

JimmyKater, 2020-12-02:
as much as I understand your points @nachoparker it's nevertheless recommended for being used with NC 20.x (and sooner or later there won't be any other help than upgrading it, though)

nachoparker, 2020-12-02:
we'll upgrade whenever we move to the next Debian release

nachoparker, 2020-12-02:
we'll upgrade whenever we move to the next Debian release. It is simply not a big enough speed improvement to justify the hassle.

JimmyKater, 2020-12-02:
well it's not only about any speed improvement (for me) but e.g. passwords app is gonna stop working from january 2021 on as they rely upon php 7.4 from then on.
and we shall see if nc 21 will make use of php 7.3 still.

teemue, 2021-03-06:
Any update on this?

ovpc, 2021-03-06:
I dont expect it untill it makes it into stable Debian.
As Nacho pointed out: It is not worth the troubles.

mohamnag, 2021-11-20:
I'm also having trouble with this. Some apps including passwords, are now stuck on legacy/maintenance/... versions. which means clients are not fully compatible and lots of small things that don't work. sure they are "functional" but thats running on lowest acceptable level.

why is this considered that much of work? which older versions of NC has to be supported? its almost a year later and this could mean its time to move on.

ovpc, 2021-11-20:
@mohamnag 
Bullseye will come with php7.4, and is not far away. see https://github.com/nextcloud/nextcloudpi/issues/1321
It is all volunteer work, no one "works" here. So there are no timelines.
You are most welcome to spend some of your free time and help testing or doing a PR if you feel up to the task.