nextcloud/nextcloudpi #1219: Debian Buster: do not expect dphys-swapfile to work everywhere

supersexy, 2020-12-09:
Next error: 
```
Setting up dphys-swapfile (20100506-5) ...
Job for dphys-swapfile.service failed because the control process exited with error code.
See "systemctl status dphys-swapfile.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dphys-swapfile, action "start" failed.
● dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file
   Loaded: loaded (/lib/systemd/system/dphys-swapfile.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-12-09 18:33:46 CET; 15ms ago
     Docs: man:dphys-swapfile(8)
  Process: 32755 ExecStart=/sbin/dphys-swapfile setup (code=exited, status=0/SUCCESS)
  Process: 322 ExecStart=/sbin/dphys-swapfile swapon (code=exited, status=255/EXCEPTION)
 Main PID: 322 (code=exited, status=255/EXCEPTION)

Dec 09 18:33:46 server.name systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Dec 09 18:33:46 server.name dphys-swapfile[32755]: computing size, want /var/swap=2068MByte, restricting to config limit: 2048MBytes, checking existing: keeping it
Dec 09 18:33:46 server.name dphys-swapfile[322]: swapon: /var/swap: swapon failed: Operation not permitted
Dec 09 18:33:46 server.name systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 09 18:33:46 server.name systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Dec 09 18:33:46 server.name systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

This happens on an OpenVZ virtual server - this is still a technology used at big hosters and it does not allow to create swap files.
Your installation script should check for that or make it configurable.

montvid, 2020-12-13:
Hi, got the same error on my local debian buster 10.7 clean netinstall. How to fix this? I am using btrfs on kernel 4.19 so swapfiles are not supported.

montvid, 2020-12-13:
Solved the interupted installation by:
`systemctl disable dphys-swapfile`

closed 2021-01-20 by nachoparker