nextcloud/nextcloudpi #1224: Can't connect to the database on nextcloud login screen

montvid, 2020-12-13:
After the script install I ran into some problems. The solution was to do nc-init to reinit nextcloud from the web interface but had to enable the web interface from the command line as it was not enabled after install.

 The web interface shows version 1.18 after install and that version 1.33 is available but the webinterface does not help install it. How to upgrade the web interface?

The end of the installation script is cryptic too as 'NC init done' never shows see:

Downloading updates
Performing updates

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin password.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.18.0
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir               (doesn't exist)
rootfs usage         4.2G/816G
swapfile             none
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enp3s0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               41min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

                                                            
  There are no commands defined in the "config" namespace.  
                                                            

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sat Dec 12 19:52:30.923904 2020] [mpm_event:notice] [pid 693:tid 139745907971200] AH00493: SIGUSR1 received.  Doing graceful restart
[Sat Dec 12 19:52:30.933615 2020] [ssl:warn] [pid 693:tid 139745907971200] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat Dec 12 19:52:30.933683 2020] [ssl:error] [pid 693:tid 139745907971200] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=debesis.pkjc.lt / issuer: CN=debesis.pkjc.lt / serial: 09C44F92B960810B16E75D6F62448428613C90AA / notbefore: Dec 12 23:57:01 2020 GMT / notafter: Dec 10 23:57:01 2030 GMT]
[Sat Dec 12 19:52:30.933711 2020] [ssl:error] [pid 693:tid 139745907971200] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Sat Dec 12 19:52:30.934145 2020] [mpm_event:notice] [pid 693:tid 139745907971200] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Sat Dec 12 19:52:30.934160 2020] [core:notice] [pid 693:tid 139745907971200] AH00094: Command line: '/usr/sbin/apache2'
[Sat Dec 12 19:52:46.743671 2020] [authnz_external:error] [pid 11949:tid 139745592350464] [client 192.168.1.232:58510] AuthExtern pwauth [/usr/sbin/pwauth]: Failed (1) for user admin
[Sat Dec 12 19:52:46.743733 2020] [auth_basic:error] [pid 11949:tid 139745592350464] [client 192.168.1.232:58510] AH01617: user admin: authentication failure for "/": Password Mismatch
[Sat Dec 12 19:52:55.044594 2020] [authnz_external:error] [pid 11948:tid 139745886095104] [client 192.168.1.232:58512] AuthExtern pwauth [/usr/sbin/pwauth]: Failed (1) for user admin
[Sat Dec 12 19:52:55.044644 2020] [auth_basic:error] [pid 11948:tid 139745886095104] [client 192.168.1.232:58512] AH01617: user admin: authentication failure for "/": Password Mismatch
[Sat Dec 12 19:53:42.550376 2020] [proxy_fcgi:error] [pid 11948:tid 139745886095104] [client 192.168.1.232:58548] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/index.php on line 252PHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/index.php on line 252'
[Sat Dec 12 19:56:35.396508 2020] [proxy_fcgi:error] [pid 11948:tid 139745869309696] [client 192.168.1.232:58548] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/index.php on line 252PHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/index.php on line 252', referer: https://192.168.1.212:4443/wizard/
[Sat Dec 12 19:56:41.251140 2020] [proxy_fcgi:error] [pid 11948:tid 139745877702400] [client 192.168.1.232:58548] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/backups.php on line 127'
[Sat Dec 12 19:57:53.573400 2020] [proxy_fcgi:error] [pid 11949:tid 139745592350464] [client 192.168.1.232:58734] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/index.php on line 252PHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/index.php on line 252', referer: https://192.168.1.212:4443/wizard/
[Sat Dec 12 19:57:58.794619 2020] [proxy_fcgi:error] [pid 11949:tid 139745592350464] [client 192.168.1.232:58734] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/backups.php on line 127'
[Sat Dec 12 19:58:16.390742 2020] [proxy_fcgi:error] [pid 11948:tid 139745877702400] [client 192.168.1.232:58736] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/index.php on line 252PHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/index.php on line 252', referer: https://192.168.1.212:4443/wizard/
[Sat Dec 12 19:58:21.564803 2020] [proxy_fcgi:error] [pid 11948:tid 139745869309696] [client 192.168.1.232:58736] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/backups.php on line 127'
[Sat Dec 12 19:58:41.262815 2020] [proxy_fcgi:error] [pid 11948:tid 139745835738880] [client 192.168.1.232:58738] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/index.php on line 252PHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/index.php on line 252', referer: https://192.168.1.212:4443/wizard/
[Sat Dec 12 19:58:46.371118 2020] [proxy_fcgi:error] [pid 11948:tid 139745844131584] [client 192.168.1.232:58738] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: CONFIG in /var/www/ncp-web/backups.php on line 127'
[Sat Dec 12 20:01:22.974109 2020] [proxy_fcgi:error] [pid 11948:tid 139745860916992] [client 192.168.1.232:58750] AH01071: Got error 'PHP message: PHP Notice:  Undefined index: datadirectory in /var/www/ncp-web/backups.php on line 127'

```
</details>

<details>
<summary>Database logs</summary>

```

2020-12-12 20:18:02 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2020-12-12 20:18:02 0 [Note] InnoDB: Number of pools: 1
2020-12-12 20:18:02 0 [Note] InnoDB: Using generic crc32 instructions
2020-12-12 20:18:02 0 [Note] InnoDB: Initializing buffer pool, total size = 256M, instances = 1, chunk size = 128M
2020-12-12 20:18:02 0 [Note] InnoDB: Completed initialization of buffer pool
2020-12-12 20:18:02 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2020-12-12 20:18:02 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-12-12 20:18:02 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-12-12 20:18:02 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2020-12-12 20:18:02 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-12-12 20:18:02 0 [Note] InnoDB: Waiting for purge to start
2020-12-12 20:18:02 0 [Note] InnoDB: 10.3.27 started; log sequence number 4906785; transaction id 3382
2020-12-12 20:18:02 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2020-12-12 20:18:02 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-12-12 20:18:02 0 [Note] InnoDB: Buffer pool(s) load completed at 201212 20:18:02
2020-12-12 20:18:02 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-12-12 20:18:02 0 [Note] Reading of all Master_info entries succeeded
2020-12-12 20:18:02 0 [Note] Added new Master_info '' to hash table
2020-12-12 20:18:02 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.27-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

montvid, 2020-12-13:
Ok, got into nextcloud - clicked on the top ncp icon and landed into a activation page that does not work - spins infinitely. 
The error ncp shows:

[ nc-admin ] (Sat Dec 12 20:53:09 EST 2020)
User does not exist

nachoparker, 2020-12-13:
your install didnt

closed 2020-12-13 by nachoparker

montvid, 2020-12-13:
I ran the install script - what else should i do to install ncp? I got this error and solved it https://github.com/nextcloud/nextcloudpi/issues/1219#issuecomment-743925477 then ran the script again.

montvid, 2020-12-13:
the script always ends the same whatever I do:

Performing updates

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin password.
root@debesis:~#

montvid, 2020-12-13:
people are having the same problem after the script as I. 
https://help.nextcloud.com/t/nextcloudplus-not-yet-initialized-trying-again-in-a-few-seconds/31019/58

doniom, 2021-02-07:
my solution is... a clean install debian net...
and install a swap file...
https://www.elektronik-kompendium.de/sites/raspberry-pi/2002131.htm

and now it works all fine... the script works like a charm...