nextcloud/nextcloudpi #1227: Update to Nextcloud 2.0.3 via nc-update-nextcloud results in flooding nextcloud logs

PDriver993, 2020-12-22:
When updating nextcloudpi docker image latest from nextcloud 2.0.2. to 2.0.3 the nextcloud logs will be flooded by following messages:

```
[no app in context] Warning: Temporary directory /var/www/nextcloud/data/tmp is not present or writable

GET /index.php/apps/logreader/poll?lastReqId=X%40IcawiYtWz9-SS-5djckgAAAgU
from 172.19.0.1 by ncp at 2020-12-22T16:18:52+00:00
```

**Howto reproduce:**
- do a docker run with the image from 2020-12-22 and login to 127.0.0.1 to activate nextcloud installation.
  (docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 my.nextcloud.server)
- Then login to 127.0.0.1:4443 and set credentials for admin and nextcloud andmin
- login to nextcloud at 127.0.0.1, goto "settings->logging". see that log holds only some messages, but nothing more.
- get back to ncp panel at 127.0.0.1:4443 and update ncp from 1.32.0 to 1.33.0 by using "nc update"
- next update nextcloud by the ncp panel from 2.0.2 to 2.0.3 by using "nc-update-nextcloud". wait until updating finished.
- get back to nextcloud at 127.0.0.1 and log in as ncp and look at "settings->logging". Now log will be flooded by the messages from missing temporary folder...

Also restarting of container or rebooting does not change this.

**Expected behavior:** no flooding log entries after update.

ovpc, 2020-12-22:
Thanks for reporting.
I cannot reproduce this tho, updated a test docker on local machine from 18.0.4 via 19.0.6 to 20.0.4
using first ncp-update, then nc-update-nextcloud. No problems.

Your error  Warning: Temporary directory /var/www/nextcloud/data/tmp
suggest an other issue, with settings of sys_temp_dir value in php.ini, Nothing to do with update imho.
Please search the forum as this is not the place to provide support

PDriver993, 2020-12-22:
@ovpc  Thanks for checking. I'm not sure what happens on updating.
I can reproduce this if I deleted all containers/images/volumes (e.g. docker system prune -a -f --volumes) and then make the docker run from my inital post, pulling the latest docker image. I don't know if this can be reproduced by those version steps you mentioned. Anyway I changed the config.php  from

  'tempdirectory' => '/var/www/nextcloud/data/tmp',
to
  'tempdirectory' => '/tmp',

This solves to flooding log. But why happened that?

BTW: my docker installation uses "data-root" entry in daemon.json because all docker volumes/images should be on separate large ssd. 

----------
Now I have a second problem: 
If I start the container with -v ncdata:/data, and opened a shell for the container I do not see any user added data in the **"/data"** folder.  The folder user data was added to is now **"/data-ro"**

root folder content of container:
```
cd /
# ls
bin   data     dev  home  lib64  mnt  proc  run		  sbin	sys  usr
boot  data-ro  etc  lib   media  opt  root  run-parts.sh  srv	tmp  var

```

IMHO using "-v ncdata:/data" moves not the user data into "ncdata" volume, only the installation files. Can you check this too?

ovpc, 2020-12-23:
>sudo ls /home/user/ncpdockerdrive/ncdata/
bin  database  etc  ncp  nextcloud

Looks to me, like you got more in there then you should. Please post the forum for further support.

closed 2020-12-23 by ovpc

PDriver993, 2020-12-23:
> > ~ $ sudo ls /home/user/ncpdockerdrive/ncdata/
> > bin  database  etc  ncp  nextcloud
> 
> Looks to me, like you got more in there then you should. Please post the forum for further support.

Sure this is on your host visible. Please try on your host to access user added data in "/home/user/ncpdockerdrive/ncdata/". It will not be there....

I mentioned to do this in a docker shell login to the running container and then do a "ls /". It will list "/data" and "/data-ro". 

Anyway it seems I'm not the only one which faces this issue. I replied in nextcloud forum on some of those posts:
https://help.nextcloud.com/t/docker-volume-does-not-contain-user-data/101808

ovpc, 2020-12-23:
in mine userdata is present:
> ~ $ sudo ls ncpdrive/ncdata/nextcloud/data/
appdata_ocheyzvxylq7  files_external  index.html  ncp  ncp-update-backups  news  nextcloud.log	tmp

>  ~ $ sudo ls ncpdrive/ncdata/nextcloud/data/ncp/files/
Documents  Nextcloud intro.mp4	Nextcloud Manual.pdf  Nextcloud.png  Photos

I will check link later today, if it turns out to be a bug, we can reopen this thread.

PDriver993, 2020-12-23:
Many thanks 👍 . But can you please add a file by drag&drop via nextcloud web page. I see also those initial files but not the files I added later by a webdav client or drag&drop by browser. They appear only in that data-ro folder in an overlay volume....
  
I looked to my older instance running on a raspi and there everything is ok, but on fresh install (linux mint 20.04 and docker 20.10.1) something happend ....   strange.

ovpc, 2020-12-23:
Did as you asked and just updated https://help.nextcloud.com/t/omv5-amd64-all-updates-applied-docker-20-nextcloudpi-does-not-use-external-volume/101090/3