nextcloud/nextcloudpi #1228: Auto-delete Shieldy Telegram replies

sunjam, 2020-12-24:
This is a request for the Shieldy Telegram bot to auto-delete spam detection replies required of new users automatically.  Thanks for considering!

ovpc, 2020-12-24:
there are several options...
/deleteEntryMessages 
is one of them, is that what you are suggesting?
or
/deleteEntryOnKick 
?

all shield commands:
/help — shows this message
/language — switches language
/captcha — changes type of captcha used
/timeLimit — changes amount of time given to newcomers
/lock — makes commands accessible only by admins
/restrict — restricts newcomers to send only text messages in the first 24 hours
/deleteEntryMessages — delete messages about user entry
/greeting — greets users who pass the test
/trust — reply with this command to a message sent by user that you don't want to check
/ban — same as trust, but the opposite
/strict — use when you don't want to receive any newcomers' messages but captcha solutions until they pass captcha
/customCaptchaMessage — setup custom captcha message
/deleteGreetingTime — setup when to delete Shieldy's greeting in seconds like /deleteGreetingTime 100, reset by setting it to 0
/banUsers — whether to ban or to kick users
/deleteEntryOnKick — whether to delete entry messages for users who failed captcha
/cas — whether to use Combot Anti-Spam or not
/underAttack — toggle the mode to automatically kick all newcomers
/noAttack — disables Shieldy
/noChannelLinks — automatically delete messages that link to other telegram channels
/viewConfig — view the current Shieldy configuration
/buttonText — change the captcha button text in the form of /buttonText I'm not a bot!
/allowInvitingBots — whether users can invite other bots or not
/greetingButtons — setup buttons for greeting message

sunjam, 2020-12-24:
yeah, delete entry messages seems to be it... Assuming it would remove the
user response after they complete the challenge.

ovpc, 2020-12-24:
done ;-)

closed 2020-12-24 by ovpc