nextcloud/nextcloudpi #1229: shieldy bot - deleteEntryOnKick & Greetings

sunjam, 2020-12-24:
/deleteEntryOnKick also?

_Originally posted by @ovpc in https://github.com/nextcloud/nextcloudpi/issues/1228#issuecomment-750924923_

Yes.  If the user passes challenge/response, I was looking at customizing a greeting for them.  It'd be useful to message the user.  [Seeing if a private message is possible here](https://github.com/backmeupplz/shieldy/issues/121) as opposed to in-channel.

Do you think in-line messaging is a good answer?  My understanding is the user receives in-line messages without alerting the whole channel.  This could be useful for giving "welcome, quick basic channel info" to that user once they successfully pass Shieldy.

ovpc, 2020-12-24:
What kind of greeting are you thinking of?
I can think of following:
- Request to check pinned message
- provide link to docs
- and link to forum

- ?

sunjam, 2020-12-24:
Users reported being confused up joining, so:

Welcome!  You are limited to text-only posts for 24 hours.  If you are seeking technical support, please create a forum post at https://help.nextcloud.com and add the tag "ncp".  See https://docs.nextcloudpi.com for guides and more info.

ovpc, 2020-12-24:
implemented ;-)

sunjam, 2020-12-24:
Cool. We can re-open if further issues pop up.  Happy Holidays.

closed 2020-12-24 by sunjam