nextcloud/nextcloudpi #1232: ncp.log every day empty

daschmidt1994, 2020-12-29:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.33.0
NextCloudPi image    NextCloudPi_docker_11-27-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          usb1 synoboot 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ecryptfs
data disk usage      742G/1.8T
rootfs usage         742G/1.8T
swapfile             /dev/zram3
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    20.0.3.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
          REMOVED
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "20.0.3.2",
        "overwrite.cli.url": "https:\/\/cloud.daschmidt.at\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
[... 108 lines cut ...]
{"reqId":"z07gfMyDWegIw4iOSKe9","level":3,"time":"2020-12-29T16:30:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":{"Exception":"RuntimeException","Message":"image not found: image:actions/delete.png webroot: serverroot:/data/nextcloud","Code":0,"Trace":[{"file":"/data/nextcloud/apps/settings/lib/Activity/SecurityProvider.php","line":91,"function":"imagePath","class":"OC\\URLGenerator","type":"->","args":["core","actions/delete.png"]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":454,"function":"parse","class":"OCA\\Settings\\Activity\\SecurityProvider","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":359,"function":"parseEvent","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":164,"function":"sendEmailToUser","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["daniel","danielschmidt1@gmx.at","de","Europe/Vienna",1609259401]},{"file":"/data/nextcloud/apps/activity/lib/BackgroundJob/EmailNotification.php","line":65,"function":"sendEmails","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":[500,1609259401]},{"file":"/data/nextcloud/lib/private/BackgroundJob/Job.php","line":52,"function":"run","class":"OCA\\Activity\\BackgroundJob\\EmailNotification","type":"->","args":[null]},{"file":"/data/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":59,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"/data/nextcloud/cron.php","line":127,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"/data/nextcloud/lib/private/URLGenerator.php","Line":236,"CustomMessage":"Failed sending activity email to user \"{user}\""},"userAgent":"--","version":"20.0.3.2"}
{"reqId":"qB5slZqBpg6YIPk7ZJ8i","level":3,"time":"2020-12-29T17:00:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":{"Exception":"RuntimeException","Message":"image not found: image:actions/delete.png webroot: serverroot:/data/nextcloud","Code":0,"Trace":[{"file":"/data/nextcloud/apps/settings/lib/Activity/SecurityProvider.php","line":91,"function":"imagePath","class":"OC\\URLGenerator","type":"->","args":["core","actions/delete.png"]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":454,"function":"parse","class":"OCA\\Settings\\Activity\\SecurityProvider","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":359,"function":"parseEvent","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":164,"function":"sendEmailToUser","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["daniel","danielschmidt1@gmx.at","de","Europe/Vienna",1609261201]},{"file":"/data/nextcloud/apps/activity/lib/BackgroundJob/EmailNotification.php","line":65,"function":"sendEmails","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":[500,1609261201]},{"file":"/data/nextcloud/lib/private/BackgroundJob/Job.php","line":52,"function":"run","class":"OCA\\Activity\\BackgroundJob\\EmailNotification","type":"->","args":[null]},{"file":"/data/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":59,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"/data/nextcloud/cron.php","line":127,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"/data/nextcloud/lib/private/URLGenerator.php","Line":236,"CustomMessage":"Failed sending activity email to user \"{user}\""},"userAgent":"--","version":"20.0.3.2"}
{"reqId":"oLRhSU03dM2GpRLZTRMa","level":3,"time":"2020-12-29T17:30:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":{"Exception":"RuntimeException","Message":"image not found: image:actions/delete.png webroot: serverroot:/data/nextcloud","Code":0,"Trace":[{"file":"/data/nextcloud/apps/settings/lib/Activity/SecurityProvider.php","line":91,"function":"imagePath","class":"OC\\URLGenerator","type":"->","args":["core","actions/delete.png"]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":454,"function":"parse","class":"OCA\\Settings\\Activity\\SecurityProvider","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":359,"function":"parseEvent","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["de",{"__class__":"OC\\Activity\\Event"}]},{"file":"/data/nextcloud/apps/activity/lib/MailQueueHandler.php","line":164,"function":"sendEmailToUser","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":["daniel","danielschmidt1@gmx.at","de","Europe/Vienna",1609263001]},{"file":"/data/nextcloud/apps/activity/lib/BackgroundJob/EmailNotification.php","line":65,"function":"sendEmails","class":"OCA\\Activity\\MailQueueHandler","type":"->","args":[500,1609263001]},{"file":"/data/nextcloud/lib/private/BackgroundJob/Job.php","line":52,"function":"run","class":"OCA\\Activity\\BackgroundJob\\EmailNotification","type":"->","args":[null]},{"file":"/data/nextcloud/lib/private/BackgroundJob/TimedJob.php","line":59,"function":"execute","class":"OC\\BackgroundJob\\Job","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]},{"file":"/data/nextcloud/cron.php","line":127,"function":"execute","class":"OC\\BackgroundJob\\TimedJob","type":"->","args":[{"__class__":"OC\\BackgroundJob\\JobList"},{"__class__":"OC\\Log"}]}],"File":"/data/nextcloud/lib/private/URLGenerator.php","Line":236,"CustomMessage":"Failed sending activity email to user \"{user}\""},"userAgent":"--","version":"20.0.3.2"}
{"reqId":"tSazgmYw2CuE2bxrdSU4","level":3,"time":"2020-12-29T18:00:02+00:00","remoteAddr":"","user":"--","app":"activity","method":"","url":"--","message":{"Exception":"RuntimeException","Message":"image not found: image:actions [... cut ...] r.php","Line":236,"CustomMessage":"Failed sending activity email to user \"{user}\""},"userAgent":"--","version":"20.0.3.2"}

```
</details>

I almost look daily at my ncp.log for my photo previes. But the ncp.log is every day only a few lines long.

my /etc/logrodate.d/ncp is:

root@nextcloudpi:/# cat etc/logrotate.d/ncp 
/var/log/ncp.log
{
        rotate 4
        size 100M
        missingok
        notifempty
        compress
}

is it possible, that ncp-backup or ncp-rsync-auto the log clear?

schoetju, 2021-07-15:
Hi daschmidt1994,

i've noticed similar. Today I realized the root-cause:
My /etc/cron.daily/ncp-autoupdate has been quite old and the script is based on a previous nextcloudpi-version.
Inside this script there has been a 
exec > /var/log/ncp.log 2>&1
(line 4) which obviously starts a new ncp.log

The current nextcloudpi-version (1.37.0) generates this script correctly with 
exec >> /var/log/ncp.log 2>&1

Solution in my case:
1.) Start _sudo ncp-config_
2.) Navigate to _Updates_
3.) start _nc-autoupdate-ncp_
4.) Choose _yes_
This recreates the  /etc/cron.daily/ncp-autoupdate 
and fixes this issue.

Hope that helps

Best regards
schoetju

theCalcaholic, 2021-07-16:
For an according issue was in `/etc/cron.d/ncp-backup-auto`

schoetju, 2021-07-17:
@theCalcaholic Good find too... My ncp-backup-auto also still clears the ncp.log :-) - I've been too soon...

In case of ncp-backup-auto the root-cause is still there: The script
[https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup-auto.sh](https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-backup-auto.sh)
creates in line 43 the incorrect 
/etc/cron.d/ncp-backup-auto
file. 
@nachoparker: Do you agree, that there should be a 
`... >> /var/log/ncp.log 2>&1" ...
`instead of
`... > /var/log/ncp.log 2>&1" ...`
?

nachoparker, 2021-08-03:
yes, good find folks

nachoparker, 2021-08-07:
fixed in https://github.com/nextcloud/nextcloudpi/pull/1309 

thanks

closed 2021-08-07 by nachoparker