nextcloud/nextcloudpi #1233: unattendes upgrades does not apply the upgrades available in stable

svenb1234, 2021-01-01:
From reading

https://ownyourbits.com/2017/03/21/automatic-security-updates-on-a-debian-system/#comment-2319

> For now, unattended upgrades will upgrade the _whole_ system, there is no way to apply only security updates. I will now update the post with this information

and

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=82863&p=1191453#p1210645
`In Debian Stable (Stretch) which Raspbian Stretch follows, there are only security updates. That's what "Stable" means.`

I understand that all upgrades to stable are security updates. If I run unattendes-upgrades, no packages to upgrade are shown.

> pi@nextcloudpi:~ $ sudo unattended-upgrades --dry-run -v
> Initial blacklist :
> Initial whitelist:
> Starting unattended upgrades script
> Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspberry Pi Foundation,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
> package raspi-config upgradable but fails to be marked for upgrade ()
> Option --dry-run given, *not* performing real actions
> Packages that will be upgraded:

In more detail, there are packages available but not installed, which seems wrong since "apt upgrade" would install them and they are part of the stable repo AFAICS. E.g. Checking: apache2 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])

> pi@nextcloudpi:~ $ sudo unattended-upgrades -d
> Initial blacklist :
> Initial whitelist:
> Starting unattended upgrades script
> Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspberry Pi Foundation,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
> Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
> Using (^linux-image-5\.4\.79\-v7l\+$|^linux-headers-5\.4\.79\-v7l\+$|^linux-image-extra-5\.4\.79\-v7l\+$|^linux-modules-5\.4\.79\-v7l\+$|^linux-modules-extra-5\.4\.79\-v7l\+$|^linux-signed-image-5\.4\.79\-v7l\+$|^linux-image-unsigned-5\.4\.79\-v7l\+$|^kfreebsd-image-5\.4\.79\-v7l\+$|^kfreebsd-headers-5\.4\.79\-v7l\+$|^gnumach-image-5\.4\.79\-v7l\+$|^.*-modules-5\.4\.79\-v7l\+$|^.*-kernel-5\.4\.79\-v7l\+$|^linux-backports-modules-.*-5\.4\.79\-v7l\+$|^linux-modules-.*-5\.4\.79\-v7l\+$|^linux-tools-5\.4\.79\-v7l\+$|^linux-cloud-tools-5\.4\.79\-v7l\+$|^linux-buildinfo-5\.4\.79\-v7l\+$|^linux-source-5\.4\.79\-v7l\+$) regexp to find running kernel packages
> Checking: apache2 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-bin ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-data ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-utils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apt ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apt-utils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: base-files ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: bind9-host ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils-arm-linux-gnueabihf ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils-common ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: ca-certificates ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: curl ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: dbus ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: distro-info-data ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: dnsutils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: e2fsprogs ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: freetype2-doc ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: fuse ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: git ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: git-man ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: iputils-ping ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libapt-inst2.0 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libapt-pkg5.0 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libbind9-161 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libbinutils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libbrotli1 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libcom-err2 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libcups2 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libcurl3-gnutls ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libcurl4 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libdbi-perl ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libdbus-1-3 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libdns-export1104 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: libdns1104 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
[... cut ...]
        "data-fingerprint": "597c55b577e23a7077d2ff90438b8d4f",
        "theme": "",
        "log_type": "file",
        "has_rebuilt_cache": true,
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "enable_previews": true,
        "enabledPreviewProviders": [
            "OC\\Preview\\Image",
            "OC\\Preview\\Movie",
            "OC\\Preview\\TXT",
            "OC\\Preview\\MKV",
            "OC\\Preview\\MP4",
            "OC\\Preview\\AVI"
        ]
    }
}

```
</details>

nachoparker, 2021-01-01:
good catch! it seems like they change the origin from `o=Raspberry Pi Foundation` to `o=Raspbian`.

I sent a fix, can you try it out please? run `sudo ncp-update devel` and try again `sudo unattended-upgrades -d` after that

svenb1234, 2021-01-01:
Thanks for the quick response! One question in advance, `sudo ncp-update devel` is a one time devel update, any following updates will be taken from the "stable" ncp?

The change seems to work:

> pi@nextcloudpi:~ $ sudo ncp-update devel
> INFO: updating to development branch 'devel'
> Downloading updates
> Performing updates
> Running unattended-upgrades
> Unattended upgrades active: yes (autoreboot true)
> NextCloudPi updated to version v1.34.1
> pi@nextcloudpi:~ $ sudo unattended-upgrades -d
> Initial blacklist :
> Initial whitelist:
> Starting unattended upgrades script
> Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspbian,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
> Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
> Using (^linux-image-5\.4\.79\-v7l\+$|^linux-headers-5\.4\.79\-v7l\+$|^linux-image-extra-5\.4\.79\-v7l\+$|^linux-modules-5\.4\.79\-v7l\+$|^linux-modules-extra-5\.4\.79\-v7l\+$|^linux-signed-image-5\.4\.79\-v7l\+$|^linux-image-unsigned-5\.4\.79\-v7l\+$|^kfreebsd-image-5\.4\.79\-v7l\+$|^kfreebsd-headers-5\.4\.79\-v7l\+$|^gnumach-image-5\.4\.79\-v7l\+$|^.*-modules-5\.4\.79\-v7l\+$|^.*-kernel-5\.4\.79\-v7l\+$|^linux-backports-modules-.*-5\.4\.79\-v7l\+$|^linux-modules-.*-5\.4\.79\-v7l\+$|^linux-tools-5\.4\.79\-v7l\+$|^linux-cloud-tools-5\.4\.79\-v7l\+$|^linux-buildinfo-5\.4\.79\-v7l\+$|^linux-source-5\.4\.79\-v7l\+$) regexp to find running kernel packages
> Checking: apache2 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-bin ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-data ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apache2-utils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apt ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: apt-utils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: base-files ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: bind9-host ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils-arm-linux-gnueabihf ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: binutils-common ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: ca-certificates ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: curl ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: dbus ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: distro-info-data ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: dnsutils ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: e2fsprogs ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: freetype2-doc ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: fuse ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: git ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: git-man ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
> Checking: iputils-ping ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
[... cut ...]
> Checking: libjpeg62-turbo ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
[... 159 lines cut ...]
> ncurses-bin
> ncurses-term
> nfs-common
> nfs-kernel-server
> openssh-client
> openssh-server
> openssh-sftp-server
> php7.3
> postfix
> python-apt
> python-apt-common
> python3-apt
> python3.7
> python3.7-minimal
> redis-server
> redis-tools
> rpcbind
> ssh
> sudo
> systemd
> systemd-sysv
> tzdata
> udev
> wpasupplicant
> ...

nachoparker, 2021-01-01:
> Thanks for the quick response! One question in advance, sudo ncp-update devel is a one time devel update, any following updates will be taken from the "stable" ncp?

Correct,

Thanks!

closed 2021-01-01 by nachoparker

svenb1234, 2021-01-02:
Just to sum this issue up. I let `unattended-upgrades` do its work this night and it did upgrade most packages. But some are still missing/ignored. 

> Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspbian,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
> Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
> Using (^linux-image-5\.4\.79\-v7l\+$|^linux-headers-5\.4\.79\-v7l\+$|^linux-image-extra-5\.4\.79\-v7l\+$|^linux-modules-5\.4\.79\-v7l\+$|^linux-modules-extra-5\.4\.79\-v7l\+$|^linux-signed-image-5\.4\.79\-v7l\+$|^linux-image-unsigned-5\.4\.79\-v7l\+$|^kfreebsd-image-5\.4\.79\-v7l\+$|^kfreebsd-headers-5\.4\.79\-v7l\+$|^gnumach-image-5\.4\.79\-v7l\+$|^.*-modules-5\.4\.79\-v7l\+$|^.*-kernel-5\.4\.79\-v7l\+$|^linux-backports-modules-.*-5\.4\.79\-v7l\+$|^linux-modules-.*-5\.4\.79\-v7l\+$|^linux-tools-5\.4\.79\-v7l\+$|^linux-cloud-tools-5\.4\.79\-v7l\+$|^linux-buildinfo-5\.4\.79\-v7l\+$|^linux-source-5\.4\.79\-v7l\+$) regexp to find running kernel packages
> Checking: libp11-kit0 **([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])**
> Checking: raspi-config **([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])**
> pkgs that look like they should be upgraded: libp11-kit0
> Fetched 0 B in 0s (0 B/s)
> fetch.run() result: 0
> dpkg is configured not to cause conffile prompts
> Packages that will be upgraded: libp11-kit0
> Writing dpkg log to /var/log/unattended-upgrades/unattended-upgrades-dpkg.log
> applying set ['libp11-kit0']
> apt-listchanges: Reading changelogs...
> (Reading database ... 50635 files and directories currently installed.)
> Preparing to unpack .../libp11-kit0_0.23.15-2+deb10u1_armhf.deb ...
> Unpacking libp11-kit0:armhf (0.23.15-2+deb10u1) over (0.23.15-2) ...
> Setting up libp11-kit0:armhf (0.23.15-2+deb10u1) ...
> Processing triggers for libc-bin (2.28-10+rpi1) ...
> left to upgrade set()
> All upgrades installed
> InstCount=0 DelCount=0 BrokenCount=0
> Extracting content from /var/log/unattended-upgrades/unattended-upgrades-dpkg.log since 2021-01-02 09:10:31
> Found /var/run/reboot-required, rebooting
> Shutdown msg: b"Shutdown scheduled for Sun 2021-01-03 04:00:00 GMT, use 'shutdown -c' to cancel."

`sudo apt upgrade ` as comparison:

> pi@nextcloudpi:~ $ sudo apt upgrade
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> The following NEW packages will be installed:
>   busybox initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base
> The following packages will be upgraded:
>   raspi-config
> 1 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
> Need to get 774 kB of archives.
> After this operation, 1,708 kB of additional disk space will be used.
> Do you want to continue? [Y/n]

Rebooted NCP manually. After that everything seems ok since` raspi-config` is from `archive testing` which is not supposed to be installed, isn't? And the packages to be installed instead of upgraded, like busybox, linux-base etc. depend on raspi-config I guess and are thus not needed as well?

> pi@nextcloudpi:~ $ sudo apt upgrade
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> The following NEW packages will be installed:
>   busybox initramfs-tools initramfs-tools-core klibc-utils libklibc linux-base
> The following packages will be upgraded:
>   raspi-config
> 1 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
> Need to get 774 kB of archives.
> After this operation, 1,708 kB of additional disk space will be used.
> Do you want to continue? [Y/n] n
> Abort.

> pi@nextcloudpi:~ $ sudo unattended-upgrades -d
> Initial blacklist :
> Initial whitelist:
> Starting unattended upgrades script
> Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspbian,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
> Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source- [... cut ...]
> Checking: raspi-config ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
> pkgs that look like they should be upgraded:
> Fetched 0 B in 0s (0 B/s)
> fetch.run() result: 0
> dpkg is configured not to cause conffile prompts
> No packages found that can be upgraded unattended and no pending auto-removals
> Extracting content from /var/log/unattended-upgrades/unattended-upgrades-dpkg.log since 2021-01-02 09:25:15