nextcloud/nextcloudpi #1235: Odroid HC4 | Debian Buster Server arm64  | install script fails

eyduh, 2021-01-04:
Hi,
I'm running this debian buster server image:

debian-buster-server-odroidhc4-20201218.img.xz

from 

http://ppa.linuxfactory.or.kr/images/raw/arm64/buster/

on an SSD on an odroid hc4

I ran the install script a few times which seems to work fine at first but it always ends with:
```
Downloading updates
Performing updates
/usr/local/etc/library.sh: line 263: [: v0: integer expression expected
/usr/local/etc/library.sh: line 265: [: v0: integer expression expected
/usr/local/etc/library.sh: line 267: [: v0: integer expression expected
/usr/local/etc/library.sh: line 263: [: v0: integer expression expected
/usr/local/etc/library.sh: line 265: [: v0: integer expression expected
/usr/local/etc/library.sh: line 267: [: v0: integer expression expected
/usr/local/etc/library.sh: line 263: [: v0: integer expression expected
/usr/local/etc/library.sh: line 265: [: v0: integer expression expected
/usr/local/etc/library.sh: line 267: [: v0: integer expression expected

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin password.
```

Which leaves me with an ncp activation page looking for an aborted nextcloud init to finish.

`NextCloudPi not yet initialized, trying again in a few seconds ...`

Output of `sudo ncp-report`:

<details>
<summary>NextCloudPi diagnostics</summary>

```

cat: /usr/local/etc/ncp-version: No such file or directory
/usr/local/bin/ncp-diag: line 36: swapon: command not found
NextCloudPi version
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir               (doesn't exist)
rootfs usage         3.9G/16G
swapfile             none
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:01

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud is not installed - only a limited number of commands are available

                                                            
  There are no commands defined in the "config" namespace.  
                                                            

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Mon Jan 04 01:20:10.614863 2021] [mpm_event:notice] [pid 6501:tid 281473801749872] AH00489: Apache/2.4.38 (Debian) configured -- resuming normal operations
[Mon Jan 04 01:20:10.615267 2021] [core:notice] [pid 6501:tid 281473801749872] AH00094: Command line: '/usr/sbin/apache2'
[Mon Jan 04 01:24:19.545370 2021] [mpm_event:notice] [pid 6501:tid 281473801749872] AH00491: caught SIGTERM, shutting down
[Mon Jan 04 01:24:19.685239 2021] [mpm_event:notice] [pid 23593:tid 281473755911536] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Mon Jan 04 01:24:19.744823 2021] [core:notice] [pid 23593:tid 281473755911536] AH00094: Command line: '/usr/sbin/apache2'
[Mon Jan 04 01:27:35.919752 2021] [mpm_event:notice] [pid 23593:tid 281473755911536] AH00493: SIGUSR1 received.  Doing graceful restart
AH00112: Warning: DocumentRoot [/var/www/ncp-web/] does not exist
[Mon Jan 04 01:27:35.951482 2021] [ssl:warn] [pid 23593:tid 281473755911536] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Mon Jan 04 01:27:35.951760 2021] [ssl:error] [pid 23593:tid 281473755911536] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=buster-server / issuer: CN=buster-server / serial: 623222D86104BD952BE953D0D2D988D19F9B2DF8 / notbefore: Jan  4 01:21:35 2021 GMT / notafter: Jan  2 01:21:35 2031 GMT]
[Mon Jan 04 01:27:35.951792 2021] [ssl:error] [pid 23593:tid 281473755911536] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Mon Jan 04 01:27:35.953048 2021] [mpm_event:notice] [pid 23593:tid 281473755911536] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Mon Jan 04 01:27:35.953098 2021] [core:notice] [pid 23593:tid 281473755911536] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2021-01-04  1:48:26 0 [Note] InnoDB: Uses event mutexes
2021-01-04  1:48:26 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-01-04  1:48:26 0 [Note] InnoDB: Number of pools: 1
2021-01-04  1:48:26 0 [Note] InnoDB: Using generic crc32 instructions
2021-01-04  1:48:26 0 [Note] InnoDB: Initializing buffer pool, total size = 256M, instances = 1, chunk size = 128M
2021-01-04  1:48:26 0 [Note] InnoDB: Completed initialization of buffer pool
2021-01-04  1:48:26 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-01-04  1:48:26 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-01-04  1:48:26 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-01-04  1:48:26 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-01-04  1:48:26 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-01-04  1:48:26 0 [Note] InnoDB: 10.3.27 started; log sequence number 1625639; transaction id 20
2021-01-04  1:48:26 0 [Note] InnoDB: Loading buffer pool(s) from /var/lib/mysql/ib_buffer_pool
2021-01-04  1:48:26 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-01-04  1:48:26 0 [Note] InnoDB: Buffer pool(s) load completed at 210104  1:48:26
2021-01-04  1:48:26 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-01-04  1:48:26 0 [Note] Reading of all Master_info entries succeeded
2021-01-04  1:48:26 0 [Note] Added new Master_info '' to hash table
2021-01-04  1:48:26 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.27-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/nextcloud.log' for reading: No such file or directory

```
</details>

ovpc, 2021-01-04:
Please note: A clean and minimal Debian Buster is expected. See https://docs.nextcloudpi.com/en/curl-installer/
Try the ncp image for odroid from https://ownyourbits.com/downloads/

eyduh, 2021-01-04:
> Please note: A clean and minimal Debian Buster is expected. See https://docs.nextcloudpi.com/en/curl-installer/

The compressed image is 205mb, I would say that's quite minimal. The only thing installed before the install script was curl to fetch the install script.

> Try the ncp image for odroid from https://ownyourbits.com/downloads/

It won't boot from SSD.

ovpc, 2021-01-04:
Yes, no probs, but back up now, I think someone posted an issue with odroid was fixed by there devs.
Closing this for now, we can re-open if needed for ncp bug reporting

closed 2021-01-04 by ovpc