nextcloud/nextcloudpi #1236: Passwords (20.0.3) update fail

FriendlyGecko, 2021-01-04:
When attempting to access my cloud I am forced into a mandatory update of an application. I have attempted updating with both ncp-config and with the web console, both fail. When updating through web console, the detailed log gives me:

Preparing update

Set log level to debug

Turned on maintenance mode

Repair step: Repair MySQL collation

Repair info: All tables already have the correct collation -> nothing to do

Repair step: Repair SQLite autoincrement

Repair step: Copy data from accounts table when migrating from ownCloud

Repair step: Drop account terms table when migrating from ownCloud

Updating database schema

Updated database

An error occurred.

ovpc, 2021-01-04:
Maybe ask password devs or use forum, doesnt look like its ncp relates, ncp does not use SQLite. Dont see how Owncloud fits the equation either, although it still shares codebase with Nextcloud...

FriendlyGecko, 2021-01-04:
I am going to upstream it. In the meantime do you know of a way for me to view all nextcloud applications/remove applications via the console?

ovpc, 2021-01-04:
https://help.nextcloud.com/t/how-to-remove-apps-from-the-terminal/63569

closed 2021-01-04 by ovpc