nextcloud/nextcloudpi #1237: Docker image does not store user added files in /data folder beginning with 1.32.0

PDriver993, 2021-01-04:
Other people and I observed docker image starting with 1.32.0 not storing user added files to /data folder on a fresh installed machine. This could be an essential problem when user wants to mount the /data folder to another storage than the docker's data-root is located on. See discussion on https://help.nextcloud.com/t/docker-volume-does-not-contain-user-data/101808/24

It seems that pre-initialization of the docker image causes the problem.
After a quick revert of the following files to the files of version 1.31.0 the problem was solved.

- docker/nextcloud/Dockerfile
- docker /nextcloud/020nextcloud
- docker/nextcloudpi/Dockerfile

IMHO pre-initialization does mix up the /data and /data-ro folders, and as a result the user added data will not be available in a mounted volume of a container. 

Howto reproduce with build-docker.sh:
# ensure you have a fresh installed Debian 10.x with up-to-date git and docker installed
# ENSURE that the ncdata volume is not existing before as it is on any fresh installed system. If a working/initialized ncdata volume is available from a prior installation lower than 1.32.0, the volume is mounted correctly and user data will be store on this as expected. But this will hide the problem, so use a completely new/uninitialized volume
```
git clone https://github.com/nextcloud/nextcloudpi.git
cd nextcloudpi
./build-docker x86
```
#wait until image created

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-x86 my.domain
```

#login to ncp using browser https://127.0.0.1 
#check datadir entry in system info of ncp
#    false behavior: "/data-ro/nextcloud/data" since 1.32.0
#    should be: "/data/nextcloud/data", until 1.31.0
#login via browser to 127.0.0.1:443 and add a folder "TEST-TEST", add some files to this folder
```
cd /var/lib
sudo find -iname "TEST-TEST"
```
# check where the folder now is stored
#    false behavior: folder is now located under /var/lib/docker/overlay2/l/<SOME-HASH-HERE>/diff/data-ro/nextcloud/data/ncp/files/TEST-TEST and /var/lib/docker/overlay2/l/<SOME-HASH-HERE>/merged/data-ro/nextcloud/data/ncp/files/TEST-TEST
#    should be: at /var/lib/docker/volumes/ncdata/_data/nextcloud/data/ncp/files/TEST-TEST

kapsner, 2021-01-06:
+1

experiencing the same issue

ddimatos, 2021-01-07:
Also experience the same for ARM

kapsner, 2021-01-07:
Let me add some more details:

The issue occurred using the docker image `ownyourbits/nextcloudpi-armhf:latest` from docker hub on a raspberry pi.

As indicated by @PDriver993, the issue does not occur using `v1.31.0`. I was able to solve it by building the `v1.31.0` docker image myself:

```bash
git clone -b v1.31.0 https://github.com/nextcloud/nextcloudpi.git
cd nextcloudpi
./build-docker.sh armhf
```

PDriver993, 2021-01-07:
@kapsner 
same with x86. I built different versions since 1.29.8 and the issue started with 1.32.0. So 1.31.0 was the latest working one for a fresh installation. 

My workaround for now: 
- did a docker run with the v1.31.0 tag including the volume wanted to use for data, which is initialized by it (this needs some time ;-) ).
- stopping the container
- then a "docker system prune" to remove all the containers and images which are not running
- same docker run but with "latest" tag and then updating ncp and nc to the latest versions.

Eukanuba863, 2021-01-08:
Same here... +1 to the list.

rednaz1337, 2021-01-09:
Same here, on a RPi 4 with `/data` on an external HDD.

jzcad1828, 2021-01-11:
I ran into this with Raspberry Pi 4, external USB SSD, docker repo (not git) using armhf (32-bit) on RaspbianOS Lite 20.04. Version 1.29.8 is available for using the aforementioned workarounds, however I had already deployed this with latest and had some data and used a different solution:

Stop the container and copy the data from the containers data-ro folder to the appropriate docker data volume from the docker host (using root account if possible):
`cp -a <path to container folder>/data-ro/nextcloud <path to nextcloud data volume>/_data/`

Edit the freshly copied config.php file:
`nano <path to nextcloud data volume>/_data/nextcloud/config/config.php`
Change the line:
`'datadirectory' => '/data-ro/nextcloud/data',`
to:
`'datadirectory' => '/data/nextcloud/data',`

Start the container. Verify that the NCP status page shows the correct path and that you can login with the ncp user (or another user) and access/create data. Verify data is created in the docker volume.

capio92, 2021-01-15:
Will this setting be overwritten when updating?

jzcad1828, 2021-01-15:
> Will this setting be overwritten when updating?

> Will this setting be overwritten when updating?

If you’re referring to my post, no, this change persists through recompose and software updates in NCP.

capio92, 2021-01-15:
> > Will this setting be overwritten when updating?
> 
> If you’re referring to my post, no, this change persists through recompose and software updates in NCP.

Thanks for fast response

nachoparker, 2021-01-16:
This is a bug. Either use an older version, or apply @jzcad1828 permanent fix.

nachoparker, 2021-01-19:
fixed it, you can `ncp-update`on that image or use the new one I'll upload shortly (v1.34.6)

closed 2021-01-19 by nachoparker

nachoparker, 2021-01-20:
make sure to `ncp-update` to at least `v1.34.7`

wsegatto, 2021-02-24:
Still happening in 1.34.7 for me.

Hyperspeed1313, 2021-03-07:
> I ran into this with Raspberry Pi 4, external USB SSD, docker repo (not git) using armhf (32-bit) on RaspbianOS Lite 20.04. Version 1.29.8 is available for using the aforementioned workarounds, [...]

These instructions saved my NCP instance that's been running for 4 months without me realizing this problem. Thanks!