nextcloud/nextcloudpi #1238: Problems at the Setup with No-IP

calledMonkey, 2021-01-06:
I wanted to allow external access so i followed the steps in the setup assistant and i chose noip. I registered everything at noip but if i click finish on the nextcloud setup assistant, i have this error: 
`[ no-ip ] (Wed Jan  6 03:13:50 GMT 2021)

Auto configuration for Linux client of no-ip.com.

Only one host [churchnercloud.ddns.net] is registered to this account.
It will be used.

New configuration file '/usr/local/etc/no-ip2.conf' created.

Failed to enable unit: Unit /run/systemd/generator.late/noip2.service is transient or generated.
System config value trusted_domains => 3 set to string churchnercloud.ddns.net
System config value overwrite.cli.url set to string https://churchnercloud.ddns.net/
noip DDNS enabled

[ nc-autoupdate-ncp ] (Wed Jan  6 03:14:08 GMT 2021)
automatic NextCloudPi updates enabled

[ dnsmasq ] (Wed Jan  6 03:14:09 GMT 2021)
System config value trusted_domains => 2 set to string churchnercloud.ddns.net
System config value overwrite.cli.url set to string https://churchnercloud.ddns.net/
dnsmasq enabled

[ letsencrypt ] (Wed Jan  6 03:14:15 GMT 2021)
usage: 
  certbot [SUBCOMMAND] [options] [-d DOMAIN] [-d DOMAIN] ...

Certbot can obtain and install HTTPS/TLS/SSL certificates.  By default,
it will attempt to use a webserver both for obtaining and installing the
certificate. 
certbot: error: argument -m/--email: expected one argument`

ovpc, 2021-01-06:
an email is required

closed 2021-01-06 by ovpc

calledMonkey, 2021-01-06:
I know, my Email was in Email, i filled everything out.

ovpc, 2021-01-06:
`certbot: error: argument -m/--email: expected one argument``

nachoparker, 2021-01-19:
I fixed this so it accepts an empty email