nextcloud/nextcloudpi #1239: Breaking changes in Nextcloud 21

ChristophWurst, 2021-01-12:
Hi there,

I found that your app uses some classes of Nextcloud's third party libraries. Since an upgrade of those for php8 compatibility, your app will have to be adjusted as well. See https://docs.nextcloud.com/server/latest/developer_manual/app_publishing_maintenance/upgrade-guide.html#doctrine-dbal for details. Your app uses `Doctrine\DBAL\Types\Type` which lost the old constants: https://github.com/doctrine/dbal/blob/3.0.x/UPGRADE.md#deprecated-type-constants.

Cheers

ChristophWurst, 2021-01-12:
Please note that we just merged a PR with our own constants for the types, so the app can be fully independent of any 3rdparty libraries: https://github.com/nextcloud/server/pull/25089.

nachoparker, 2021-05-23:
Thanks, updated

closed 2021-05-23 by nachoparker