nextcloud/nextcloudpi #1242: Required diskspace calculation is wrong

AlexanderNitsch, 2021-01-15:
### PROBLEMS
I'm trying to make a backup of my docker-nextcloudpi data through the nextcloudpi panel.
It calculates, that I need more than 30 TiB despite my folder of nextcloud data being just 133 GiB.
[image]

Output of df -h /media/mypassport:
Filesystem      Size  Used Avail Use% Mounted on
/dev/sdb        916G   77M  870G   1% /media/mypassport

Output of du -h /media/SanDiskSSD (where my docker volume lies):
Filesystem      Size  Used Avail Use% Mounted on
/dev/sda1       930G  133G  750G  16% /media/SanDiskSSD

**System information**

output of `sudo ncp-report` inside docker container:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.34.1
NextCloudPi image    NextCloudPi_docker_04-06-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda sdb 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      133G/930G
rootfs usage         46G/59G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    18.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "213.55.162.158",
            "1": "172.17.0.2",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
[... 105 lines cut ...]
```

{"reqId":"xaoWIXA2f9V0E7eIuI02","level":2,"time":"2020-12-31T13:15:15+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 10001 milliseconds with 0 bytes received (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}
{"reqId":"C0A306cwmRi4lVkuW5ho","level":2,"time":"2021-01-02T13:45:12+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 10001 milliseconds with 0 bytes received (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}
{"reqId":"WmyN50Kq12YZcyGIillb","level":2,"time":"2021-01-03T13:45:13+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Operation timed out after 10001 milliseconds with 0 bytes received (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}
{"reqId":"X-LDqzeeLLgBvb5MSrf8VAAAQAE","level":3,"time":"2021-01-04T07:28:43+00:00","remoteAddr":"192.168.1.1","user":"--","app":"index","method":"GET","url":"/svg/spreed/app-dark?color=000&v=1","message":{"Exception":"OCP\\App\\AppPathNotFoundException","Message":"Could not find path for spreed","Code":0,"Trace":[{"file":"/data/nextcloud/core/Controller/SvgController.php","line":100,"function":"getAppPath","class":"OC\\App\\AppManager","type":"->","args":["spreed"]},{"file":"/data/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":170,"function":"getSvgFromApp","class":"OC\\Core\\Controller\\SvgController","type":"->","args":["spreed","app-dark","000"]},{"file":"/data/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":99,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OC\\Core\\Controller\\SvgController"},"getSvgFromApp"]},{"file":"/data/nextcloud/lib/private/AppFramework/App.php","line":125,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OC\\Core\\Controller\\SvgController"},"getSvgFromApp"]},{"file":"/data/nextcloud/lib/private/AppFramework/Routing/RouteActionHandler.php","line":47,"function":"main","class":"OC\\AppFramework\\App","type":"::","args":["OC\\Core\\Controller\\SvgController","getSvgFromApp",{"__class__":"OC\\AppFramework\\DependencyInjection\\DIContainer"},{"app":"spreed","fileName":"app-dark","_route":"core.Svg.getSvgFromApp"}]},{"function":"__invoke","class":"OC\\AppFramework\\Routing\\RouteActionHandler","type":"->","args":[{"app":"spreed","fileName":"app-dark","_route":"core.Svg.getSvgFromApp"}]},{"file":"/data/nextcloud/lib/private/Route/Router.php","line":299,"function":"call_user_func","args":[{"__class__":"OC\\AppFramework\\Routing\\RouteActionHandler"},{"app":"spreed","fileName":"app-dark","_route":"core.Svg.getSvgFromApp"}]},{"file":"/data/nextcloud/lib/base.php","line":1008,"function":"match","class":"OC\\Route\\Router","type":"->","args":["/svg/spreed/app-dark"]},{"file":"/data/nextcloud/index.php","line":38,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"/data/nextcloud/lib/private/App/AppManager.php","Line":431,"CustomMessage":"--"},"userAgent":"Mozilla/5.0 (Android 10; Mobile; rv:85.0) Gecko/85.0 Firefox/85.0","version":"18.0.3.0"}
{"reqId":"B3aUe7ho2iz33qNvfLms","level":2,"time":"2021-01-04T14:00:13+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Connection timed out after 10000 milliseconds (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}
{"reqId":"XWqOuftFwb3ykkJY1RiI","level":2,"time":"2021-01-08T15:00:20+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Connection timed out after 10001 milliseconds (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}
{"reqId":"X-lkBTeeLLgBvb5MSrczuwAAAFQ","level":3,"time":"2021-01-09T08:06:29+00:00","remoteAddr":"193.118.53.210","user":"--","app":"no app in context","method":"GET","url":"/remote/login","message":{"Exception":"Symfony\\Component\\Routing\\Exception\\RouteNotFoundException","Message":"Unable to generate a URL for the named route \"calendar.view.index\" as such route does not exist.","Code":0,"Trace":[{"file":"/data/nextcloud/lib/private/Route/Router.php","line":339,"function":"generate","class":"Symfony\\Component\\Routing\\Generator\\UrlGenerator","type":"->","args":["calendar.view.index",[],1]},{"file":"/data/nextcloud/lib/private/Route/CachingRouter.php","line":59,"function":"generate","class":"OC\\Route\\Router","type":"->","args":["calendar.view.index",[],false]},{"file":"/data/nextcloud/lib/private/URLGenerator.php","line":82,"function":"generate","class":"OC\\Route\\CachingRouter","type":"->","args":["calendar.view.index",[]]},{"file":"/data/nextcloud/lib/private/NavigationManager.php","line":293,"function":"linkToRoute","class":"OC\\URLGenerator","type":"->","args":["calendar.view.index"]},{"file":"/data/nextcloud/lib/private/NavigationManager.php","line":114,"function":"init","class":"OC\\NavigationManager","type":"->","args":[]},{"file":"/data/nextcloud/apps/theming/lib/ThemingDefaults.php","line":182,"function":"getAll","class":"OC\\NavigationManager","type":"->","args":["guest"]},{"file":"/data/nextcloud/lib/private/legacy/defaults.php","line":263,"function":"getShortFooter","class":"OCA\\Theming\\ThemingDefaults","type":"->","args":[]},{"file":"/data/nextcloud/lib/public/Defaults.php","line":169,"function":"getLongFooter","class":"OC_Defaults","type":"->","args":[]},{"file":"/data/nextcloud/core/templates/layout.guest.php","line":55,"function":"getLongFooter","class":"OCP\\Defaults","type":"->","args":[]},{"file":"/data/nextcloud/lib/private/Template/Base.php","line":179,"args":["/data/nextcloud/core/templates/layout.guest.php"],"function":"include"},{"file":"/data/nextcloud/lib/private/Template/Base.php","line":151,"function":"load","class":"OC\\Template\\Base","type":"->","args":["/data/nextcloud/core/templates/layout.guest.php",null]},{"file":"/data/nextcloud/lib/private/legacy/template.php","line":181,"function":"fetchPage","class":"OC\\Template\\Base"," [... cut ...] CustomMessage":"--"},"userAgent":"Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:84.0) Gecko/20100101 Firefox/84.0","version":"18.0.3.0"}
{"reqId":"LGg325hjfxnIfO1bXJQt","level":2,"time":"2021-01-14T16:00:28+00:00","remoteAddr":"","user":"--","app":"appstoreFetcher","method":"","url":"--","message":"Could not connect to appstore: cURL error 28: Connection timed out after 10000 milliseconds (see http://curl.haxx.se/libcurl/c/libcurl-errors.html)","userAgent":"--","version":"18.0.3.0"}

```
</details>

nachoparker, 2021-01-16:
that is 30 GiB

AlexanderNitsch, 2021-01-19:
You are right. The real issue was, that I did not mount the backup folder into my docker container when starting.
adding a 
-v /media/mypassport/backup:/backup 
to the docker run command and selecting /backup in the ui solved the issue.

It's just weird, that it said "path exists".

Thank you for all your great work nachoparker :)

closed 2021-01-19 by AlexanderNitsch