nextcloud/nextcloudpi #1243: > Will this setting be overwritten when updating?

capio92, 2021-01-15:
> Will this setting be overwritten when updating?

If you’re referring to my post, no, this change persists through recompose and software updates in NCP.

_Originally posted by @jzcad1828 in https://github.com/nextcloud/nextcloudpi/issues/1237#issuecomment-760602292_

i got this issue:
mv: cannot move 'nextcloud-old/data' to 'nextcloud/data': Device or resource busy
Abort
Simple roll back...
Clean up...

someone had this issue to: https://help.nextcloud.com/t/docker-nextcloudpi-cannot-update-device-or-ressource-is-busy/101100

Potjoe-97, 2021-03-07:
Hi @capio92, I published the thread on the nextcloud forum. Still no luck, despite a full reinstall. I'm upgrading from within nextcloud now (not from ncp panel). If anyone is able to adress this...

computerproblems, 2021-03-14:
I get this same issue when trying to update with NCP (installed on debian with curl).  I've been stuck on an old nextcloud version for months.  

mv: cannot move 'nextcloud-old/data' to 'nextcloud/data': Device or resource busy
Abort
Simple roll back...
Clean up...

nachoparker, 2021-03-21:
@computerproblems what's the output of `lsof | grep nextcloud` and `mount`?

Potjoe-97, 2021-03-21:
Hi @nachoparker, here is the output of sudo lsof | grep nextcloud (too heavy for pastebin...): 
[report.txt](https://github.com/nextcloud/nextcloudpi/files/6178031/report.txt) 

And here is the output for mount : https://pastebin.com/imC70Yhi

computerproblems, 2021-03-22:
@nachoparker  

For sudo lsof | grep mount: https://pastebin.com/ZmyXn8mn

For sudo lsof | grep nextcloud: https://pastebin.com/V1qJq9iu There is more, but it's too long for free pastebin.  Do you need the rest?

computerproblems, 2021-04-11:
I was having this issue on a virtual machine with the datadir on separate mounted disk.  I made a new virtual machine, restored from backup with data and OS on same virtual disk, then I was able to upgrade nextcloud.

JonasDaWi, 2021-08-03:
I have the same problem with the datadir on a separate disk..

LennyKopisch, 2023-07-05:
Closed long ago but use ` sudo cp /var/www/nextcloud-old/data/* /var/www/nextcloud/data -a` then it will work.