nextcloud/nextcloudpi #1249: When removing the letsencrypt config it still tries to renew SSL Cert

weidi, 2021-01-21:
I just placed my NCP instance behind a reverse proxy / traefik that is taking care now of the ssl cert.
Now i keep getting alerts about failed SSL renewal and even after removing the SSL Config from NCP it still seems to try something here.

aphex008, 2021-04-08:
Same here. Waiting for PR to be accepted.

nachoparker, 2021-05-09:
PR in `devel` 

thanks

closed 2021-05-09 by nachoparker

cver, 2021-07-28:
@weidi would you mind telling me how to setup Nextcloudpi behind Traefik

weidi, 2021-08-01:
> @weidi would you mind telling me how to setup Nextcloudpi behind Traefik

Hi @cver, this is my docker-compose.yml

````
version: '3'
services:
  nextcloudpi:
    image: ownyourbits/nextcloudpi-x86
    restart: always
    command: "${IP}"
    ports:
#     - "80:80"
#     - "443:443"
     - "4443:4443"
    volumes:
     - ./nextcloud_data:/data
     - /etc/localtime:/etc/localtime:ro
     - ./ncdata:/media/USBdrive
    container_name: nextcloudpi
    networks:
      - proxy
    labels:
      - "traefik.enable=true"
      - "traefik.http.routers.nextcloudpi.entrypoints=http"
      - "traefik.http.routers.nextcloudpi.rule=Host(`nextcloud.your-url.com`)"
      - "traefik.http.middlewares.nextcloudpi-https-redirect.redirectscheme.scheme=https"
      - "traefik.http.routers.nextcloudpi.middlewares=nextcloudpi-https-redirect"
      - "traefik.http.routers.nextcloudpi-secure.entrypoints=https"
      - "traefik.http.routers.nextcloudpi-secure.rule=Host(`nextcloud.your-url.com`)"
      - "traefik.http.routers.nextcloudpi-secure.tls=true"
      - "traefik.http.routers.nextcloudpi-secure.tls.certresolver=http"
      - "traefik.http.routers.nextcloudpi-secure.service=nextcloudpi"
      - "traefik.http.services.nextcloudpi.loadbalancer.server.port=80"
      - "traefik.docker.network=proxy"
networks:
  proxy:
    external: true

```