nextcloud/nextcloudpi #1250: Since latest Update, new Docker Instances do not finish initializing

timethy96, 2021-01-27:
Hey hi hello from Germany and thank you a lot for the excellent work on this project.

For the past few days I spent hours and hours trying to find out, why my new Docker Containers of NCP suddenly do not initialize correctly anymore.

**Stuck on white screen with "Initializing NextcloudPi for the first time..." but with backend working and *Init done* in the logs**

I was experimenting around, so I was sure, that the problem came from there. It was only after trying out a lot of different things (changing sd-cards and power-supplies) and reading through every post I found, that I tried **rolling back** to an earlier version I still had (**v1.32.0**). Here no problems with the Init at all. Only that the now fixed bug with the data dir assigned to /data-ro/ was still there... (From there I can upgrade to the latest Version (**v1.34.6**) without loosing the functionality and manually change the datadir in the Nextcloud config.php)
From that I presume that there is a bug in the updates from the last few days.

**System information**

After getting it to work, I quickly set it back up to reproduce the steps to the bug and get a log to post here. I apologize for any inconsistencies in the logs due to the short running time.

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.34.6
NextCloudPi image    NextCloudPi_docker_01-19-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.9G/116G
rootfs usage         1.9G/116G
swapfile             /dev/zram0
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    18.0.3.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               9:14

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "172.17.0.2",
            "6": "192.168.0.108"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "18.0.3.0",
[... 72 lines cut ...]
2021-01-27  8:03:14 0 [ERROR] mysql.db: 1 client is using or hasn't closed the table properly
2021-01-27  8:03:14 0 [Note] Reading of all Master_info entries succeeded
2021-01-27  8:03:14 0 [Note] Added new Master_info '' to hash table
2021-01-27  8:03:14 0 [Note] mysqld: ready for connections.
Version: '10.3.22-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-01-27  8:03:20 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2021-01-27  8:03:20 0 [Note] Event Scheduler: Purging the queue. 0 events
2021-01-27  8:03:20 0 [Note] InnoDB: FTS optimize thread exiting.
2021-01-27  8:03:21 0 [Note] InnoDB: Starting shutdown...
2021-01-27  8:03:21 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2021-01-27  8:03:21 0 [Note] InnoDB: Buffer pool(s) dump completed at 210127  8:03:21
2021-01-27  8:03:22 0 [Note] InnoDB: Shutdown completed; log sequence number 4297011; transaction id 2780
2021-01-27  8:03:22 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-01-27  8:03:22 0 [Note] mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

timethy96, 2021-01-27:
@ovpc I got it working, so no support requested.

ovpc, 2021-01-27:
Sorry, thanks for repporting, but still think this should be in forum, if it is not a bug....
Weird,  how can you be on NC18 ? Thought latest docker would be on NC 20.0.4

timethy96, 2021-01-27:
Ok thank you and sorry I will post it there...
True, this is weird. I am working on arm64 (RPI4) and directly pulling through docker-compose from ownyourbits/nextcloudpi

ovpc, 2021-01-27:
Ok, closing this for now, we can re-open if it turns out to be a bug.

closed 2021-01-27 by ovpc

cheesyfries7, 2021-02-05:
> 
> 
> @ovpc I got it working, so no support requested.

hello, wondering how you got it working as I am encountering the same problem.

codeclinic, 2021-02-20:
> 
> 
> > @ovpc I got it working, so no support requested.
> 
> hello, wondering how you got it working as I am encountering the same problem.

ME too. I've followed all the instructions on the NCP website for Docker but get nowhere.

nachoparker, 2021-02-24:
make sure you use the latest image (use a tag if you want to be double sure), stop and remove all running containers and volumes.

cheesyfries7, 2021-02-24:
> 
> 
> > > @ovpc I got it working, so no support requested.
> > 
> > 
> > hello, wondering how you got it working as I am encountering the same problem.
> 
> ME too. I've followed all the instructions on the NCP website for Docker but get nowhere.

I fixed my issue by upgrading my nextcloud image via the nextcloudpi admin panel from 18 to 19, then 19 to 20. Then I was able to view the Nextcloud homepage. You can see #1256