nextcloud/nextcloudpi #1252: Download of Backups via NCP Panel not working

hansaushamburg, 2021-01-28:
After the update to NC 20.0.6 downloading of backupfiles via the NCP panel is not working. Error message is "unauthorized access". Backupfiles are shown in the panel with a minus (-1356 MiB).

I checked via SSH. The backups are carried out properly. The restore function also works fine.

It's just the download function ...

-->

**System information**
NextCloudPi version  v1.34.7
NextCloudPi image    NextCloudPi_09-05-20
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      78G/3.7T
rootfs usage         2.9G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.6.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               23:07

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "2a02:8108:483f:c0e4:fb82:e09c:6764:f91",
            "1": "192.168.178.61",
            "12": "cloud.hanslangen.de"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "20.0.6.1",
        "overwrite.cli.url": "https:\/\/cloud.hanslangen.de\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
[... 68 lines cut ...]

```

{"reqId":"ZW18HkjjpwaULY3xf41m","level":1,"time":"2021-01-28T20:21:06+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"20.0.6.1"}
{"reqId":"YBMjDqD0naDWHNvks25JmgAAAFY","level":4,"time":"2021-01-28T20:48:14+00:00","remoteAddr":"31.17.236.67","user":"Hans","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/Hans/","message":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":458,"function":"emit","class":"Sabre\\DAV\\Server","type":"->","args":["beforeMethod:PROPFIND",[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":251,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":319,"function":"start","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":332,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/remote.php","line":167,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"},"userAgent":"Mozilla/5.0 (Macintosh) mirall/2.6.5legacy (build 20200710) (Nextcloud)","version":"20.0.6.1"}
{"reqId":"YBMjLKD0naDWHNvks25JnAAAAFE","level":4,"time":"2021-01-28T20:48:45+00:00","remoteAddr":"31.17.236.67","user":"Hans","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/Hans/","message":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":458,"function":"emit","class":"Sabre\\DAV\\Server","type":"->","args":["beforeMethod:PROPFIND",[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":251,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":319,"function":"start","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":332,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/remote.php","line":167,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"},"userAgent":"Mozilla/5.0 (Macintosh) mirall/2.6.5legacy (build 20200710) (Nextcloud)","version":"20.0.6.1"}
{"reqId":"YBMjSqD0naDWHNvks25JnQAAAE0","level":4,"time":"2021-01-28T20:49:14+00:00","remoteAddr":"31.17.236.67","user":"Hans","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/Hans/","message":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/event/lib/WildcardEmitterTrait.php","line":89,"function":"checkMaintenanceMode","class":"OCA\\DAV\\Connector\\Sabre\\MaintenancePlugin","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":458,"function":"emit","class":"Sabre\\DAV\\Server","type":"->","args":["beforeMethod:PROPFIND",[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":251,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"/var/www/nextcloud/3rdparty/sabre/dav/lib/DAV/Server.php","line":319,"function":"start","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/lib/Server.php","line":332,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/dav/appinfo/v2/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/remote.php","line":167,"args":["/var/www/nextcloud/apps/dav/appinfo/v2/remote.php"],"function":"require_once"}],"File":"/var/www/nextcloud/apps/dav/lib/Connector/Sabre/MaintenancePlugin.php","Line":85,"CustomMessage":"--"},"userAgent":"Mozilla/5.0 (Macintosh) mirall/2.6.5legacy (build 20200710) (Nextcloud)","version":"20.0.6.1"}
{"reqId":"YBMjaKD0naDWHNvks25JnwAAAFc","level":4,"time":"2021-01-28T20:49:45+00:00","remoteAddr":"31.17.236.67","user":"Hans","app":"webdav","method":"PROPFIND","url":"/remote.php/dav/files/Hans/","message":{"Exception":"Sabre\\DAV\\Exception\\ServiceUnavailable","Message":"Das System befindet sich im Wartungsmodus.","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/sabre/ev [... cut ...] 2.6.5legacy (build 20200710) (Nextcloud)","version":"20.0.6.1"}
{"reqId":"YBMmdfMB4C301QPFWjmezgAAABM","level":2,"time":"2021-01-28T21:02:46+00:00","remoteAddr":"31.17.236.67","user":"Hans","app":"no app in context","method":"GET","url":"/ocs/v2.php/core/navigation/apps?absolute=true&format=json","message":"Temporary directory /media/myCloudDrive/ncdata/tmp is not present or writable","userAgent":"Mozilla/5.0 (Macintosh) mirall/2.6.5legacy (build 20200710) (Nextcloud)","version":"20.0.6.1"}

```

nachoparker, 2021-04-29:
Thanks,

This bug happens due to 32bit. We have 64bit images for the RPi now, but this has an easy workaround, so we'll fix.

nachoparker, 2021-04-29:
fixed in `devel` v1.35.1, thanks

closed 2021-04-29 by nachoparker