nextcloud/nextcloudpi #1253: apps and app store missing

nephilim75, 2021-01-30:
Hello

while trying to install NCPi on a RasPi 3+ the installation shows errors and at the end apps like Calendar and Contacts are missing. In the left hand navigation pane the apps store is missing as well.

```
curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
```

Has something changed? Past days (about 10 days ago) the installation went fine. No issues. But now a lot of PHP errors in a fresh installed cloud.

Kind regards
//neph

Lauffrosch, 2021-01-31:
Hello,
I experienced the same problem after updating from Nextcloud 20.0.2.0 to 20.0.4.0. All manually added apps are removed and the app store is not available-

ovpc, 2021-01-31:
Nothing changed, worked fine me, day or two ago, on a VM

closed 2021-01-31 by ovpc