nextcloud/nextcloudpi #1256: Nextcloud version in latest Docker image published doesn't match

jush, 2021-02-03:
I used the following docker compose yaml to start a fresh nextcloud container:
```
version: '3'
services:
  nextcloudpi:
    image: ownyourbits/nextcloudpi-arm64
    command: "192.168.1.162"
    ports:
     - "80:80"
     - "443:443"
     - "4443:4443"
    volumes:
     - ./data:/data
     - /etc/localtime:/etc/localtime:ro
    container_name: nextcloudpi
    restart: unless-stopped
```
Which should use this [docker image](https://hub.docker.com/layers/ownyourbits/nextcloudpi/latest/images/sha256-93b8c5a846add666db340a6740fce9927675f00bbb078c6245360db3672c1803).
In the system info interface I see:
```
NextCloudPi version | v1.34.6

Nextcloud version | 18.0.3.0
```

Unfortunately I run into the issue reported #1250.

f I read the [Changelod.md](https://github.com/nextcloud/nextcloudpi/blob/b978184ce6a5e7f7171bb73cdce8009367cfdaa9/changelog.md) file properly Nextcloud version should be `20.0.4` since v1.34.0 instead of `18.0.3.0`

Then I decided to build the docker image myself by cloning this repo and running command:
```
sudo docker build --progress=plain . -f docker/nextcloudpi/Dockerfile -t jush/nextcloudpi-arm64:latest --build-arg release=buster --build-arg arch=arm64 --build-arg ncp_ver=1.34.6
```

And the system info for that container is indeed showing Nextcloud version `20.0.4.0 ` and it initializes properly (that is, I can't replicate issue #1250)

The bug seems to be that docker images published are somehow using an old Nextcloud version.

rozpuszczalny, 2021-02-03:
+1.

I also had to clone repo and build an image with `./build-docker.sh arm64` to get around #1250.

IMO stale image is one problem. Another is lack of other version tags in Docker Hub. Currently it's possible to only install latest version, but not possible to install other versions (at least from `ownyourbits/nextcloudpi` image).

I think someone should take into consideration CI/CD pipeline

cheesyfries7, 2021-02-05:
Noob here, and was recently trying to deploy nextcloudpi via docker. Could not figure out what I was doing wrong as I was always stuck on initializing page, even after trying reinstalling my OS, trying on both 32 bit and even the beta 64 bit (am on an RPI 4 8gb). I found #1250 via my internet search as I have the exact same problem, which led me here. I do not know how to build an image but am hopeful this mismatch with the docker hub images is the cause of my problems. Hope it gets fixed soon so I can finally try nextcloudpi.

jakern, 2021-02-07:
For now I got around this by updating the version to 19.0.0 and then updating again to 20 (you can't jump 2 version numbers). Hope it helps you if you want to try it out and don't mind waiting for the updates to install

cheesyfries7, 2021-02-07:
> 
> 
> For now I got around this by updating the version to 19.0.0 and then updating again to 20 (you can't jump 2 version numbers). Hope it helps you if you want to try it out and don't mind waiting for the updates to install

This worked for me. Thank you very much!

nachoparker, 2021-02-16:
Thanks for reporting.

Not sure what happened, maybe the automatic upload failed and nobody noticed before, which is quite surprising.

Anyhow, I re-uploaded the multi-arch image. Also I added tags to it. The machine specific images they all have tags btw.

https://hub.docker.com/r/ownyourbits/nextcloudpi/tags?page=1&ordering=last_updated

https://hub.docker.com/r/ownyourbits/debian-ncp-arm64/tags?page=1&ordering=last_updated

closed 2021-02-16 by nachoparker

jush, 2021-02-16:
Gracias Nacho!

Ungetym, 2021-02-17:
I still have the exact problem that was initially described by jush on a system I set up today (RP4 with Raspberry Pi OS). Tried pulling the correct image directly (ownyourbits/nextcloudpi-armhf) as well as ownyourbits/nextcloudpi during setup and always end up with NextCloudPi v1.34.6, but Nextcloud version 18.0.3.0. Of course I can upgrade the NC or build the image myself, but that is only a workaround.

nachoparker, 2021-02-19:
1.34.6 is pretty recent, you might be working on an old volume?

https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md

alexlawrence, 2021-02-27:
I experienced the same issue with the same version as @Ungetym on a freshly installed Raspbian. The only thing I did was to install Docker and to execute `docker run` as described in the instructions.

nachoparker, 2021-02-28:
I triple checked, It works for me

```
docker pull ownyourbits/nextcloudpi-armhf           SIGINT(2) ↵  0.87 L 
Using default tag: latest
latest: Pulling from ownyourbits/nextcloudpi-armhf
Digest: sha256:487a68a096b4ac95aa6b4b959501b97b9e297a91196eae493c83d607d69ba057
Status: Image is up to date for ownyourbits/nextcloudpi-armhf:latest
docker.io/ownyourbits/nextcloudpi-armhf:latest
```

```
$ docker manifest inspect --verbose ownyourbits/nextcloudpi:latest
...
       {                                                                                                                      
                "Ref": "docker.io/ownyourbits/nextcloudpi:latest@sha256:487a68a096b4ac95aa6b4b959501b97b9e297a91196eae493c83d60
7d69ba057",                                                                                                                    
                "Descriptor": {                                                                                                
                        "mediaType": "application/vnd.docker.distribution.manifest.v2+json",                                   
                        "digest": "sha256:487a68a096b4ac95aa6b4b959501b97b9e297a91196eae493c83d607d69ba057",                   
                        "size": 5743,                                                                                          
                        "platform": {                                                                                          
                                "architecture": "arm",                                                                         
                                "os": "linux"                                                                                  
                        }                                                                                                      
                },         
```
ocker inspect --format='{{.RepoDigests}}' ownyourbits/nextcloudpi-armhf:v1.34.6
[ownyourbits/nextcloudpi-armhf@sha256:487a68a096b4ac95aa6b4b959501b97b9e297a91196eae493c83d607d69ba057]

```

Also, it matches 

https://hub.docker.com/r/ownyourbits/nextcloudpi/tags?page=1&ordering=last_updated

hermesonfg, 2021-03-01:
I am stuck on the same issue.... any news??!!

iarbina, 2021-03-13:
I just followed what @nachoparker did and got something different:

```$ docker pull ownyourbits/nextcloudpi-armhf
Using default tag: latest
latest: Pulling from ownyourbits/nextcloudpi-armhf
98098fa10651: Already exists 
198e6f06aa57: Already exists 
677eecee6870: Already exists 
a8e76f60f395: Already exists 
57e40f2eb49d: Already exists 
472b9dfcb495: Already exists 
a29bdd869c0a: Already exists 
4f4fb700ef54: Already exists 
a72791775c14: Already exists 
68d17a9ff599: Already exists 
af8ac99e4913: Already exists 
e196bb4e8f7a: Already exists 
7d63e5bd4ad3: Already exists 
0167f0c53fe3: Already exists 
3d9a4acc084e: Already exists 
6c6fac4c6061: Already exists 
8abe5dc0b1a3: Already exists 
d0ebeaf178b4: Already exists 
db8c232256e1: Already exists 
e95d03eec6c7: Already exists 
a8a94cebb807: Already exists 
1866704c268a: Already exists 
8b8932a299e9: Already exists 
27793c66ea08: Already exists 
3a81a6a639f8: Already exists 
42e4c73cf56f: Already exists 
Digest: sha256:487a68a096b4ac95aa6b4b959501b97b9e297a91196eae493c83d607d69ba057
Status: Downloaded newer image for ownyourbits/nextcloudpi-armhf:latest
docker.io/ownyourbits/nextcloudpi-armhf:latest
```
For the moment the same. But then running:

```
$ docker manifest inspect --verbose ownyourbits/nextcloudpi:latest
[
	{
		"Ref": "docker.io/ownyourbits/nextcloudpi:latest@sha256:93b8c5a846add666db340a6740fce9927675f00bbb078c6245360db3672c1803",
		"Descriptor": {
			"mediaType": "application/vnd.docker.distribution.manifest.v2+json",
			"digest": "sha256:93b8c5a846add666db340a6740fce9927675f00bbb078c6245360db3672c1803",
			"size": 5743,
			"platform": {
				"architecture": "arm64",
				"os": "linux"
			}
		},
```
Why do I'm getting `"architecture": "arm64"`? Is that correct? And then the image looks like is different one...

jush, 2021-03-16:
After many days I tried again by pulling the latest image from docker and the versions are still wrong.

The SHA matches the image @nachoparker published a month ago:
```
[
        {
                "Ref": "docker.io/ownyourbits/nextcloudpi:latest@sha256:93b8c5a846add666db340a6740fce9927675f00bbb078c6245360db3672c1803",
                "Descriptor": {
                        "mediaType": "application/vnd.docker.distribution.manifest.v2+json",
                        "digest": "sha256:93b8c5a846add666db340a6740fce9927675f00bbb078c6245360db3672c1803",
                        "size": 5743,
                        "platform": {
                                "architecture": "arm64",
                                "os": "linux"
                        }
                },
...
```

But the `Nextcloud version` is still `18.0.3.0` instead of the expected `20.0.4.0`. See:
[image]

Somehow it seems that when Nacho builds the image is building with the wrong Nextcloud version. When I build locally in a clean clone of this repo then I get the 20.* version.

ovpc, 2021-03-16:
I was under impression it was no longer platform bound,
and one can simply use 
> docker pull ownyourbits/nextcloudpi:latest

It is what I used on my PC to test upgrading to 21 yesterday, from the initial 20.0.8

So if you are using -armhf, you pull an older image, I think, and appearently....

mickev36, 2021-04-12:
Same issue here, Nextcloud is in version 18 with a fresh ownyourbits/nextcloudpi:latest image