nextcloud/nextcloudpi #1258: Raspberry Pi : Mircosoft repo add on my Nextcloudpi

rimko, 2021-02-04:
Hi, 

I read on reddit that [Raspbian have add Microsoft repo on Raspbian](https://www.reddit.com/r/raspberry_pi/comments/lc07um/microsoft_repo_stealth_added_in_latest_rpios/) (for vscode, to programing on pico raspberry...)
So i check on my raspberry only and unfortunately those with Raspbian was affected and my unfortunately nextcloudpi also.

So yeah personally i will think twice before using raspbian again (outside of prototyping or testing for exemple..) it's not a big deal at the end but the close thread on raspberry forum and the official respond send a message : Raspbian OS is made for our product and with a education purpose first.

szaimen, 2021-02-14:
Since NextcloudPi itself isn't based on Raspbian OS but on Debian Linux, I don't think that it should be affacted by this.
cc @nachoparker

dotlineX, 2021-02-18:
This has nothing to do with NextCloudPI, raspberrypi added the repo for ease of use. You can easly remove the repo by removing it form `/etc/apt/sources.list.d/vscode.list`
Please clode this issue

closed 2021-02-18 by ovpc