nextcloud/nextcloudpi #1259: Information on the "nc-restore" page is misleading for data-less backups

Raanelom, 2021-02-05:
Currently, the NCP web panel contains the following information for `nc-restore`: "This new installation will cleanup current
NextCloud instance, including files and database."
However, for data-less backups, the files are preserved.

Suggestion for new info-text:
"This new installation will cleanup the current NextCloud instance. For data-less backups, files will remain untouched. In case of a full backup *your current files will be removed*."

Raanelom, 2021-02-05:
I'll setup a simple PR

nachoparker, 2021-02-05:
in case of a full backup a copy will be stored and you will see

```
backing up existing $DATADIR to $DATADIR-$( date "+%m-%d-%y" )...
```

closed 2021-02-05 by nachoparker

Raanelom, 2021-02-06:
Ah, I've overlooked that line indeed. Will include it in the solution.
Could you perhaps give me a pointer on how to create a pull-request for NCP? I've created a branch locally, but when trying to push it to remote I get a permission denied.

ovpc, 2021-02-06:
Always use dev branch for PR's, Nacho can then push it. Or link to your branch here, if not member of the team yet, so he can have look at your code.
Thank you.

Raanelom, 2021-02-07:
If I understand you correctly, you say that all PRs should end up in `devel` before pushing them to master, right?
The problem is that I can't create a remote (feature) branch, thus being unable to create a PR. My code doesn't even reach the repository but lives locally. I guess I need your permission to create a new remote branch.

Raanelom, 2021-02-15:
Any update on my last comment?