nextcloud/nextcloudpi #1270: Suggestion: Save 28% on diskimage size. Use XZ compression instead of Bzip2.

rekcodocker, 2021-03-06:
**Description**  
NextcloudPi asks users to use the Torrent and please seed - to save on network costs.
Here is 25% savings on network costs with no downsides:
I suggest to use the widely availabel 'xz' compression.
It can be decompressed on Windows and Linux, just like the older bzip2. It is installed by default on many systems.

**Example**  
The raspberry pi image: 
RaspberryPi.tar.bz2 : 888MB
RaspberryPi.tar.xz : 647MB
Savings: 28% in internet-transfer-time + 28% in diskspace.

The VM ware image:
Nextcloud.Vmware.bz2: 1.5G
Nextcloud.Vmware.xz: 1.1G
26% savings in transfer time + 26% savings in diskspace.

People will feast and take to the streets, shouting 'let's all use xz compression!'
In 60 years people will tell their grandchildren about this improvement that saved a quarter of diskspace just by using a different algorithm. Their children will be bored because their latest quantum compression compresses everything into a single bit.
But until that day, the whole world will be a little happier.

ovpc, 2021-03-07:
Not sure of implications, but seems like a good idea. Thanks.
Let 's wait for Nacho to view and comment.

eroyee, 2021-06-23:
As a result of another project I've been interested in I recently came across this article:

http://www.nongnu.org/lzip/xz_inadequate.html

Having subsequently done a little research I suggest that perhaps zstd could be a better option? In my tests I find it generally results in a smaller file (than lzma-derived compression) and it _may_ be a better algorithm long term.

Definitely not something to feast or shout about, but mebbe a quiet beer?

rekcodocker, 2021-10-03:
Mind you, this is about distributing the image.  It is not about long term storage. 

For distribution I would recommend going for the smallest possible filesize, while accepting paying for it in longer compression/decompression times.
Bzip2 is ancient - there are better alternatives. 
That article is about long term storage and about compression-file-container-design.  All of it might be true, but it does not apply if I simply want to send a file from A to B in the cheapest way possible.

eroyee, 2021-10-03:
> Mind you, this is about distributing the image. It is not about long term storage.

Ah, but didn't you want your grandkids to see all this worked 60 years from now ;-)

rekcodocker, 2021-12-03:
> > Mind you, this is about distributing the image. It is not about long term storage.
> 
> Ah, but didn't you want your grandkids to see how all this worked 60 years from now ;-)

First of all, those grandkids will be able to decompress it without issue as every conceivable decompression algorithm is already built in to their 3D glasses. 
They will not be interested in ... a cloud server package.
I know. I have kids and I can't get them interested in my own 40 year old computer games, let alone a boring cloud server.   "And then... you can ... wait for it... store a file on it!" 
"Seriously grandpa? That's 30 seconds of my life I won't get back".

closed 2021-12-03 by rekcodocker