nextcloud/nextcloudpi #1276: support Collabora Office by default

Coernel82, 2021-04-10:
As wished before in #51 and #343 now could be the time to integrate Collabora Office (including the Server) in Nextcloud Pi as it is now possible.

https://nextcloud.com/blog/the-nextcloud-hub-ubuntu-appliance-now-available-for-raspberry-pi-4-and-more/

nachoparker, 2021-04-10:
unfortunately we still support some 32bit boards, so it cannot come installed by default, but it is as easy as installing the app for those who want that

closed 2021-04-10 by nachoparker

Coernel82, 2021-04-11:
Thank you for your reply! For other users who try to active the 64 Bit Collabora CODE server, there might be an error:
`App "Collabora Online - Built-in CODE Server (ARM64)" cannot be installed because the following dependencies are not fulfilled: The following architectures are supported: aarch64`

At your own risk (backups, in this case probable an SD-Card image are your friends) switch the Kernel to 64 bit
https://www.sindastra.de/p/1345/switch-your-raspberry-pi-to-64-bit-kernel-raspbian/

After installing the CODE server and the collabora app and having selected "use built in server" you might not be able to edit or create documents as the server cannot be found. ("Please try again later" - error message) Also rebooting may take very long (5 minutes or so) till you can use your Nextcloud again. Don't panic in that case - just wait.

However the error is known: https://github.com/CollaboraOnline/richdocumentscode/issues/116

So at the moment: Be patient!

nachoparker, 2021-04-11:
thanks for the info. The raspberry pi images are NOT 32bit Raspbian, they are 64bit Debian so that shouldn't be an issue.

Coernel82, 2021-04-11:
Should not be an issue but in fact was an issue, I had to enable the 64 bit kernel which was easy but not the default.

nachoparker, 2021-04-11:
you must be using an older image, the latest one is 64bits

sunjam, 2022-10-19:
Wondering about coming back to this now that we are defaulting to 64-bit builds.  The process of setting up Collabora is currently very awkward in testing and could use some possible finesse.