nextcloud/nextcloudpi #1278: Reopen #1256  build issue

jakern, 2021-04-17:
My only guess is that if you're building arm on a different server, the repo's etc/ncp.cfg isn't being updated before the build-docker.sh command is run?

#1256

jakern, 2021-04-17:
I built the image myself and I do get version 20.0.8.1 by default which aligns with what it in the ncp.cfg `"nextcloud_version": "20.0.8"`. That's really the only place where this could have gone wrong and not catastrophically wrong, no?

jakern, 2021-04-17:
It doesn't appear that using 21.0.1 in that file works though

WeiChihChern, 2021-04-20:
the latest docker image provided is outdated and either docker or this repo have been updated for two months. Not sure why tho.

nachoparker, 2021-04-29:
@jakern what is your issue with the build exactly?

@WeiChihChern you can always download the latest available image and then nc-update -> nc-update-nextcloud

closed 2021-10-15 by nachoparker