nextcloud/nextcloudpi #1280: Initialization process of NexcloudPi ends in eternal loop

JanBrink42, 2021-04-24:
Here is a bug report on the initialization process of NexcloudPi:

On my Raspberry Pi4, I started a container based on the following docker-compose.yaml:
```
version: '3'
services:
  nextcloudpi:
    image: ownyourbits/nextcloudpi-armhf
    command: "${IP}"
    ports:
     - "80:80"
     - "443:443"
     - "4443:4443"
    volumes:
     - ncdata4:/data
     - /etc/localtime:/etc/localtime:ro
    container_name: nextcloudpi

volumes:
  ncdata4:
```
IP is set to the local IP address of the Raspberry Pi.

After starting, I can access the web interface, receiving two passwords, and I can log into the admin area (Port 4443).
But when I navigate to the nextcloud itself (Port 80 or 443), a page telling the following is shown:
```
Initializing NextCloudPi for the first time
Please wait...
```
This site is refreshing itself in an eternal loop. Thus, Nextcloud cannot be used.

ovpc, 2021-04-25:
_ image: ownyourbits/nextcloudpi-armhf_
instead try:
image: ownyourbits/nextcloudpi:latest
There is only one docker image for all architectures.

closed 2021-04-25 by ovpc

JanBrink42, 2021-04-25:
Unfortunately, the other images leads to the same issue.

ovpc, 2021-04-25:
Then please start a thread on the forum, the ncp docker image works fine for me and most other users.
You can use the ncp and docker tag on the Nextcloud forum to get help concerning docker and ncp.
https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33