nextcloud/nextcloudpi #1281: Improve Default Security Headers

mucke5, 2021-04-26:
**Description**  
Scanning with https://securityheaders.com/ revealed some warnings, upcoming headers and additional information.
Especially the following:

> Content-Security-Policy | This policy contains 'unsafe-inline' which is dangerous in the style-src directive.

**Example**  
https://securityheaders.com/?q=try.nextcloud.com&followRedirects=on

nachoparker, 2021-04-29:
thanks for the heads up, I'll take a look

nachoparker, 2021-04-29:
is this with ncp-web or with nextcloud itself?

mucke5, 2021-04-29:
For security reasons, ncp-web is configured so that it cannot be accessed remotely.
It's only with nextcloud, see an example of NextcloudPi instance
https://securityheaders.com/?q=nc.jcjm.fr&followRedirects=on
`
Permissions Policy is a new header that allows a site to control which features and APIs can be used in the browser.
`

Additionally, I recommend setting FLoC opt-out in the permission policy by default, as mentioned here
https://paramdeo.com/blog/opting-your-website-out-of-googles-floc-network

UN-pi, 2021-05-05:
I checked my nextcloudpi-Website with [internet.nl](internet.nl) and [securityheaders.com](securityheaders.com).
Not all new header-options are known by different browsers.

For me there are 3 new different Security-header-options:

X-Frame-Options
Referrer-Policy-Options
Content-Security-Options

I added 3 lines to my /etc/apache2/sites-available/nextcloud.conf and it works for me without changing other code:
`# New Headeroptions:

Header always append X-Frame-Options SAMEORIGIN

Header always set Referrer-Policy "same-origin"

Header always set Permissions-Policy "geolocation=(),fullscreen=(self)"
`
I also tried to add a Content-Security-policy, but after that there where problems with the screen-design.
So I leave it for now. After checking the site with securityheader.com there are already options set by nextcloud.
For example STUN nextcloud.com. The problems occur with options like: "unsafe-inline" "blob" and "data".

I also added a line for HSTS:
`    Header always set Strict-Transport-Security "max-age=31536000; includeSubDomains "
`
I also changed the configuration for /etc/apache2/conf-available/http2.conf, because the fault-configuration uses weak ciphers and prefer AES-encryption while still enabling TLS 1. The pi is faster with Chacha20:
`#allow alll and disable TLS1

SSLProtocol all -SSLv3 -TLSv1 -TLSv1.1
`
`#define SSL TLS 1.2
#prefer CHACHA20 and with 1 weak cipher for max. compatibility (older safari/android-versions) and no DH2048

 SSLCipherSuite CHACHA20:ALL:!RSA:!DHE:!CAMELLIA:!aNULL:!eNULL:!LOW:!3DES:!MD5:!EXP:!PSK:!SRP:!DSS:!RC4:!SHA1:!SHA256:+TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384
#define SSL TLS 1.3

SSLCipherSuite TLSv1.3 TLS_CHACHA20_POLY1305_SHA256:TLS_AES_256_GCM_SHA384:TLS_AES_128_GCM_SHA256:TLS_AES_128_CCM_SHA256:TLS_AES_128_CCM_8_SHA256
`
This works for me on Pi4/2GB,Debian64bit, Kernel 5.10.17

nachoparker, 2021-05-09:
CSP and such are handled by Nextcloud itself, so please refer to them for suggestions.

As for NCP, we only accept TLS13 and we already use HSTS and CHACHA first.

Cheers

closed 2021-05-09 by nachoparker