nextcloud/nextcloudpi #1282: Internal Server Error after shutdown

KUGA2, 2021-04-29:
Hi,

I run ncp in docker on x86. (I am stuck on ncp 1.25 since my docker image is to old)
My system recently crashed (powerfail), and about an week later i rebootet it and started the docker again.

I manually ran updates dyndns and letsencrypt via web interface, no I get:

```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log
```

I tried a few reboots/container restarts without success.

**System information**
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.25.0
NextCloudPi image    NextCloudPi_docker_01-18-19
distribution         Debian GNU/Linux 9 \n \l
automount            no
USB devices          sda 
datadir              /data/nextcloud/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      133G/916G
rootfs usage         8.3G/93G
swapfile             /dev/dm-1
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    20.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               14min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
[... 159 lines cut ...]
root@280bb67ab9c4:/# ls /etc/php/7.2/fpm/conf.d/
10-mysqlnd.ini	20-bz2.ini	 20-dom.ini	  20-gd.ini	  20-igbinary.ini  20-mbstring.ini   20-phar.ini      20-simplexml.ini	20-sysvshm.ini	  20-xmlwriter.ini
10-opcache.ini	20-calendar.ini  20-exif.ini	  20-gettext.ini  20-intl.ini	   20-mysqli.ini     20-posix.ini     20-sockets.ini	20-tokenizer.ini  20-xsl.ini
10-pdo.ini	20-ctype.ini	 20-fileinfo.ini  20-gmp.ini	  20-json.ini	   20-opcache.ini    20-readline.ini  20-sysvmsg.ini	20-wddx.ini	  20-zip.ini
15-xml.ini	20-curl.ini	 20-ftp.ini	  20-iconv.ini	  20-ldap.ini	   20-pdo_mysql.ini  20-shmop.ini     20-sysvsem.ini	20-xmlreader.ini
root@280bb67ab9c4:/# apt show php-redis
Package: php-redis
Version: 5.3.2+4.3.0-7+0~20210228.34+debian9~1.gbp893bd4
Priority: optional
Section: php
Maintainer: Debian PHP PECL Maintainers <team+php-pecl@tracker.debian.org>
Installed-Size: 14.3 kB
Pre-Depends: php-common (>= 2:69~)
Depends: php8.0-redis, php-igbinary
Suggests: redis-server
Homepage: https://pecl.php.net/package/redis
Download-Size: 5754 B
APT-Manual-Installed: yes
APT-Sources: https://packages.sury.org/php stretch/main amd64 Packages
Description: PHP extension for interfacing with Redis

N: There is 1 additional record. Please use the '-a' switch to see it
```

Do you need full ncp.log?

ovpc, 2021-04-29:
Pull a fresh image, afraid there is no longer "NCP" support for: NextCloudPi_docker_01-18-19 & Debian GNU/Linux 9 \n \l
Should not be a problem if you have nc-backups to nc-restore

closed 2021-04-29 by ovpc

KUGA2, 2021-04-29:
Can you please point me to a guide for updating the container image?

I am afraid, that i will loose my nextcloud data but most of all the config (users, calendars etc)