nextcloud/nextcloudpi #1283: Ports are shown as closed where they are in fact open

Menneken, 2021-05-13:
Hi, it seems I have a well known problem: NCP shows ports as closed where they are in fact open (as shown by portchecker.co). As a consequence, my installation is unable to refresh the Let´s Encrypt Certificate or do any updates.
My Installation:
<html><body>

NextCloudPi version | v1.34.7
-- | --
NextCloudPi image | NextCloudPi_12-20-19
distribution | Raspbian GNU/Linux 10 \n \l
automount | yes
USB devices | sda sdb sdc
datadir | /media/myCloudDrive/ncdata
data in SD | no
data filesystem | btrfs
data disk usage | 495G/932G
rootfs usage | 6.4G/59G
swapfile | /var/swap
dbdir | /var/lib/mysql/ncdatabase
Nextcloud check | ok
Nextcloud version | 20.0.4.0
HTTPD service | up
PHP service | up
MariaDB service | up
Redis service | up
Postfix service | up
internet check | no
port check 80 | closed
port check 443 | closed

</body>
</html>

ovpc, 2021-05-13:
Thought already fixed in v1.36.2 
https://github.com/nextcloud/nextcloudpi/releases/tag/v1.36.2

stoneubi, 2021-05-13:
Updated from 1.35 to 1.36.2 and ports are shown as closed. Worked before update and nextcloud instance is reachable from internet. So this information is missleading. 

[image]

System: Raspberry4 8 GB RAM, Dietpi as distro, Nextcloud 20.0.8

nachoparker, 2021-05-13:
There was a bug where they would show as open when they were actually closed.

Seems like the fix was incomplete, can't test right now since my ports are not open yet on my new ISP.

In any case, this is informational only, and does not affect Letsencrypt or anything else.

nachoparker, 2021-05-13:
I might have fixed, but can't test. Can someone please run `sudo ncp-update devel` and try again, see if it shows correctly now?

ovpc, 2021-05-13:
Nope still showing closed for me in develk v1.36.3

nachoparker, 2021-05-13:
ok, then I need someone who can debug this, since my ports are closed ATM

cc @dsmic

vcbranco, 2021-05-13:
version 1.36.3 showing closed here to

vcbranco, 2021-05-13:
I would like to help but I don't know how

GNEUGNEUGNEU, 2021-05-24:
i'm here to help too if needed

nachoparker, 2021-05-25:
If someone has the skills to fix this, please send a PR. Otherwise we'll have to wait until my ports are open.

dsmic, 2021-05-25:
Sorry I can not reproduce the problem. The version before your changes still works on my setup…

Haraade, 2021-05-26:
> Sorry I can not reproduce the problem. The version before your changes still works on my setup…

Worked, before the upgrade. This feature now creates false information for the user and is useless. A user of nextclod should know how to open router ports. Nextcloudpi fixes firewall (UFW) for required gates.

suxlex, 2021-05-30:
Before the change from 1.35.0 to 1.36.0 both ports were shown as open and now unfortunately as closed.

nachoparker, 2021-06-02:
Fixed in `v1.36.3`, thanks for the help!

closed 2021-06-02 by nachoparker