nextcloud/nextcloudpi #1292: Image on Docker-Hub not up to date

mad73923, 2021-06-19:
Same issue as in #1256 
Somehow the Docker-Hub images don't get updated any more.
Stuck on v1.34.6

ovpc, 2021-06-19:
I dont think it the container is supposed to get any updates at all, just run ncp-update inside the container.

mad73923, 2021-06-19:
In my point of view it should. This is needed if you migrate your instance to a new machine including your data.
You can't start a new docker container with the image pulled from the hub and your dataset from a newer version.
I'll purpose an automated dockerhub push via Travis. PR pending

nachoparker, 2021-06-19:
when you restore a backup the nextcloud files and database are included, so no issues there

nachoparker, 2021-06-19:
> I'll purpose an automated dockerhub push via Travis. PR pending

:clap:

closed 2021-06-19 by nachoparker