nextcloud/nextcloudpi #1293: add Gitter (Matrix chat) to main Github repo

sunjam, 2021-06-23:
**Description**  
https://gitter.im is fully open source and available as public chat for any github repo.  We could add it to the main repository for NextcloudPi.  

- This would also serve as Matrix chat support for NextcloudPi
- can be bridged into our existing Telegram chat.  Could setup a test Telegram room bridge, then enable integration for the real room

theCalcaholic, 2021-06-24:
That sounds interesting... I've been thinking about providing a public bridged room for the nextcloudpi telegram group.

(In fact, I myself am using telegram via matrix exclusively).

I don't completely understand how the repository integration of gitter works, yet, but I'll look into that.

As I'm currently quite invested in the Matrix ecosystem (hosting my own server, working on my own client...), I'm very interested in learning more about that :D

sunjam, 2021-09-11:
Closing this issue now that #nextcloudpi:matrix.org is bridged into Telegram.

closed 2021-09-11 by sunjam