nextcloud/nextcloudpi #1294: Progress bar during Backup

dm807cam, 2021-06-28:
**Description**  
I have about 1 TB on my NextcloudPi instance. Running backups through the backup option on the NCP panel in the shell takes very long and I  cannot verify the progress of if the backup process is still running at all.

A simple progress bar or a percentage indicator in the shell during the backup would be of immense help.

dm807cam, 2021-09-18:
Can I please add to this: A confirmation if rsync backups were successful is also urgently needed. I am backing my instance up to another server through ssh and the rsync function but have no idea if the back was successful.

sunjam, 2021-11-23:
You could write your own rsync command based on the ncp shell script, but include `--progress` option

dm807cam, 2021-11-23:
@sunjam I have duplicati running on the same Pi to deal with my backups for now. I just believe that if a backup feature is provided by ncp, it should be able to provide basic functions such as a progress bad and backup verification.

conectionist, 2022-01-30:
Hello, I would also be interested in such a feature.
Same use-case basically, 1TB backup.
I started a backup at 18:45 and after 6 hours (0:45 at night) it still wasn't finished, so I had to pull the plug and start fresh the second day.
I don't necessarily mind waiting, but I'd like to know how long I have to wait, so I can plan accordingly.
Thanks!