nextcloud/nextcloudpi #1295: Implementation of deSEC | Secure DDNS clients

mucke5, 2021-06-28:
**Description**  
Running on open-source software and supported by SSE, [deSEC](https://desec.io/) is free for everyone to use.

**Example**  
DNSSEC
DNS information hosted with deSEC is signed using DNSSEC, always. We use state-of-the-art elliptic-curve cryptography and follow operational best practice.

Cloud Integration
Thanks to cloud integrations and language bindings, deSEC works out of the box in automated environments. Examples include Terraform providers and Go, Python, and JavaScript bindings.

Modern Record Types
We support a broad array of record types, including novel types such as HTTPS/SVCB (for CNAME-like behavior at the apex), CDNSKEY/CDS (RFC 8078, RFC 8901), or OPENPGPKEY, SMIMEA, and TLSA.

Scalability
Are you a web hoster? Start using deSEC, even with thousands of domains. Our global network ensures high availability and performance everywhere. Talk to us about your use case.

REST API
Configure your DNS information via a modern API. You can easily integrate our API into your scripts, tools, or even CI/CD pipeline.

IPv6
deSEC is fully IPv6-aware: administration can be done using v6, AAAA-records containing IPv6 addresses can be set up, our name servers are reachable via IPv6.

DANE / TLSA
Secure your web service with TLSA records, hardening it against fraudulently issued SSL certificates. You can also use other DANE techniques, such as OPENPGPKEY key exchange.

Let's Encrypt Integration
We provide easy integration with Let's Encrypt and their certbot tool. Further integration with ACME clients like acme.sh, lego, and Terraform is available.

Fast Updates
Updates to your DNS information will be published world-wide within a few seconds. Minimum required TTLs are low.

Low-latency Anycast
We run global networks of high-performance frontend DNS servers. Your query is routed to the closest server via Anycast, so clients receive answers as fast as possible.

Open Source
deSEC runs 100% on free and open-source software. Start hacking away!

Non-profit
deSEC is organized as a non-profit organization based in Berlin. We make sure that privacy is not compromised by business interest.

theCalcaholic, 2021-07-05:
I've had a look at deSec and it does look nice. I general, I think it would be a good idea to have a generic dyndns option as well (where you can just enter a url that's being called).

Also, the deSec API might be one way to offer an alternative to the letsencrypt verification process (that doesn't require port 80 - not sure if we still have that issue).
However, tying ourselves to non-standard specific provider drawbacks obviously has its own issues.

mucke5, 2022-03-09:
@theCalcaholic @nachoparker 
Can this be a way to implement it?
https://github.com/desec-io/certbot-dns-desec

jerabaul29, 2022-04-10:
Any update on this? Would be great to be able to use deSec as a free DNS provider :) .

theCalcaholic, 2022-04-12:
@jerabaul29 This would be a nice feature, but unfortunately we don't have the capacity to implement it ourselves in the near future.
We're open for PRs though.