nextcloud/nextcloudpi #1297: docker build for arm64 fails with 'Device or resource busy'

combr, 2021-07-05:
Hello!

I try to build docker image for raspberry pi 4, as shown in readme:
```
git clone https://github.com/nextcloud/nextcloudpi.git
cd nextcloudpi
./build-docker.sh arm64
```
It builds for some minutes, but end with this error
```
#17 579.9 NextCloudPi updated to version v1.37.0
#17 580.3 /bin/sed: cannot rename /etc/sedGbGAIJ: Device or resource busy
#17 ERROR: executor failed running [/bin/bash -c set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; apt-get install --n
all-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /usr/local/etc/ncp-config.d/nc-init-copy.cfg /usr/local/
p-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/config.php; sed -i 's|/media/USBdrive|/data/backups|' /u
al/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto.cfg; cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/nc
ig.d/nc-init.cfg; rm /.ncp-image; source /usr/local/etc/library.sh; find /usr/local/bin/ncp -name '*.sh' | while read l; do cleanup_script $l; done; apt-get autoremove -y; a
 clean; find /var/lib/apt/lists -type f | xargs rm; find /var/log -type f -exec rm {} \; ; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*; rm -f /var/log/alternatives.log
og/apt/*; rm /var/cache/debconf/*-old; echo "${ncp_ver}" > /usr/local/etc/ncp-version]: exit code: 4
------
 > [stage-0 11/12] RUN --mount=type=cache,target=/var/cache/apt --mount=type=cache,target=/var/lib/apt set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; tou
ocker-image; apt-get update; apt-get install --no-install-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /u
al/etc/ncp-config.d/nc-init-copy.cfg /usr/local/etc/ncp-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/co
hp; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto
cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/ncp-config.d/nc-init.cfg; rm /.ncp-image; source /usr/local/etc/library.sh; find /usr/local/bin/ncp -name '*.sh' | while read l
leanup_script $l; done; apt-get autoremove -y; apt-get clean; find /var/lib/apt/lists -type f | xargs rm; find /var/log -type f -exec rm {} ; ; rm -rf /usr/share/man/*; rm -
r/share/doc/*; rm -f /var/log/alternatives.log /var/log/apt/*; rm /var/cache/debconf/*-old; echo "v1.37.0" > /usr/local/etc/ncp-version:
------
executor failed running [/bin/bash -c set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; ....
```

Environment:

```
Kubuntu 20.04 
 Server Version: 20.10.2
 Storage Driver: overlay2
 Kernel Version: 5.11.6-051106-generic
 Operating System: Ubuntu 20.04.2 LTS
 Architecture: x86_64  
```
"git, docker, qemu-user-static, chroot and all the usual building tools." installed (but what is 'chroot'?)

Buschkampf, 2021-07-06:
Hi together,

I have had the same problems and error messages.

But having a look at:

https://hub.docker.com/r/ownyourbits/nextcloudpi

I followed this guide:

`docker pull ownyourbits/nextcloudpi`

Which worked without any problems.

CU

combr, 2021-07-06:
> I followed this guide:
> 
> `docker pull ownyourbits/nextcloudpi`
> 
> Which worked without any problems.

Thanks, i know about it. But that image have an nextcloud 18 (very outdated) version
And build-docker must build image with last 20.0.11 version, it's the difference.

nachoparker, 2021-07-07:
I can verify this. Looks like newer docker versions don't like that step.

That being said, you don't need to build from scratch. You can pull the latest from dockerhub and run the updates from there and you will be in the same spot.

combr, 2021-07-07:
> I can verify this. Looks like newer docker versions don't like that step.
> 
> That being said, you don't need to build from scratch. You can pull the latest from dockerhub and run the updates from there and you will be in the same spot.

Yes, thanks for your attention.
At first attempt I tried to update, but direct update to 'latest ver' (0) failed: it show error about 'cannot update between multiple versions' and stuck with show 'ver 20.0.11' in ncp info.
therefore in nc start page it show 'update required' and when I press  'update' it show error about 'multiple versions', and ncp-update-nc in ncp interface say 'no new version'
I forced to remove all docker image, all data, and re-download docker image, start it (it show ver 18) and update step by step:
18->19->20->21
This way was successful.

combr, 2021-07-07:
But there is no ncp-samba now, althrough in git it is:
 nextcloudpi/bin/ncp/NETWORKING/samba.sh 

I cannot "enable samba", also as "enable ssh" via ncp-config

nachoparker, 2021-07-27:
fixed

closed 2021-07-27 by nachoparker