nextcloud/nextcloudpi #1299: Nextcloud 22 Support

sunjam, 2021-07-07:
Nextcloud 22 When?

[Nextcloud 22 has been announced as of 7/6/2021](https://nextcloud.com/blog/nextcloud-hub-22-introduces-approval-workflows-integrated-knowledge-management-and-decentralized-group-administration/)
**It is not yet tested** on NextcloudPi.  Please be patient.  The community does not want to roll out an untested update, plus we are all volunteers.

**Can I help test?  Yes!**

If you do want to test it on your own instance you must do so manually by updating version-to-version from 20 to current 21.* to current 22.*

Please always maintain backups and be ready to use them.  Better yet, don't test this major update on your actual production instance you depend on.  Or just wait with the rest of us.  :+1: 

Please report any issues you encounter here so they can be addressed.

sunjam, 2021-08-02:
quick note for anyone following: #1298 will be resolved first.

sunjam, 2021-11-10:
[Preview Generation app is now available for version 22](https://apps.nextcloud.com/apps/previewgenerator).  Unsure on future maintenance.

ovpc, 2021-11-17:
Update to 22.2.1 worked fine in a test VM https://pastebin.com/PD7k982F preview-genarater throwing error when attempt update

ovpc, 2021-11-17:
Some one else reported the calander app broken on [forum](https://help.nextcloud.com/t/changes-to-calendar-events-and-contacts-are-not-saved-any-ideas/127462)

sunjam, 2021-11-17:
Update failed for me and auto-rolled back to 21.  I'll try again later and
update this post.

nachoparker, 2021-11-17:
the previewsgenerator issue should be fixed now on `devel`

ovpc, 2021-11-17:
Yep version is 99.99.99 ?

nachoparker, 2021-11-17:
> Yep version is 99.99.99 ?

Correct, we don't ever want updates from the app store for this forked app

vcbranco, 2021-11-17:
Runing ncp-update devel again, the Preview generator is working

closed 2021-11-18 by nachoparker