nextcloud/nextcloudpi #1306: unattendes-upgrades wrongfully skipping updates

tbruckmaier, 2021-07-31:
Hi, I have noticed that `unattended-upgrades` misses updates again. Same symptoms as in #1233. It seems like the package origins have changed back from `Raspbian` to `Raspberry Pi Foundation` (as it was before #1233)

When changing the label in `/etc/apt/apt.conf.d/20ncp-upgrades` to `o="Raspberry Pi Foundation",n=buster;`, updates work again.

Before changing:
```sh
$ cat /etc/apt/apt.conf.d/20ncp-upgrades
APT::Periodic::Update-Package-Lists "1";
APT::Periodic::Unattended-Upgrade "1";
APT::Periodic::MaxAge "14";
APT::Periodic::AutocleanInterval "7";
Unattended-Upgrade::Automatic-Reboot "true";
Unattended-Upgrade::Automatic-Reboot-Time "04:00";
Unattended-Upgrade::Origins-Pattern {
o=Debian,n=buster;
o=deb.sury.org,n=buster;
o="Raspbian",n=buster;
}
Dpkg::Options {
   "--force-confdef";
   "--force-confold";
};

// Enable the update/upgrade script, disabled by Armbian in 02-armbian-periodic
APT::Periodic::Enable "1";
$ sudo unattended-upgrades --debug
Initial blacklist : 
Initial whitelist: 
Starting unattended upgrades script
Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspbian,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
Using (^linux-image-5\.10\.17\-v7l\+$|^linux-headers-5\.10\.17\-v7l\+$|^linux-image-extra-5\.10\.17\-v7l\+$|^linux-modules-5\.10\.17\-v7l\+$|^linux-modules-extra-5\.10\.17\-v7l\+$|^linux-signed-image-5\.10\.17\-v7l\+$|^linux-image-unsigned-5\.10\.17\-v7l\+$|^kfreebsd-image-5\.10\.17\-v7l\+$|^kfreebsd-headers-5\.10\.17\-v7l\+$|^gnumach-image-5\.10\.17\-v7l\+$|^.*-modules-5\.10\.17\-v7l\+$|^.*-kernel-5\.10\.17\-v7l\+$|^linux-backports-modules-.*-5\.10\.17\-v7l\+$|^linux-modules-.*-5\.10\.17\-v7l\+$|^linux-tools-5\.10\.17\-v7l\+$|^linux-cloud-tools-5\.10\.17\-v7l\+$|^linux-buildinfo-5\.10\.17\-v7l\+$|^linux-source-5\.10\.17\-v7l\+$) regexp to find running kernel packages
Checking: firmware-atheros ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg firmware-atheros not in allowed origin
sanity check failed
Checking: firmware-brcm80211 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg firmware-brcm80211 not in allowed origin
sanity check failed
Checking: firmware-libertas ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg firmware-libertas not in allowed origin
sanity check failed
Checking: firmware-misc-nonfree ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg firmware-misc-nonfree not in allowed origin
sanity check failed
Checking: firmware-realtek ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg firmware-realtek not in allowed origin
sanity check failed
Checking: libavcodec58 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libavcodec58=7:4.1.6-1~deb10u1
Checking: libavformat58 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libavformat58=7:4.1.6-1~deb10u1
Checking: libavutil56 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libavutil56=7:4.1.6-1~deb10u1
Checking: libgl1-mesa-dri ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libgl1-mesa-dri=18.3.6-2+deb10u1
Checking: libglapi-mesa ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libglapi-mesa=18.3.6-2+deb10u1
Checking: libglx-mesa0 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libglx-mesa0=18.3.6-2+deb10u1
Checking: libraspberrypi-bin ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: libraspberrypi-dev ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: libraspberrypi-doc ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: libraspberrypi0 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
Checking: libswresample3 ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: libswresample3=7:4.1.6-1~deb10u1
Checking: linux-libc-dev ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
pkg linux-libc-dev not in allowed origin
sanity check failed
Checking: mesa-va-drivers ([<Origin component:'main' archive:'testing' origin:'Raspberry Pi Foundation' label:'Raspberry Pi Foundation' site:'archive.raspberrypi.org' isTrusted:True>])
adjusting candidate version: mesa-va-drivers=18.3.6-2+deb10u1
[... 112 lines cut ...]
data in SD           yes
data filesystem      ext2/ext3
data disk usage      6.3G/59G
rootfs usage         6.3G/59G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.11.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      down
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               14days

```
</details>

nachoparker, 2021-08-01:
Many thanks for noticing. I changed back in `devel`.

Can you type `sudo ncp-update devel` and verify that it works for you after that so I can push the fix for everyone?

Thanks!

tbruckmaier, 2021-08-01:
Yes, it is working again, thank you very much for the quick fix!

tbruckmaier, 2021-08-17:
Sorry to bother you again, but the origin has changed back to `Raspbian` once again, and unattended-upgrades skips updates...
```sh
$ sudo unattended-upgrades --debug --dry-run
Initial blacklist : 
Initial whitelist: 
Starting unattended upgrades script
Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspberry Pi Foundation,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
Using (^linux-image-5\.10\.52\-v7l\+$|^linux-headers-5\.10\.52\-v7l\+$|^linux-image-extra-5\.10\.52\-v7l\+$|^linux-modules-5\.10\.52\-v7l\+$|^linux-modules-extra-5\.10\.52\-v7l\+$|^linux-signed-image-5\.10\.52\-v7l\+$|^linux-image-unsigned-5\.10\.52\-v7l\+$|^kfreebsd-image-5\.10\.52\-v7l\+$|^kfreebsd-headers-5\.10\.52\-v7l\+$|^gnumach-image-5\.10\.52\-v7l\+$|^.*-modules-5\.10\.52\-v7l\+$|^.*-kernel-5\.10\.52\-v7l\+$|^linux-backports-modules-.*-5\.10\.52\-v7l\+$|^linux-modules-.*-5\.10\.52\-v7l\+$|^linux-tools-5\.10\.52\-v7l\+$|^linux-cloud-tools-5\.10\.52\-v7l\+$|^linux-buildinfo-5\.10\.52\-v7l\+$|^linux-source-5\.10\.52\-v7l\+$) regexp to find running kernel packages
Checking: bluez ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
adjusting candidate version: bluez=5.50-1.2~deb10u1+rpt2
Checking: libpam-systemd ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: libsystemd0 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: libudev1 ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: syncthing ([<Origin component:'stable' archive:'syncthing' origin:'Syncthing' label:'Syncthing' site:'apt.syncthing.net' isTrusted:True>])
Checking: systemd ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: systemd-sysv ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: udev ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
pkgs that look like they should be upgraded: 
Fetched 0 B in 0s (0 B/s)                                                                                                                                                                                      
fetch.run() result: 0
dpkg is configured not to cause conffile prompts
No packages found that can be upgraded unattended and no pending auto-removals
```

I have searched a bit, but could not figure out why or when that change happened. So maybe we can put both origins into `/etc/apt/apt.conf.d/20ncp-upgrades` just to be safe? The following works for example:
```sh
$ cat /etc/apt/apt.conf.d/20ncp-upgrades 
APT::Periodic::Update-Package-Lists "1";
APT::Periodic::Unattended-Upgrade "1";
APT::Periodic::MaxAge "14";
APT::Periodic::AutocleanInterval "7";
Unattended-Upgrade::Automatic-Reboot "true";
Unattended-Upgrade::Automatic-Reboot-Time "04:00";
Unattended-Upgrade::Origins-Pattern {
o=Debian,n=buster;
o=deb.sury.org,n=buster;
o="Raspberry Pi Foundation",n=buster;
o="Raspbian",n=buster;
}
Dpkg::Options {
   "--force-confdef";
   "--force-confold";
};

// Enable the update/upgrade script, disabled by Armbian in 02-armbian-periodic
APT::Periodic::Enable "1";

$ sudo unattended-upgrades --debug --dry-run
Initial blacklist : 
Initial whitelist: 
Starting unattended upgrades script
Allowed origins are: o=Debian,n=buster, o=deb.sury.org,n=buster, o=Raspberry Pi Foundation,n=buster, o=Raspbian,n=buster, origin=Debian,codename=buster,label=Debian, origin=Debian,codename=buster,label=Debian-Security
Using (^linux-image-[0-9]+\.[0-9\.]+-.*|^linux-headers-[0-9]+\.[0-9\.]+-.*|^linux-image-extra-[0-9]+\.[0-9\.]+-.*|^linux-modules-[0-9]+\.[0-9\.]+-.*|^linux-modules-extra-[0-9]+\.[0-9\.]+-.*|^linux-signed-image-[0-9]+\.[0-9\.]+-.*|^linux-image-unsigned-[0-9]+\.[0-9\.]+-.*|^kfreebsd-image-[0-9]+\.[0-9\.]+-.*|^kfreebsd-headers-[0-9]+\.[0-9\.]+-.*|^gnumach-image-[0-9]+\.[0-9\.]+-.*|^.*-modules-[0-9]+\.[0-9\.]+-.*|^.*-kernel-[0-9]+\.[0-9\.]+-.*|^linux-backports-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-modules-.*-[0-9]+\.[0-9\.]+-.*|^linux-tools-[0-9]+\.[0-9\.]+-.*|^linux-cloud-tools-[0-9]+\.[0-9\.]+-.*|^linux-buildinfo-[0-9]+\.[0-9\.]+-.*|^linux-source-[0-9]+\.[0-9\.]+-.*) regexp to find kernel packages
[... cut ...]
Checking: systemd ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: systemd-sysv ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
Checking: udev ([<Origin component:'main' archive:'stable' origin:'Raspbian' label:'Raspbian' site:'raspbian.raspberrypi.org' isTrusted:True>])
pkgs that look like they should be upgraded: bluez
libpam-systemd
libsystemd0
libudev1
systemd
systemd-sysv
udev
...
```

nachoparker, 2021-08-17:
good idea. Done

closed 2021-08-17 by nachoparker