nextcloud/nextcloudpi #1311: Request: Show CPU temperature in System Info

PennyWafel, 2021-08-05:
**Description**  

I would love to see the temperature of the Raspberry Pi displayed on the System Info screen of the NCP admin panel.  

As you know, RPI4's tend to run a bit hot. For those like me who like to overclock them, it's useful and nice to keep an eye on how the temperature is doing. One could of course run commands through ssh, but it would be nice to see it displayed on the System Info screen. It seems like a suitable place and even if it's just a static snapshot of the temperature it would still give some indication if further investigation is warranted.

Thanks for the great work on NCP, by the way! I feel empowered running my own cloud and wouldn't have been able to without NCP.

theCalcaholic, 2021-08-06:
While it would be possible to include that information in the web panel, this would only not be very useful, if it's not showing either realtime data or past data (i.e. a graph or something like that).
In my opinion, both of these options aren't very easy to implement with the current architecture.

But maybe the [new metrics app](https://github.com/theCalcaholic/ncp-monitoring-dashboard) fits your use case better. It provides (among many other things) metrics for CPU usage, temperature, etc.

PennyWafel, 2021-08-08:
Wow, that metrics app looks more amazing than anything I would dare to ask for! 
Looking forward to it!

theCalcaholic, 2021-08-08:
@PennyWafel You can already try it by running `ncp-update devel`. Or you can wait until it arrives in master :)

closed 2021-08-08 by theCalcaholic