nextcloud/nextcloudpi #1312: apache did install and start as requested

tomAufGithub, 2021-08-07:
I am new to nextcloud.
I tried to use 
wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
sudo bash install.sh

to install it on my raspberry4.

in the install there was a hint that apache did not start.

==> systemctl status apache2.service 
shows:
 apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2021-08-07 11:39:28 CEST; 6min ago
     Docs: https://httpd.apache.org/docs/2.4/
  Process: 5273 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Aug 07 11:39:28 nextcloudpi systemd[1]: Starting The Apache HTTP Server...
Aug 07 11:39:28 nextcloudpi apachectl[5273]: (98)Address already in use: AH00072: make_sock: could not bind to address
Aug 07 11:39:28 nextcloudpi apachectl[5273]: (98)Address already in use: AH00072: make_sock: could not bind to address
Aug 07 11:39:28 nextcloudpi apachectl[5273]: no listening sockets available, shutting down
Aug 07 11:39:28 nextcloudpi apachectl[5273]: AH00015: Unable to open logs
Aug 07 11:39:28 nextcloudpi apachectl[5273]: Action 'start' failed.
Aug 07 11:39:28 nextcloudpi apachectl[5273]: The Apache error log may have more information.
Aug 07 11:39:28 nextcloudpi systemd[1]: apache2.service: Control process exited, code=exited, status=1/FAILURE
Aug 07 11:39:28 nextcloudpi systemd[1]: apache2.service: Failed with result 'exit-code'.
Aug 07 11:39:28 nextcloudpi systemd[1]: Failed to start The Apache HTTP Server.

what should I do?

theCalcaholic, 2021-08-07:
Looks like you already have some process running that's occupying port 80 or 443.
The installer is meant to be used on a fresh debian system, so if you have modified it, there might be errors

x0rld, 2021-08-07:
if you do ` ss -ta |grep http`
you have something display ?
th `t` show the tcp connection and the `a` show all include the listen

tomAufGithub, 2021-08-07:
Thank you for your hints. I forgot that I have running Open Media Vault and this seems to run on the standard http port..
Is there a way to still use this nextcloud install process ?
Can I maybe modify the script a bit with your help?

"wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
sudo bash install.sh"

closed 2021-08-07 by nachoparker