nextcloud/nextcloudpi #1316: build-docker.sh armhf fails

mad73923, 2021-08-10:
I tried to build the latest docker image for armhf but it fails.
Using the master (v1.37.2) and devel (v1.38.0) versions gives me the same results.

I'm calling
```
build-docker.sh armhf
```

```
[...]
#17 684.5 System config value mail_smtpmode set to string sendmail
#17 684.8 Cannot load Zend OPcache - it was already loaded
#17 685.5 System config value mail_smtpauthtype set to string LOGIN
#17 685.8 Cannot load Zend OPcache - it was already loaded
#17 686.6 System config value mail_from_address set to string admin
#17 686.9 Cannot load Zend OPcache - it was already loaded
#17 687.6 System config value mail_domain set to string ownyourbits.com
#17 688.3 Cannot load Zend OPcache - it was already loaded
#17 689.1 nextcloudpi 0.0.1 enabled
#17 689.4 Cannot load Zend OPcache - it was already loaded
#17 695.1 calendar 2.3.1 installed
#17 695.2 calendar enabled
#17 695.4 Cannot load Zend OPcache - it was already loaded
#17 698.2 contacts 4.0.1 installed
#17 698.3 contacts enabled
#17 698.6 Cannot load Zend OPcache - it was already loaded
#17 703.3 notes 4.1.1 installed
#17 703.4 notes enabled
#17 703.6 Cannot load Zend OPcache - it was already loaded
#17 707.1 tasks 0.14.1 installed
#17 707.1 tasks enabled
#17 707.4 Cannot load Zend OPcache - it was already loaded
#17 711.0 Error: App "News" cannot be installed because the following dependencies are not fulfilled: 64bit or higher PHP required.
#17 ERROR: executor failed running [/bin/bash -c set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; apt-get install --no-install-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /usr/local/etc/ncp-config.d/nc-init-copy.cfg /usr/local/etc/ncp-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/config.php; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto.cfg; cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/ncp-config.d/nc-init.cfg; rm /.ncp-image; source /usr/local/etc/library.sh; find /usr/local/bin/ncp -name '*.sh' | while read l; do cleanup_script $l; done; apt-get autoremove -y; apt-get clean; find /var/lib/apt/lists -type f | xargs rm; find /var/log -type f -exec rm {} \; ; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*; rm -f /var/log/alternatives.log /var/log/apt/*; rm /var/cache/debconf/*-old; echo "${ncp_ver}" > /usr/local/etc/ncp-version]: exit code: 1
------
 > [stage-0 11/12] RUN --mount=type=cache,target=/var/cache/apt --mount=type=cache,target=/var/lib/apt set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; apt-get install --no-install-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /usr/local/etc/ncp-config.d/nc-init-copy.cfg /usr/local/etc/ncp-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/config.php; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto.cfg; cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/ncp-config.d/nc-init.cfg; rm /.ncp-image; source /usr/local/etc/library.sh; find /usr/local/bin/ncp -name '*.sh' | while read l; do cleanup_script $l; done; apt-get autoremove -y; apt-get clean; find /var/lib/apt/lists -type f | xargs rm; find /var/log -type f -exec rm {} ; ; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*; rm -f /var/log/alternatives.log /var/log/apt/*; rm /var/cache/debconf/*-old; echo "v1.38.0" > /usr/local/etc/ncp-version:
------
executor failed running [/bin/bash -c set -e; rm -f ncp-web/{wizard.cfg,ncp-web.cfg}; touch /.ncp-image; touch /.docker-image; apt-get update; apt-get install --no-install-recommends -y wget ca-certificates; source /usr/local/etc/library.sh; cd /tmp/ncp-build/; install_app ncp.sh; mv /usr/local/etc/ncp-config.d/nc-init-copy.cfg /usr/local/etc/ncp-config.d/nc-init.cfg; run_app_unsafe bin/ncp/CONFIG/nc-init.sh; sed -i 's|data-ro|data|' /data-ro/nextcloud/config/config.php; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup.cfg; sed -i 's|/media/USBdrive|/data/backups|' /usr/local/etc/ncp-config.d/nc-backup-auto.cfg; cd /; rm -r /tmp/ncp-build; rm /usr/local/etc/ncp-config.d/nc-init.cfg; rm /.ncp-image; source /usr/local/etc/library.sh; find /usr/local/bin/ncp -name '*.sh' | while read l; do cleanup_script $l; done; apt-get autoremove -y; apt-get clean; find /var/lib/apt/lists -type f | xargs rm; find /var/log -type f -exec rm {} \; ; rm -rf /usr/share/man/*; rm -rf /usr/share/doc/*; rm -f /var/log/alternatives.log /var/log/apt/*; rm /var/cache/debconf/*-old; echo "${ncp_ver}" > /usr/local/etc/ncp-version]: exit code: 1

```

nachoparker, 2021-08-11:
https://github.com/nextcloud/server/issues/27797

nachoparker, 2021-08-11:
Sadly, we'll have to remove news from 32-bit builds :/

nachoparker, 2021-08-17:
done

closed 2021-08-17 by nachoparker