nextcloud/nextcloudpi #1317: Cannot run nc-update

lokisilverblade, 2021-08-11:
Hello,

i cannot perform nc-update it ends with error:

"/usr/local/bin/ncp-update: line 23: ./update.sh: File or dir not found"

Installed is NCP 1.37.2

Thank you.

ovpc, 2021-08-11:
Please go to https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33 and use the ncp tag for support.

closed 2021-08-11 by ovpc

lokisilverblade, 2021-08-16:
Hey, after there is a new Version i cannot perform Update. The Script means "/usr/local/bin/ncp-update: line 23: ./update.sh: File or dir not found" - Is there any chance to fix the Script?

nachoparker, 2021-08-16:
let me look at that

nachoparker, 2021-08-16:
that is not happening to me

nachoparker, 2021-08-16:
are you doing `sudo ncp-update`? or how are you doing it?

lokisilverblade, 2021-08-16:
Yes i do sudo ncp-update or update via sudo ncp-config doesent work.

nachoparker, 2021-08-16:
try this `sudo rm /.ncp-image` and then retry

lokisilverblade, 2021-08-17:
Thank you it works :-)