nextcloud/nextcloudpi #1321: Debian 11 Bullseye

sunjam, 2021-08-15:
Debian 11 Bullseye was released to stable.
https://www.debian.org/News/2021/20210814

BJKle, 2021-09-03:
> Debian 11 Bullseye was released to stable.
> https://www.debian.org/News/2021/20210814

Interesting news. Are there any plans to support this in the near future?
A timeline would be great.

nachoparker, 2021-09-03:
this is volunteer work, so no timelines. If you want it faster, you can always help out

svenb1234, 2021-09-07:
Bullseye means there will be a 64bit raspberry pi OS and thus the news app as well as the trashbin (https://github.com/nextcloud/server/issues/13160) will work again?

sunjam, 2021-09-07:
> as well as the trashbin (nextcloud/server#13160

Raspbian 64 is being developed.
support for 32bit News app was dropped.
not sure on trashbin

nachoparker, 2021-09-08:
we have been using 64bit Rpi images for a long time FYI

sunjam, 2021-09-08:
Raspberry Pi OS, formerly known as Raspbian, has not yet received a
Bullseye release in stable.  Just fyi, it hasn't come out yet.  Will take
more time since it is a spinoff of Debian. :)

https://raspberrytips.com/update-raspberry-pi-latest-version/

nachoparker, 2021-09-08:
yes, we'll have to wait until it's available for rpi to generate the new release

svenb1234, 2021-09-10:
> we have been using 64bit Rpi images for a long time FYI

That's true. But if you run ncp as docker container, the host would be 32 bit and the image 64bit. Thus you have to use the 32bit container and thus the trashbin does not work anymore.

nachoparker, 2021-09-12:
@svenb1234 not sure what you mean. Our non docker images for RPi are 64 bits. If you want to run docker, you have both architectures available.

A 32 bit host can never run 64 bit containers (but the other way around works)

nachoparker, 2021-09-21:
https://github.com/nextcloud/nextcloudpi/pull/1342

nachoparker, 2021-10-05:
bullseye works fine, only one package missing, and ofc rpi/armbian support

https://github.com/nextcloud/nextcloudpi/pull/1342

sunjam, 2021-11-09:
Raspberry Pi OS for Bullseye is being released.  However, only armhf packages are listed so far.

https://downloads.raspberrypi.org/?C=M;O=D

UN-pi, 2021-12-15:
I am running bullseye von a Pi4 and ncp 1.46.7 and noticed a few things.

1. /usr/local/bin/ncp-diag still contains:
echo "MariaDB service|$( pgrep -c mysqld &>/dev/null && echo up || echo down )"

2. /usr/local/etc/ncp-recommended.cfg
 is thinking about php7.3 and buster

3./etc/apt/apt.conf.d/20ncp-upgrades still contains:
 buster and not bullseye

4.ncp-dist-upgrade is removing a already higher php-Version and downgrade to php7.4
   Maybe the php-version should be checked first like:
  
  get the realy installed php-version:

   sphp=`php -v | grep PHP -m 1 | awk '{print $2}'`
   curphp=$(echo ${sphp:0:3})
   ...
   curphp shows: '7.3', '7.4', '8.0', '8.1'     or whatever

Hope that helps.

jvonau, 2021-12-15:
Here is how I installed on a RPi4 running on 2021-10-30-raspios-bullseye-arm64-lite RaspiOS.
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh > install.sh`
edit install.sh master -> devel near the top.
`sudo bash install.sh`
Ran to completion. No 'news' issue found as in #1316
>news 17.0.1 installed
news enabled
news already enabled
```
pi@raspberrypi:~ $ php -v | grep PHP -m 1 | awk '{print $2}'
7.4.25
pi@raspberrypi:~ $ ls /etc/php
7.4
```
Found this:
```
--2021-12-15 06:40:06--  https://github.com/nachoparker/noip-DDNS/archive/master/latest.tar.gz
Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/nachoparker/noip-DDNS/tar.gz/master [following]
--2021-12-15 06:40:07--  https://codeload.github.com/nachoparker/noip-DDNS/tar.gz/master
Resolving codeload.github.com (codeload.github.com)... 140.82.114.10
Connecting to codeload.github.com (codeload.github.com)|140.82.114.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘STDOUT’

-                                 [  <=>                                            ] 136.62K   407KB/s    in 0.3s    

2021-12-15 06:40:08 (407 KB/s) - written to stdout [139904]

/tmp/nextcloudpi.epnFCL/nextcloudpi
gcc -Wall -g -Dlinux -DPREFIX=\"/usr/local\" noip2.c -o noip2 
noip2.c: In function ‘dynamic_update’:
noip2.c:1595:6: warning: variable ‘i’ set but not used [-Wunused-but-set-variable]
 1595 |  int i, x, is_group, retval, response;
      |      ^
noip2.c: In function ‘domains’:
noip2.c:1826:13: warning: variable ‘x’ set but not used [-Wunused-but-set-variable]
 1826 |         int x;
      |             ^
noip2.c: In function ‘hosts’:
noip2.c:1838:20: warning: variable ‘y’ set but not used [-Wunused-but-set-variable]
 1838 |         int     x, y, z;
      |                    ^
noip2.c: In function ‘autoconf’:
noip2.c:2538:19: warning: ‘%s’ directive writing up to 255 bytes into a region of size 247 [-Wformat-overflow=]
 2538 |  sprintf(line, "%s%s%s%s", USTRNG, login, PWDSTRNG, password);
      |                   ^~               ~~~~~
noip2.c:2538:2: note: ‘sprintf’ output between 16 and 526 bytes into a destination of size 256
 2538 |  sprintf(line, "%s%s%s%s", USTRNG, login, PWDSTRNG, password);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/nextcloudpi.epnFCL/nextcloudpi
Installing samba
```
>pi@raspberrypi:~ $ ls /usr/local/etc/ncp-recommended.cfg
ls: cannot access '/usr/local/etc/ncp-recommended.cfg': No such file or directory

>pi@raspberrypi:~ $ grep bullseye /etc/apt/apt.conf.d/20ncp-upgrades 
o=Debian,n=bullseye-security;
o=deb.sury.org,n=bullseye-security;
o="Raspberry Pi Foundation",n=bullseye-security;
o="Raspbian",n=bullseye-security;

@UN-pi think you need to be on the devel branch. 
On to configuration...

UN-pi, 2021-12-15:
Thanks for the idea.
The pi4 runs with 64bit, bullseye and php8, as it is to be expected that the 7.4 version will be replaced by 8.0 in the future. It all works very well.
The suggested install script ends with the message that the NC database already exists. Which is not surprising, since the NC is already installed. 

Perhaps it would be helpful to note somewhere or to configure in general whether you are using the branch master or devel. Maybe in ncp.cfg?
Then ncp-update would also recognize it automatically.

Alternatively, I have the file
/etc/cron.daily/ncp-autoupdate changed to:

 if / usr / local / bin / ncp-test-updates; then
   / usr / local / bin / ncp-update **devel** || exit 1
   notify_admin "NextCloudPi" "NextCloudPi was updated to $ (cat / usr / local / etc / ncp-version)"

So devel is used by default.

jvonau, 2021-12-15:
> Thanks for the idea. The pi4 runs with 64bit, bullseye and php8, as it is to be expected that the 7.4 version will be replaced by 8.0 in the future. It all works very well. The suggested install [...]

Full disclosure, I've been part of small team that started putting OC on offline hardware in 2015 over at IIAB where NC is now running under both RaspiOS and Ubuntu being trouble free so I was rather shocked when I found out about the 'news' issue on 32bit. We too are also looking at the future with 8.1 with ongoing testing on Ubuntu [22.04](https://github.com/iiab/iiab/issues/3047) while going to 21.10 with php8.0 was pretty straight forward once you understand how things were shuffled around within the bump of php.

Maxxxel, 2021-12-28:
@jvonau this doesn't work for me. Installation runs through, but ncp is not useable. The webui doesn't work and ncp-config is not responding at all.

sirdrakeistda, 2022-01-03:
@jvonau have you managed to solve this error message:
```
An unhandled exception has been thrown:
OC\HintException: [0]: Memcache \OC\Memcache\Redis not available for local cache (Is the matching PHP module installed and enabled?)
```
I got this error message on a self updated system and on a new setup with debian bullseye. php-apcu is installed and I tried to enable this in config settings, but it was not succesfull.

jvonau, 2022-01-03:
I have not circled back around with no plans to try given the forced-pushed changes on live devel branch with the claim the changes are already applied.

Wimmelbach, 2022-01-20:
@nachoparker: On September 12th you mentioned:
"Our non docker images for RPi are 64 bits."
But 'I am wondering if they are already based on bullseye or still on buster.

nachoparker, 2022-03-01:
creating images, will be published tomorrow for testers

closed 2022-03-01 by nachoparker