nextcloud/nextcloudpi #1322: Disabling nc-httpsonly does not have any effect, still can't connect via http

briankendall, 2021-08-23:
Quite simply, I want to disable https-only because I will be using a forward proxy to handle all https to my Nextcloud instance (and other webservers). However after disabling nc-httpsonly it still redirects me every time I try to connect using just http.

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.38.5
NextCloudPi image    NextCloudPi_docker_07-20-19
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      55G/472G
rootfs usage         55G/472G
swapfile             /dev/sda2
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    19.0.13.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               2:47

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "11": "73.219.140.5",
            "1": "172.17.0.2",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "6": "nextcloud-bri.1337.cx",
            "3": "nextcloud.briankendall.net",
            "2": "nextcloud.briankendall.net",
            "4": "nextcloud.briankendall.net",
            "12": "nextcloud.briankendall.net"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "19.0.13.1",
        "overwrite.cli.url": "https:\/\/nextcloud.briankendall.net\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
[... 96 lines cut ...]

```

{"reqId":"YSLuq9Q9fIdHtkkXiDyqDwAAAEY","level":1,"time":"2021-08-23T00:41:15+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLuqz4QhJcy5ncb-xz6gQAAABE","level":1,"time":"2021-08-23T00:41:15+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLuvNQ9fIdHtkkXiDyqEQAAAFU","level":1,"time":"2021-08-23T00:41:32+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLuvD4QhJcy5ncb-xz6iAAAERE","level":1,"time":"2021-08-23T00:41:32+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLuzNQ9fIdHtkkXiDyqGAAAAEI","level":1,"time":"2021-08-23T00:41:48+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLuzNQ9fIdHtkkXiDyqGQAASwU","level":1,"time":"2021-08-23T00:41:48+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu3dQ9fIdHtkkXiDyqGwAAAFE","level":1,"time":"2021-08-23T00:42:05+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu3T4QhJcy5ncb-xz6kQAABRE","level":1,"time":"2021-08-23T00:42:05+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu7NQ9fIdHtkkXiDyqIwAAAFA","level":1,"time":"2021-08-23T00:42:20+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu7D4QhJcy5ncb-xz6lgAAARE","level":1,"time":"2021-08-23T00:42:20+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu@9Q9fIdHtkkXiDyqJAAAAFI","level":1,"time":"2021-08-23T00:42:35+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLu@9Q9fIdHtkkXiDyqJQAAVwU","level":1,"time":"2021-08-23T00:42:35+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvCtQ9fIdHtkkXiDyqLQAAAFQ","level":1,"time":"2021-08-23T00:42:50+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvCj4QhJcy5ncb-xz6ngAADhE","level":1,"time":"2021-08-23T00:42:50+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvGdQ9fIdHtkkXiDyqLgAAAEs","level":1,"time":"2021-08-23T00:43:05+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvGT4QhJcy5ncb-xz6ogAABBE","level":1,"time":"2021-08-23T00:43:05+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvKdQ9fIdHtkkXiDyqOQAAAEM","level":1,"time":"2021-08-23T00:43:21+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvKT4QhJcy5ncb-xz6qQAAFxE","level":1,"time":"2021-08-23T00:43:21+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":" [... cut ...] access using \"mediaserver.local:80\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}
{"reqId":"YSLvONQ9fIdHtkkXiDyqOwAARgU","level":1,"time":"2021-08-23T00:43:36+00:00","remoteAddr":"10.0.2.2","user":"--","app":"core","method":"GET","url":"/","message":"Trusted domain error. \"10.0.2.2\" tried to access using \"mediaserver.local:443\" as host.","userAgent":"AutoMounter/91 CFNetwork/978.0.7 Darwin/18.7.0 (x86_64)","version":"19.0.13.1"}

```

nachoparker, 2021-08-23:
Hi,

Thanks for reporting. I just tested this and it works as intended.

You can check `/etc/apache2/sites-available/000-default.conf`. It should read `RewriteEngine Off` if httpsonly is disabled

closed 2021-08-23 by nachoparker

briankendall, 2021-08-23:
@nachoparker I looked into it a bit more, and I think I found the source of the problem, at least on my installation:

I needed to edit: `/data/nextcloud/config/config.php` and change line 52 from: `'overwriteprotocol' => 'https',` to `'overwriteprotocol' => 'http',`.

So long as the value was `https` it would always redirect from http to https, regardless of how `nc-httpsonly` was set. Once I made the change then `nc-httpsonly` works as expected.