nextcloud/nextcloudpi #1326: Upgrade NCP 1.39.0/NC 21.0.4 fail

vcbranco, 2021-09-05:
The upgrade does not done.

During upgrade I have the following errors just after the line "System config value trusted_proxies => 12 set to string ::1"

✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
🗴 push server can't connect to the Nextcloud server
  error sending request for url (https://10.8.6.100/index.php/apps/notify_push/test/cookie): error trying to connect: invalid dnsname: error trying to connect: invalid dnsname: invalid dnsname
Abort

Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20210905_1630827743-20.0.2.2.tar.gz...
{"reqId":"g2uhEBMHYLKFdYG1047b","level":2,"time":"2021-09-05T07:45:50+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}
{"reqId":"g2uhEBMHYLKFdYG1047b","level":2,"time":"2021-09-05T07:45:50+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"21.0.4.1"}
check free space...
extracting backup file /var/www/nextcloud-bkp_20210905_1630827743-20.0.2.2.tar.gz...
restore files...
restore database...
No datadir found in backup
Maintenance mode disabled
Starting scan for user 1 out of 1 (ncp)
+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 2       | 0     | 00:00:00     |
+---------+-------+--------------+
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value logfile set to string /var/www/nextcloud/data/nextcloud.log
System config value trusted_domains => 11 set to string 176.78.117.244
System config value trusted_domains => 1 set to string 10.8.6.100
Scanning AppData for files

+---------+-------+--------------+
| Folders | Files | Elapsed time |
+---------+-------+--------------+
| 5       | 4     | 00:00:00     |
+---------+-------+--------------+
Cleanup...
Rollback successful. Nothing was updated
Done. Press any key...

nachoparker, 2021-09-05:
many thanks for testing. Can you share the output of `sudo ncp-report` please?

nachoparker, 2021-09-05:
also, is this your main instance, or is this a test instance you have around?

nachoparker, 2021-09-05:
I made a special branch to debug your issue. That url https://10.8.6.100/index.php/apps/notify_push/test/cookie seems plain wrong.

Do you mind doing `sudo ncp-update nc-21-vcbranco` and trying again `sudo ncp-update-nc 21.0.4`, please?

We should get more info from the output this time around

vcbranco, 2021-09-05:
This is a test instance (RPI 4 4GB) with the NextCloudPi_RPi_11-27-20 image fully updated.

I'll do all the tests and place the output here

ovpc, 2021-09-05:
Like wise here on a VM (Proxmox), got rollback:
terminal output at https://pastebin.com/59CHhBJ7

WeiChihChern, 2021-09-05:
Host: M1 Mac Mini
Docker image: ownyourbits/nextcloudpi-arm64:latest
Note: Fresh docker run without any data

1. Updated apps and nextcloud from 18.0.3 to 20.0.12 first.
2. `ncp-update nc-21-vcbranco`. Successful, NCP to version 1.39.0.
3.  `ncp-update-nc 21.0.4`. Error message shown as below:

> root@c5c3b7795ebe:/# ncp-update-nc 21.0.4
Current   Nextcloud version 20.0.12.1
Available Nextcloud version 21.0.4
Download Nextcloud 21.0.4...
Back up current instance...
check free space...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode already enabled
backup database...
backup files...
tar: Nextcloud or one of the apps require upgrade - only a limited number of commands are available\nYou may use your browser or the occ upgrade command to do the upgrade\n/data/nextcloud: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
error generating backup
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode disabled
Error backing up
Clean up...

4. `ncp-report` as shown below:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.0
NextCloudPi image    NextCloudPi_docker_01-19-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      21G/24G
rootfs usage         21G/24G
swapfile             /swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    20.0.12.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

[... 114 lines cut ...]
```

{"reqId":"jrQgQkRWZTJDAgpNqAqf","level":2,"time":"2021-09-05T19:32:02+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"jrQgQkRWZTJDAgpNqAqf","level":2,"time":"2021-09-05T19:32:02+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"hTuojUtdLFlQ9VsOawHd","level":2,"time":"2021-09-05T19:32:59+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"hTuojUtdLFlQ9VsOawHd","level":2,"time":"2021-09-05T19:32:59+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"dlI2WmhuaFwjbkxOV1zM","level":2,"time":"2021-09-05T19:33:22+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"dlI2WmhuaFwjbkxOV1zM","level":2,"time":"2021-09-05T19:33:22+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"XO2MQhSgkZB4W7ksenSA","level":2,"time":"2021-09-05T19:33:22+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"XO2MQhSgkZB4W7ksenSA","level":2,"time":"2021-09-05T19:33:22+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"1lr0SxgyhahzO5hapqXI","level":2,"time":"2021-09-05T19:35:01+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"1lr0SxgyhahzO5hapqXI","level":2,"time":"2021-09-05T19:35:01+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"y8iTNXDHrT4PUUts0nkg","level":2,"time":"2021-09-05T19:35:43+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
[... cut ...]
{"reqId":"0COkiijVbO2OEpQdfRT4","level":2,"time":"2021-09-05T19:37:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0COkiijVbO2OEpQdfRT4","level":2,"time":"2021-09-05T19:37:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}

```
</details>

WeiChihChern, 2021-09-05:
2nd try.

1. Updated apps and NCP to 1.38.6.
2. Update NC from 18.0.3 to 20.0.12.
3. Update NC from 20.0.12 to 21.0.4 Failed. Run the update again, then showing 'Nextcloud data directory not found'.
4. `ncp-report`:
<details>
<summary>NextCloudPi diagnostics</summary>

```

{"reqId":"6jNcTch5ypYv98ojjUT7","level":2,"time":"2021-09-05T19:54:03+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"6jNcTch5ypYv98ojjUT7","level":2,"time":"2021-09-05T19:54:03+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
NextCloudPi version  v1.38.6
NextCloudPi image    NextCloudPi_docker_01-19-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data (doesn't exist)
rootfs usage         21G/24G
swapfile             /swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    20.0.12.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               4days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "172.17.0.3",
            "6": "192.168.1.65"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "18.0.3.0",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
[... 71 lines cut ...]
2021-09-05 19:46:54 0 [Note] mysqld: ready for connections.
Version: '10.3.22-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-09-05 19:46:56 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2021-09-05 19:46:56 0 [Note] Event Scheduler: Purging the queue. 0 events
2021-09-05 19:46:56 0 [Note] InnoDB: FTS optimize thread exiting.
2021-09-05 19:46:56 0 [Note] InnoDB: Starting shutdown...
2021-09-05 19:46:56 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2021-09-05 19:46:56 0 [Note] InnoDB: Buffer pool(s) dump completed at 210905 19:46:56
2021-09-05 19:46:58 0 [Note] InnoDB: Shutdown completed; log sequence number 4296968; transaction id 2780
2021-09-05 19:46:58 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-09-05 19:46:58 0 [Note] mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"arEevnkhRuhqtm5xPDbr","level":2,"time":"2021-09-05T19:54:08+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"arEevnkhRuhqtm5xPDbr","level":2,"time":"2021-09-05T19:54:08+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}

```
</details>

WeiChihChern, 2021-09-05:
Would it work if building a new image with NC21?

vcbranco, 2021-09-05:
ncp-report after the failed upgrade (ncp-update devel)

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.0
NextCloudPi image    NextCloudPi_11-27-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      3,0G/29G
rootfs usage         3,0G/29G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    20.0.12.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               10min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "176.78.117.244",
            "1": "10.8.6.100"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "20.0.12.1",
        "overwrite.cli.url": "https:\/\/10.8.6.100\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
[... 88 lines cut ...]
```

{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Reset generated avatar flag","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Keep legacy encryption enabled","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Check encryption key format","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove old dashboard app config data","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"20.0.12.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--" [... cut ...]
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:33+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"21.0.4.1"}
{"reqId":"sMYHaKycFJSkgt3245VR","level":1,"time":"2021-09-05T19:10:33+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"21.0.4.1"}

```
</details>

vcbranco, 2021-09-05:
Test with the instance ncp-update nc-21-vcbranco in the same image but clean (the card was cloned)
Error:

Update successful
Turned off maintenance mode
Resetting log level
Check indices of the share table.
Check indices of the filecache table.
Adding additional size index to the filecache table, this can take some time...
Filecache table updated successfully.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Check columns of the comments table.
Done.
Check primary keys.
Done.
All tables already up to date!
++ ncc status
++ grep version:
++ awk '{ print $3 }'
+ NCVER=21.0.4.1
+ is_more_recent_than 21.0.0 21.0.4.1
+ local version_A=21.0.0
+ local version_B=21.0.4.1
++ cut -d. -f1
+ local major_a=21
++ cut -d. -f2
+ local minor_a=0
++ cut -d. -f3
+ local patch_a=0
++ cut -d. -f1
+ local major_b=21
++ cut -d. -f2
+ local minor_b=0
++ cut -d. -f3
+ local patch_b=4
+ '[' 21 -gt 21 ']'
+ '[' 21 -eq 21 ']'
+ '[' 0 -gt 0 ']'
+ '[' 21 -eq 21 ']'
+ '[' 0 -eq 0 ']'
+ '[' 4 -ge 0 ']'
+ return 1
+ is_app_enabled notify_push
+ local app=notify_push
+ ncc app:list
+ sed '0,/Disabled/!d'
+ grep -q notify_push
+ ncc app:install notify_push
notify_push 0.2.2 installed
notify_push enabled
+ ncc app:enable notify_push
notify_push already enabled
+ bash /usr/local/etc/ncp-templates/nextcloud.conf.sh
+ a2enmod proxy proxy_http proxy_wstunnel
Module proxy already enabled
Considering dependency proxy for proxy_http:
Module proxy already enabled
Module proxy_http already enabled
Considering dependency proxy for proxy_wstunnel:
Module proxy already enabled
Enabling module proxy_wstunnel.
To activate the new configuration, you need to run:
  systemctl restart apache2
[... 243 lines cut ...]
```

{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Reset generated avatar flag","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Keep legacy encryption enabled","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Check encryption key format","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove old dashboard app config data","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"20.0.12.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:11+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
[... cut ...]
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:22+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"21.0.4.1"}
{"reqId":"xMjN7dikzYpVFsXMJ60M","level":1,"time":"2021-09-05T20:11:22+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"21.0.4.1"}

```
</details>

vcbranco, 2021-09-05:
The second error ncp-report is at the end.
It seems to me it's the same issue

WeiChihChern, 2021-09-06:
part 3.

Telegram pinned update instruction result:
> Running nc-update-nextcloud
> Current   Nextcloud version 18.0.3.0
> Available Nextcloud version 21.0.4
> Download Nextcloud 21.0.4...
> Back up current instance...
> check free space...
> Maintenance mode enabled
> backup database...
> backup files...
> backup /var/www//nextcloud-bkp_20210906_1630894298.tar.gz generated
> Maintenance mode disabled
> Install Nextcloud 21.0.4...
> Restore data...
> Fix permissions...
> Upgrade...
> Nextcloud or one of the apps require upgrade - only a limited number of commands are available
> You may use your browser or the occ upgrade command to do the upgrade
> Setting log level to debug
> Turned on maintenance mode
> Exception: Updates between multiple major versions and downgrades are unsupported.
> Update failed
> Maintenance mode is kept active
> Resetting log level
> Abort
> Clean up...
> Rolling back to backup /var/www/nextcloud-bkp_20210906_1630894298-18.0.3.0.tar.gz...
> {"reqId":"y5pHOagKPrj1O9DNjmEO","level":2,"time":"2021-09-06T02:11:56+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
> {"reqId":"y5pHOagKPrj1O9DNjmEO","level":2,"time":"2021-09-06T02:11:56+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
> Can only restore from ext/btrfs filesystems
> Rollback failed! Data left at /var/www/ncp-data.na7qHH
> Done. Press any key...

'ncp-report' logs:
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.0
NextCloudPi image    NextCloudPi_docker_01-19-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      21G/24G
rootfs usage         21G/24G
swapfile             /swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5days

[... 104 lines cut ...]
2021-09-06  2:09:45 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2021-09-06  2:09:45 0 [Note] Event Scheduler: Purging the queue. 0 events
2021-09-06  2:09:45 0 [Note] InnoDB: FTS optimize thread exiting.
2021-09-06  2:09:45 0 [Note] InnoDB: Starting shutdown...
2021-09-06  2:09:45 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2021-09-06  2:09:45 0 [Note] InnoDB: Buffer pool(s) dump completed at 210906  2:09:45
2021-09-06  2:09:47 0 [Note] InnoDB: Shutdown completed; log sequence number 4296968; transaction id 2780
2021-09-06  2:09:47 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2021-09-06  2:09:47 0 [Note] mysqld: Shutdown complete

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"QaPAdR4IssDJdJoZeLeo","level":2,"time":"2021-09-06T02:13:51+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"QaPAdR4IssDJdJoZeLeo","level":2,"time":"2021-09-06T02:13:51+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"tvPDA53vfkwfOVpynR0X","level":2,"time":"2021-09-06T02:13:56+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"tvPDA53vfkwfOVpynR0X","level":2,"time":"2021-09-06T02:13:56+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}

```
</details>

nachoparker, 2021-09-07:
thanks all. Please bear with me, as upgrading to this new HPB is really proving to be a pain given all the moving pieces. Also I don't think it's super well polished.

I'll go with your issues one by one.

@vcbranco

A couple questions: Do you have `nc-static-IP` activated by any chance? do you access by IP or do you have a domain name to access?

I couldn't reproduce your issue, but I think we are dealing with this https://github.com/rustls/hyper-rustls/issues/84

In any case, please run again `ncp-update nc-21-vcbranco` and then `ncp-update-nc 21.0.4`, see if my fixes work for your case.

---

@WeiChihChern 

> Would it work if just build a new image with NC21? Just be so excited with NC21.

Yes, you can checkout `devel` and then `./build-docker.sh arm64`, but then I still need to fix your issue before I can push this to everyone. A new set of images will come after this is good to go.

I'll try to replicate your upgrade path next

vcbranco, 2021-09-07:
Yes, nc-static-ip is activated.
This test intance is accessed by IP.

I going to start testing and post the results.

vcbranco, 2021-09-07:
No luck
Same problem

I'm going to setup a test DNS to the upgrade

WeiChihChern, 2021-09-07:
@nachoparker Thanks for the feedback. I've build an image from the devel branch. My personal NCP is running with the older image with an external hard drive. Can I simply stop the old image, and launch the new image and mount the external drive?

Or further setup is required to switch from one image to another with data all in an external drive? Thanks.

nachoparker, 2021-09-08:
@WeiChihChern that's not how it works, if you want to retain your users, data and so on you need to upgrade, no way around that.

> Updated apps and nextcloud from 18.0.3 to 20.0.12 first.

How did you do this? through NC updater or through NCP tools? it seems like you were in a half updated state. I updated from 18 to 20 and then did `ncp-update nc-21-vcbranco` and then upgraded to 21 all using NCP tools and it worked fine for me.

@vcbranco thanks for your testing. You never shared the output from  your second attempt with static IP so I don't know how it went. You just said it didn't work. Then you changed your setup and you shared the output, so we are mixing things here and I cannot trace whether my fix worked or needs adjustments.

> I'm going to setup a test DNS to the upgrade

what does this mean? on your last output I see https://pi.emcontacto.eu in overwrite.cli.url, but I don't know where it comes from. notify_push seems to have a problem with it. Are you behind a proxy?

WeiChihChern, 2021-09-08:
@nachoparker 
I see. I wonder what if one encountered power shortage to his device and have to relaunch the docker container again? Does that mean the users, data, etc cannot be retained? Does it mean docker users are encouraged to `docker commit` their image regularly? Thanks.

I updated apps and nextcloud using ncp-config. I would try it again with your successful run again, see how it goes.

nachoparker, 2021-09-08:
@ovpc if your proxmox VM is disposable (only if), can you also `ncp-update nc-21-vcbranco`, then edit `/usr/local/bin/ncp-update-nc` and remove the line that says `set-nc-domain "${nc_domain}"`, then

```
ncp-update-nc 21.0.4 # should finish ok this time, but notify_push won't be setup yet
ncc notify_push:log debug
source /usr/local/etc/library.sh
set -x
set-nc-domain localhost
```

You error is super weird and super unhelpful and I can't find what library it comes from :/

Thanks

nachoparker, 2021-09-08:
@WeiChihChern not encouraged to `docker commit`, your NC instance, database and data need to be in sync and they live in a volume so that's orthogonal to what image you are using, and that means that they are retained always, but they are independent on the underlying docker image, because they need to be in sync. You cannot upgrade NC without migrating the database, apps and so on.

WeiChihChern, 2021-09-08:
@nachoparker 
Tried again, but having the error of:

> Running nc-update-nextcloud
Current   Nextcloud version 20.0.4.0
Available Nextcloud version 21.0.4
Download Nextcloud 21.0.4...
Back up current instance...
check free space...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode enabled
backup database...
backup files...
tar: Nextcloud or one of the apps require upgrade - only a limited number of commands are available\nYou may use your browser or the occ upgrade command to do the upgrade\n/data/nextcloud: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
error generating backup
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode disabled
Error backing up
Clean up...
Done. Press any key...

`ncp-report`:
<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.0
NextCloudPi image    NextCloudPi_docker_01-19-21
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      overlayfs
data disk usage      23G/48G
rootfs usage         23G/48G
swapfile             /swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    20.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               7min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
{
[... 111 lines cut ...]
```

{"reqId":"IXmnKavTbH4tUIuwdorz","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"IXmnKavTbH4tUIuwdorz","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"u7j9zdrfCZbmXkxOVav8","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"u7j9zdrfCZbmXkxOVav8","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"hDJpU3GLC47rAUeu6S0D","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"hDJpU3GLC47rAUeu6S0D","level":2,"time":"2021-09-08T02:16:12+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0yCzwe1PNcqnHzBl3A6D","level":2,"time":"2021-09-08T02:16:20+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0yCzwe1PNcqnHzBl3A6D","level":2,"time":"2021-09-08T02:16:20+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"uGXvRUoI12SW88I17uKy","level":2,"time":"2021-09-08T02:16:20+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"uGXvRUoI12SW88I17uKy","level":2,"time":"2021-09-08T02:16:20+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"Vu8bf3Hg9sNvadt7krNc","level":2,"time":"2021-09-08T02:16:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"Vu8bf3Hg9sNvadt7krNc","level":2,"time":"2021-09-08T02:16:21+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextclou [... cut ...]
{"reqId":"fLCsGzLmIdLpf4Rhyluz","level":2,"time":"2021-09-08T02:17:04+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"fLCsGzLmIdLpf4Rhyluz","level":2,"time":"2021-09-08T02:17:04+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}

```
</details>

WeiChihChern, 2021-09-08:
@nachoparker 
> and that means that they are retained always

Sorry that I might not use the right words. I understand that the database would retain. In my case, in the external storage that `-v ` into the container.

What I am not sure what to do is that, as mentioned if something like power shortage resulting in relaunching a container is required, what is the correct ways to migrate my database, user, etc into a newly launch container?

It could be:
1. Using the provided docker image to start a fresh setup.
2. Upgrade to NC 20, then have everything setup and users upload their files
3. Encounter situations that needs to restart a container
4. The provided docker image has NC 18, which does not match my database
What should I do in this case? Thanks.

nachoparker, 2021-09-08:
@WeiChihChern it seems like your backup step is failing? can you just run a backup from `ncp-config`, see how that goes?

> The provided docker image has NC 18, which does not match my database
What should I do in this case? Thanks.

this is what you are not understanding I think, the image does not contain NC18. NC18 lives in a volume, along with the data and the database (and the apps). You can restart the container all you want, or you can even change container versions, as long as you include the same volume you will be in the same "NC instance" (but maybe with a different version of NCP tools)

WeiChihChern, 2021-09-08:
@nachoparker 
backup log:
> Running nc-backup
check free space...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode enabled
backup database...
backup files...
tar: Nextcloud or one of the apps require upgrade - only a limited number of commands are available\nYou may use your browser or the occ upgrade command to do the upgrade\n/data/nextcloud: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
error generating backup
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode disabled
Done. Press any key...

Ahh, now I understand it. Thanks for the explanation.

ovpc, 2021-09-08:
> @ovpc if your proxmox VM is disposable (only if), can you also `ncp-update nc-21-vcbranco`, then edit `/usr/local/bin/ncp-update-nc` and remove the line that says `set-nc-domain "${nc_domain}"`, then
> 
> ```
> ncp-update-nc 21.0.4 # should finish ok this time, but notify_push won't be setup yet
> ncc notify_push:log debug
> source /usr/local/etc/library.sh
> set -x
> set-nc-domain localhost
> journalctl -u notify_push # hopefully there's some debug info here
> ```
> 
> You error is super weird and super unhelpful and I can't find what library it comes from :/
> 
> Thanks

@nachoparker sorry this VM is not disposible. I will create a fresh one to test this, later on this week.

ovpc, 2021-09-08:
@nachoparker 
Could not reproduce the error on a new test VM it updated fine to 21.0.4 (see  https://pastebin.com/8jWdiYqh)
Got the old imagick error in NC admin settings tho

Security & setup warnings

   This instance is missing some recommended PHP modules. For improved performance and better compatibility it is highly recommended to install them.

    imagick

nachoparker, 2021-09-08:
@ovpc can you just try again on your original VM? maybe it was a one time glitch :thinking: 

@WeiChihChern can you add `set -x` to `/usr/local/bin/ncp-backup`  as a second line and try to run a backup again?

WeiChihChern, 2021-09-08:
@nachoparker 
Is this what you want?

> root@bbb3f069dfbb:/# /usr/local/bin/ncp-backup \ -x
Cannot load Zend OPcache - it was already loaded
check free space...
Cannot load Zend OPcache - it was already loaded
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode enabled
backup database...
backup files...
tar: Nextcloud or one of the apps require upgrade - only a limited number of commands are available\nYou may use your browser or the occ upgrade command to do the upgrade\n/data/nextcloud: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
error generating backup
Cannot load Zend OPcache - it was already loaded
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Maintenance mode disabled

nachoparker, 2021-09-08:
thanks @WeiChihChern let's just run these commands

```
sudo sed -i  '1aset -x' /usr/local/bin/ncp-backup
sudo ncp-config
# start a backup from the menu
```

thanks

nachoparker, 2021-09-08:
@ovpc oh gosh, that makes me a bit uneasy :/

but glad it worked

ovpc, 2021-09-08:
It is complaining about
> Module php-imagick in this instance has no SVG support. For better compatibility it is recommended to install it.

nachoparker, 2021-09-08:
can you try this?

https://help.nextcloud.com/t/how-to-enable-svg-for-php-imagick/108646/5

ovpc, 2021-09-08:
had already found same fix at https://techoverflow.net/2021/08/17/how-to-fix-docker-nextcloud-module-php-imagick-in-this-instance-has-no-svg-support-for-better-compatibility-it-is-recommended-to-install-it/

> sudo apt install libmagickcore-6.q16-6-extra

fixed it

WeiChihChern, 2021-09-08:
> thanks @WeiChihChern let's just run these commands
> 
> ```
> sudo sed -i  '1aset -x' /usr/local/bin/ncp-backup
> sudo ncp-config
> # start a backup from the menu
> ```
> 
> thanks

What I got:

> Running nc-update-nextcloud
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Cannot load Zend OPcache - it was already loaded
> Current   Nextcloud version 20.0.4.0
> Available Nextcloud version 21.0.4
> Download Nextcloud 21.0.4...
> Back up current instance...
> + set -eE
> + destdir=/var/www/
> + includedata=no
> + compress=yes
> + backuplimit=0
> ++ date +%Y%m%d_%s
> + destfile=/var/www//nextcloud-bkp_20210908_1631113525.tar
> ++ date +%Y%m%d
> + dbbackup=nextcloud-sqlbkp_20210908.bak
> + occ='sudo -u www-data php /var/www/nextcloud/occ'
> + [[ -f /.docker-image ]]
> + basedir=/data
> + [[ yes == \y\e\s ]]
> + destfile=/var/www//nextcloud-bkp_20210908_1631113525.tar.gz
> ++ sudo -u www-data php /var/www/nextcloud/occ config:system:get datadirectory
> Cannot load Zend OPcache - it was already loaded
> + datadir='Nextcloud or one of the apps require upgrade - only a limited number of commands are available
> You may use your browser or the occ upgrade command to do the upgrade
> /data/nextcloud/data'
> + trap cleanup EXIT
> + trap fail INT TERM HUP ERR
> + echo 'check free space...'
> check free space...
> + mkdir -p /var/www/
> ++ du -sb /data/nextcloud
> ++ awk '{ print $1 }'
> + nsize=408125092
> + [[ -d /data/nextcloud/data ]]
> ++ du -sb /data/nextcloud/data
> ++ awk '{ print $1 }'
> + dsize=61061140
> + margin=524288000
> + [[ no == \y\e\s ]]
> + size=871351952
> ++ df -B1 /var/www/
> ++ tail -1
> ++ awk '{ print $4 }'
> + free=23708033024
> + '[' 871351952 -ge 23708033024 ']'
> + [[ 0 != 0 ]]
> + sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on
> Cannot load Zend OPcache - it was already loaded
> Nextcloud or one of the apps require upgrade - only a limited number of commands are available
> You may use your browser or the occ upgrade command to do the upgrade
> Maintenance mode enabled
> + cd /data
> + echo 'backup database...'
> backup database...
> + mysqldump -u root --single-transaction nextcloud
> + echo 'backup files...'
[... 150 lines cut ...]
```

{"reqId":"BAF92sotBmzUpIPuSOYr","level":2,"time":"2021-09-08T15:05:16+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"BAF92sotBmzUpIPuSOYr","level":2,"time":"2021-09-08T15:05:16+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"VHtSTyaf5Q3UGVYhQx5j","level":2,"time":"2021-09-08T15:05:17+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"VHtSTyaf5Q3UGVYhQx5j","level":2,"time":"2021-09-08T15:05:17+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"JGjQ6TY6TyX2xW0RfzYn","level":2,"time":"2021-09-08T15:05:17+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"JGjQ6TY6TyX2xW0RfzYn","level":2,"time":"2021-09-08T15:05:17+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"PA1MCvHfJUqfHPtsVy78","level":2,"time":"2021-09-08T15:05:25+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"PA1MCvHfJUqfHPtsVy78","level":2,"time":"2021-09-08T15:05:25+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"Qjwn1wLKLgBA81NGYieg","level":2,"time":"2021-09-08T15:05:25+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"Qjwn1wLKLgBA81NGYieg","level":2,"time":"2021-09-08T15:05:25+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"jHF1P66pk7nWOvd6lasK","level":2,"time":"2021-09-08T15:05:25+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
[... cut ...]
{"reqId":"fE3nUl16Zb5wbQFZsRKt","level":2,"time":"2021-09-08T15:06:45+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
{"reqId":"fE3nUl16Zb5wbQFZsRKt","level":2,"time":"2021-09-08T15:06:45+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}

```
</details>

nachoparker, 2021-09-08:
@ovpc  sweet, thanks, we are back at the imagick discussion though. Not sure if we should install it for security reasons

nachoparker, 2021-09-08:
@WeiChihChern 

ugh

```
++ sudo -u www-data php /var/www/nextcloud/occ config:system:get datadirectory
Cannot load Zend OPcache - it was already loaded
datadir='Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
```

I see what's going on, we should handle this more gracefully. What is the output from 

```
# ncc config:system:get datadirectory 2> /dev/null
# echo $?
```

WeiChihChern, 2021-09-08:
@nachoparker 

> root@bbb3f069dfbb:/# ncc config:system:get datadirectory 2> /dev/null
> Nextcloud or one of the apps require upgrade - only a limited number of commands are available
> You may use your browser or the occ upgrade command to do the upgrade
> /data/nextcloud/data
> root@bbb3f069dfbb:/# echo $?
> 0

nachoparker, 2021-09-08:
@WeiChihChern 

```
ncc status
```
?

WeiChihChern, 2021-09-08:
@nachoparker 

> root@bbb3f069dfbb:/# ncc status
> Cannot load Zend OPcache - it was already loaded
> Nextcloud or one of the apps require upgrade - only a limited number of commands are available
> You may use your browser or the occ upgrade command to do the upgrade
>   - installed: true
>   - version: 20.0.4.0
>   - versionstring: 20.0.4
>   - edition:
> root@bbb3f069dfbb:/#

nachoparker, 2021-09-08:
hmm I wonder why you keep getting that message, if you log in to Nextcloud, does everything work on the web interface?

WeiChihChern, 2021-09-08:
Opening the web interface
[image]

Clicked upgrade:
[image]

nachoparker, 2021-09-08:
yes, your instance was in a weird half-upgraded state to start with

vcbranco, 2021-09-08:
> @WeiChihChern that's not how it works, if you want to retain your users, data and so on you need to upgrade, no way around that.
> 
> > Updated apps and nextcloud from 18.0.3 to 20.0.12 first.
> 
> How did you do this? through NC updater or through NCP tools? it seems like you were in a half updated state. I updated from 18 to 20 and then did `ncp-update nc-21-vcbranco` and then upgraded to 21 all using NCP tools and it worked fine for me.
> 
> @vcbranco thanks for your testing. You never shared the output from your second attempt with static IP so I don't know how it went. You just said it didn't work. Then you changed your setup and you shared the output, so we are mixing things here and I cannot trace whether my fix worked or needs adjustments.
> 
> > I'm going to setup a test DNS to the upgrade
> 
> what does this mean? on your last output I see https://pi.emcontacto.eu in overwrite.cli.url, but I don't know where it comes from. notify_push seems to have a problem with it. Are you behind a proxy? have you messed with the 'forwarded_for_headers' option in your config.php by any chance?

This is the test with the fixes nc-static-ip and letsencrypt configured

https://github.com/nextcloud/nextcloudpi/issues/1326#issuecomment-914511671

Something you said and I did not understant/noticed?

vcbranco, 2021-09-08:
I'm not behind a proxy and did not make any change in the config files.
I installed the image on the pi, updated with ncp-update and ncp-update-nextcloud and sudo apt update

WeiChihChern, 2021-09-08:
> yes, your instance was in a weird half-upgraded state to start with. I would go to the NC forums if you want to try to fix that

It's fine, this is only a test container. Looking forward to the final release.

nachoparker, 2021-09-08:
@vcbranco thanks for clarifying, by the way, you don't have to and should not do ` sudo apt update/upgrade`, let NCP do it automatically for you to make sure you don't override any configuration.

Can you run `nc-static-IP` again and then share the output of `ncp-update-nextcloud 21.0.4` so I can see why my fix failed?

Thanks

vcbranco, 2021-09-08:
Ok.
I'll do that

nachoparker, 2021-09-08:
I added another tweak for your "letsencrypt attempt". In order to test this one, please run

```
ncp-update nc-21-vcbranco # download the fix
ncp-config # run letsencrypt again
ncp-update-nc 21.0.4
```

thanks

vcbranco, 2021-09-08:
This test
Failed

-Burned image in the card
-nc-update
-nc-update-nextcloud
-nc-static-ip
-ncp-update nc-21-vcbranco
-letsencrypt
-ncp-update-nextcloud

Update successful
Turned off maintenance mode
Resetting log level
Check indices of the share table.
Check indices of the filecache table.
Adding additional size index to the filecache table, this can take some time...
Filecache table updated successfully.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Check columns of the comments table.
Done.
Check primary keys.
Done.
All tables already up to date!
++ ncc status
++ grep version:
++ awk '{ print $3 }'
+ NCVER=21.0.4.1
+ is_more_recent_than 21.0.0 21.0.4.1
+ local version_A=21.0.0
+ local version_B=21.0.4.1
++ cut -d. -f1
+ local major_a=21
++ cut -d. -f2
+ local minor_a=0
++ cut -d. -f3
+ local patch_a=0
++ cut -d. -f1
+ local major_b=21
++ cut -d. -f2
+ local minor_b=0
++ cut -d. -f3
+ local patch_b=4
+ '[' 21 -gt 21 ']'
+ '[' 21 -eq 21 ']'
+ '[' 0 -gt 0 ']'
+ '[' 21 -eq 21 ']'
+ '[' 0 -eq 0 ']'
+ '[' 4 -ge 0 ']'
+ return 1
+ is_app_enabled notify_push
+ local app=notify_push
+ ncc app:list
+ sed '0,/Disabled/!d'
+ grep -q notify_push
+ ncc app:install notify_push
notify_push 0.2.2 installed
notify_push enabled
+ ncc app:enable notify_push
notify_push already enabled
+ bash /usr/local/etc/ncp-templates/nextcloud.conf.sh
+ a2enmod proxy proxy_http proxy_wstunnel
perl: warning: Setting locale failed.
[... 290 lines cut ...]
```

{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Reset generated avatar flag","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Keep legacy encryption enabled","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Check encryption key format","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Remove old dashboard app config data","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Add job to cleanup the bruteforce entries","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Queue a one-time job to check for user uploaded certificates","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Deprecated event type for \\OC\\Repair::step: Symfony\\Component\\EventDispatcher\\GenericEvent is used","userAgent":"--","version":"20.0.12.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:15+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Repair::step: Repair step: Repair DAV shares","userAgent":"--","version":"20.0.12.1"}
[... cut ...]
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:23+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::maintenanceDisabled: Turned off maintenance mode","userAgent":"--","version":"21.0.4.1"}
{"reqId":"npYhs74dgeHDXhNKdt5N","level":1,"time":"2021-09-08T20:43:23+00:00","remoteAddr":"","user":"--","app":"updater","method":"","url":"--","message":"\\OC\\Updater::resetLogLevel: Reset log level to Warning(2)","userAgent":"--","version":"21.0.4.1"}

```
</details>

vcbranco, 2021-09-08:
I think that my router, despite being configured normally, is working as a proxy.
I will add the router address to the trusted addresses and test again

nachoparker, 2021-09-08:
hi @vcbranco, let's see

you don't need to burn a new image every time :)

First, make sure that since this is a newly flashed image, you activate nextcloud (you know, the initial activation where you get your random passwords). Let's start from zero one more time.

- Only one time:

Burn the image
Activate
After that, run `ncp-update nc-21-vcbranco`

- Now we have

## test 1
-letsencrypt
-ncp-update-nc 21.0.4

## test 2
- nc-static-IP
- ncp-update-nc 21.0.4

We can re-run tests 1 and 2 however many times we want without needing a fresh burn + activation every time.

Thanks!

nachoparker, 2021-09-08:
now, looking at your last report.. I have no idea where the IP 10.8.6.1' comes from

vcbranco, 2021-09-08:
10.8.6.1 is the router's IP

nachoparker, 2021-09-08:
oh! so somehow seems to be acting as a proxy, sending XFF headers :thinking:

vcbranco, 2021-09-08:
I noticed that only in the last test.
Wired

vcbranco, 2021-09-08:
I bought this router two weeks ago.

nachoparker, 2021-09-08:
or maybe your domain provider is sending them, something of that nature

nachoparker, 2021-09-08:
I don't think we can automate this... you would have to run `ncc config:system:set trusted_proxies 14 --value=10.8.6.1` manually, and then the upgrade should succeed.

Another thing to try, I think that in that case the letsencrypt test would work

vcbranco, 2021-09-08:
The ISP router is configured as a bridge, I can test there if necessary
I will run ncc config:system:set trusted_proxies 14 --value=10.8.6.1 before the tests

nachoparker, 2021-09-08:
are you using DDNS settings in the router maybe for your domain?

[35 more replies]

closed 2021-09-15 by nachoparker