nextcloud/nextcloudpi #1329: After upgrading to bullseye, wrong update notice

BJKle, 2021-09-09:
I've upgrade my os to Debian 11 (**bullseye**).
When I log-the following update notice is displayed:

   Update to **buster** available. Type 'sudo ncp-dist-upgrade' to upgrade

***Solution?***
The update check should check the debian version 

**System information**

NextCloudPi version  v1.39.0
distribution         Raspbian GNU/Linux 11 \n \l
automount            yes
USB devices          sda 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      55G/932G
rootfs usage         2.7G/30G
swapfile             none
dbdir                /media/myCloudDrive/ncdatabase
Nextcloud check      ok
Nextcloud version    22.1.1.2

ovpc, 2021-09-09:
Thanks for reporting, but buster is still only NCP supported version.  It will resolve itself when NCP moves to Bullseye.
No other issues 😉?

sunjam, 2021-09-10:
#1321 related.  Thank you for reporting!  I see you are on 22.*

BJKle, 2021-09-12:
I just wanted to help. Nevermind.

closed 2021-09-12 by BJKle