nextcloud/nextcloudpi #1332: error with trusted_domain after upgrade

AMstuff, 2021-09-15:
After the upgrade to 1.39.1 (from 1.38) my config contains an error on every reboot.

```    3 => 'An unhandled exception has been thrown:
RedisException: read error on connection in 
Stack trace:
#0 
#1 
#2 
#3 
#4 
#5 
#6 
#7 
#8 
#9 
#10 
#11 
#12 
#13 
#14 
#15 
#16 
#17 
#18 
#19 
#20 
#21 
#22 
#23 {main}',
```

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_12-01-18
distribution         Armbian 21.08.2 Buster \l 
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      649G/1,9T
rootfs usage         11G/15G
swapfile             /var/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    21.0.4.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
HPB service          down
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            enx001e063261be
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               6min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "1": "192.168.1.2",
            "21": "***REMOVED SENSITIVE VALUE***",
            "11": "***REMOVED SENSITIVE VALUE***",
            "20": "***REMOVED SENSITIVE VALUE***",
            "22": "***REMOVED SENSITIVE VALUE***",
            "3": "An unhandled exception has been thrown:\nRedisException: read error on connection in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 \n#13 \n#14 \n#15 \n#16 \n#17 \n#18 \n#19 \n#20 \n#21 \n#22 \n#23 {main}"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "21.0.4.1",
        "overwrite.cli.url": "https:\/\/An unhandled exception has been thrown:\nRedisException: read error on connection in \nStack trace:\n#0 \n#1 \n#2 \n#3 \n#4 \n#5 \n#6 \n#7 \n#8 \n#9 \n#10 \n#11 \n#12 \n#13 \n#14 \n#15 \n#16 \n#17 \n#18 \n#19 \n#20 \n#21 \n#22 \n#23 {main}\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "loglevel": "2",
        "log_type": "file",
        "maintenance": false,
        "tempdirectory": "\/media\/USBdrive\/ncdata\/tmp",
        "logfile": "\/media\/USBdrive\/ncdata\/nextcloud.log",
        "theme": "",
        "jpeg_quality": "60",
        "app_install_overwrite": [
            "nextcloudpi",
            "admin_notifications"
        ],
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "ssl",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "465",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "htaccess.RewriteBase": "\/",
        "updater.release.channel": "stable",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Sep 15 15:31:33.043842 2021] [ssl:warn] [pid 1581:tid 3069496704] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Sep 15 15:31:33.108778 2021] [ssl:warn] [pid 1923:tid 3069496704] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Sep 15 15:31:33.123779 2021] [mpm_event:notice] [pid 1923:tid 3069496704] AH00489: Apache/2.4.38 (Debian) OpenSSL/1.1.1d configured -- resuming normal operations
[Wed Sep 15 15:31:33.124037 2021] [core:notice] [pid 1923:tid 3069496704] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

2021-09-15 15:31:54 0 [Note] InnoDB: Uses event mutexes
2021-09-15 15:31:54 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-09-15 15:31:54 0 [Note] InnoDB: Number of pools: 1
2021-09-15 15:31:54 0 [Note] InnoDB: Using generic crc32 instructions
2021-09-15 15:31:54 0 [Note] InnoDB: Initializing buffer pool, total size = 512M, instances = 1, chunk size = 128M
2021-09-15 15:31:54 0 [Note] InnoDB: Completed initialization of buffer pool
2021-09-15 15:31:54 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-09-15 15:31:55 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-09-15 15:31:55 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-09-15 15:31:55 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-09-15 15:31:56 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-09-15 15:31:56 0 [Note] InnoDB: 10.3.29 started; log sequence number 22196869844; transaction id 67445041
2021-09-15 15:31:56 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-09-15 15:31:56 0 [Note] InnoDB: Loading buffer pool(s) from /media/Nextcloud/ncdatabase/ib_buffer_pool
2021-09-15 15:31:57 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-09-15 15:31:57 0 [Note] Reading of all Master_info entries succeeded
2021-09-15 15:31:57 0 [Note] Added new Master_info '' to hash table
2021-09-15 15:31:57 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.29-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-09-15 15:32:10 0 [Note] InnoDB: Buffer pool(s) load completed at 210915 15:32:10

```
</details>

nachoparker, 2021-09-15:
Thanks for the feedback. It seems like your redis connection wasn't working during the time of the upgrade. Let's try this

```
source /usr/local/etc/library.sh
set -x
set-nc-domain <your_domain>
```

please share output, and after that, please share again output from `sudo ncp-report`

AMstuff, 2021-09-15:
Thanks, this time it worked. When I tried ealier, it would show the same error on sourcing library.sh at this line:
```
export TRUSTED_DOMAINS=(
  [ip]=1 [dnsmasq]=2 [nc_domain]=3 [nextcloudpi-local]=5 [docker_overwrite]=6
  [nextcloudpi]=7 [nextcloudpi-lan]=8 [public_ip]=11 [letsencrypt_1]=12
  [letsencrypt_2]=13 [trusted_domain_1]=20 [trusted_domain_1]=21 [trusted_domain_1]=22)
```
I manually cleaned up my config.php afterwards, that may have fixed the issue.

Anyway, this is the output now (Replaced my ip with 999.999.999.999). Seems like this issue is resolved. I still have issues with notify_push, though.
<details>
<summary>output of set-nc-domain</summary>
```
+ set-nc-domain my-public-domain.net
+ local domain=my-public-domain.net
++ sed 's|http.\?://||;s|\(/.*\)||'
+ domain=my-public-domain.net
+ [[ my-public-domain.net == '' ]]
+ is_an_ip my-public-domain.net
+ local ip_or_domain=my-public-domain.net
+ grep -oPq '\d{1,3}(.\d{1,3}){3}'
+ local proto
++ ncc config:system:get overwriteprotocol
+ proto=https
+ [[ https == '' ]]
+ local url=https://my-public-domain.net
+ [[ '' == \-\-\n\o\-\t\r\u\s\t\e\d\-\d\o\m\a\i\n ]]
+ ncc config:system:set trusted_domains 3 --value=my-public-domain.net
System config value trusted_domains => 3 set to string my-public-domain.net
+ ncc config:system:set overwrite.cli.url --value=https://my-public-domain.net/
System config value overwrite.cli.url set to string https://my-public-domain.net/
+ is_ncp_activated
+ a2query -s ncp-activation -q
+ is_app_enabled notify_push
+ local app=notify_push
+ ncc app:list
+ sed '0,/Disabled/!d'
+ grep -q notify_push
+ ncc config:system:set trusted_proxies 11 --value=127.0.0.1
System config value trusted_proxies => 11 set to string 127.0.0.1
+ ncc config:system:set trusted_proxies 12 --value=::1
System config value trusted_proxies => 12 set to string ::1
+ ncc config:system:set trusted_proxies 13 --value=my-public-domain.net
System config value trusted_proxies => 13 set to string my-public-domain.net
++ dig +short my-public-domain.net
+ ncc config:system:set trusted_proxies 14 '--value=dyndns.dnsprovider.net.
999.999.999.999'
System config value trusted_proxies => 14 set to string dyndns.dnsprovider.net.
999.999.999.999
+ sleep 5
+ ncc notify_push:setup https://my-public-domain.net/push
✓ redis is configured
🗴 push server is not receiving redis messages (received 853811047, got 0)

```
</details>

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_12-01-18
distribution         Armbian 21.08.2 Buster \l 
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
[... 132 lines cut ...]
2021-09-15 15:31:54 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2021-09-15 15:31:54 0 [Note] InnoDB: Number of pools: 1
2021-09-15 15:31:54 0 [Note] InnoDB: Using generic crc32 instructions
2021-09-15 15:31:54 0 [Note] InnoDB: Initializing buffer pool, total size = 512M, instances = 1, chunk size = 128M
2021-09-15 15:31:54 0 [Note] InnoDB: Completed initialization of buffer pool
2021-09-15 15:31:54 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2021-09-15 15:31:55 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2021-09-15 15:31:55 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2021-09-15 15:31:55 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2021-09-15 15:31:56 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2021-09-15 15:31:56 0 [Note] InnoDB: 10.3.29 started; log sequence number 22196869844; transaction id 67445041
2021-09-15 15:31:56 0 [Note] Plugin 'FEEDBACK' is disabled.
2021-09-15 15:31:56 0 [Note] InnoDB: Loading buffer pool(s) from /media/Nextcloud/ncdatabase/ib_buffer_pool
2021-09-15 15:31:57 0 [Note] Server socket created on IP: '127.0.0.1'.
2021-09-15 15:31:57 0 [Note] Reading of all Master_info entries succeeded
2021-09-15 15:31:57 0 [Note] Added new Master_info '' to hash table
2021-09-15 15:31:57 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.29-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2021-09-15 15:32:10 0 [Note] InnoDB: Buffer pool(s) load completed at 210915 15:32:10
2021-09-15 19:28:44 1080 [Warning] Aborted connection 1080 to db: 'nextcloud' user: 'ncadmin' host: 'localhost' (Got an error reading communication packets)

```
</details>

</details>

nachoparker, 2021-09-15:
thanks, please reboot and try again

AMstuff, 2021-09-15:
Same results after the reboot:

```bash
~$ ncc notify_push:setup https://mydomain.net/push
✓ redis is configured
🗴 push server is not receiving redis messages (received 309396003, got 0)

~$ ncc notify_push:self-test
🗴 no push server configured
```

Iolaum, 2021-09-17:
I was able to fix the problem with the above commands:

```
source /usr/local/etc/library.sh
set -x
set-nc-domain <your_domain>
```

it didn't work the first time. What I did the second time was to manually delete all entries in `trusted_domains`, `trusted_proxies` and `overwrite.cli.url` and then re-run the commands. After re-booting 3 times `config.php` remains at a sane condition! :+1:

Iolaum, 2021-09-17:
Regarding the requested outputs.

Here's the output of the commands when they fixed stuff:

```
$ source /usr/local/etc/library.sh
pi@nextcloud:~ $ set -x
pi@nextcloud:~ $ set-nc-domain $mysubdomain
+ set-nc-domain $mysubdomain
+ local domain=$mysubdomain
++ sed 's|http.\?://||;s|\(/.*\)||'
+ domain=$mysubdomain
+ [[ $mysubdomain == '' ]]
+ is_an_ip $mysubdomain
+ local ip_or_domain=$mysubdomain
+ grep --color=auto -oPq '\d{1,3}(.\d{1,3}){3}'
+ local proto
++ ncc config:system:get overwriteprotocol
+ proto=https
+ [[ https == '' ]]
+ local url=https://$mysubdomain
+ [[ '' == \-\-\n\o\-\t\r\u\s\t\e\d\-\d\o\m\a\i\n ]]
+ ncc config:system:set trusted_domains 3 --value=$mysubdomain
System config value trusted_domains => 3 set to string $mysubdomain
+ ncc config:system:set overwrite.cli.url --value=https://$mysubdomain/
System config value overwrite.cli.url set to string https://$mysubdomain/
+ is_ncp_activated
+ a2query -s ncp-activation -q
+ is_app_enabled notify_push
+ local app=notify_push
+ sed '0,/Disabled/!d'
+ ncc app:list
+ grep --color=auto -q notify_push
+ ncc config:system:set trusted_proxies 11 --value=127.0.0.1
System config value trusted_proxies => 11 set to string 127.0.0.1
+ ncc config:system:set trusted_proxies 12 --value=::1
System config value trusted_proxies => 12 set to string ::1
+ ncc config:system:set trusted_proxies 13 --value=$mysubdomain
System config value trusted_proxies => 13 set to string $mysubdomain
++ dig +short $mysubdomain
+ ncc config:system:set trusted_proxies 14 --value=$my_local_ip
System config value trusted_proxies => 14 set to string $my_local_ip
+ sleep 5
+ ncc notify_push:setup https://$mysubdomain/push
✓ redis is configured
✓ push server is receiving redis messages
✓ push server can load mount info from database
✓ push server can connect to the Nextcloud server
✓ push server is a trusted proxy
✓ push server is running the same version as the app
  configuration saved
```

Also here's the output of ncp-report:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.39.1
NextCloudPi image    NextCloudPi_03-09-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      299G/916G
rootfs usage         2.6G/15G
[... 143 lines cut ...]
```

{"reqId":"YUSOAtXNa1pDzMUYq5yrQwAAAAA","level":4,"time":"2021-09-17T12:46:26+00:00","remoteAddr":"192.168.1.211","user":"ncp","app":"oauth2","method":"PUT","url":"/index.php/apps/user_status/heartbeat","message":{"Exception":"RedisException","Message":"read error on connection","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/Memcache/Redis.php","line":55,"function":"get","class":"Redis","type":"->","args":["9028ef9023c2ceb9f76cd2b1ba004ff9/AutoloaderOCA\\OAuth2\\AppInfo\\Application"]},{"file":"/var/www/nextcloud/lib/autoloader.php","line":153,"function":"get","class":"OC\\Memcache\\Redis","type":"->","args":["OCA\\OAuth2\\AppInfo\\Application"]},{"function":"load","class":"OC\\Autoloader","type":"->","args":["OCA\\OAuth2\\AppInfo\\Application"]},{"function":"spl_autoload_call","args":["OCA\\OAuth2\\AppInfo\\Application"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Bootstrap/Coordinator.php","line":195,"function":"class_exists","args":["OCA\\OAuth2\\AppInfo\\Application"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":171,"function":"isBootable","class":"OC\\AppFramework\\Bootstrap\\Coordinator","type":"->","args":["oauth2"]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":138,"function":"loadApp","class":"OC_App","type":"::","args":["oauth2"]},{"file":"/var/www/nextcloud/lib/base.php","line":972,"function":"loadApps","class":"OC_App","type":"::","args":[["authentication"]]},{"file":"/var/www/nextcloud/index.php","line":37,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":55,"CustomMessage":"Error during app loading: read error on connection"},"userAgent":"Mozilla/5.0 (X11; Linux x86_64; rv:92.0) Gecko/20100101 Firefox/92.0","version":"21.0.4.1"}
{"reqId":"LJB8HnESr0N6Vwzx2xpm","level":4,"time":"2021-09-17T12:46:26+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":{"Exception":"RedisException","Message":"No such file or directory","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":92,"function":"connect","class":"Redis","type":"->","args":["/var/run/redis/redis.sock",0,0]},{"file":"/var/www/nextcloud/lib/private/RedisFactory.php","line":108,"function":"create","class":"OC\\RedisFactory","type":"->","args":[]},{"file":"/var/www/nextcloud/apps/notify_push/lib/AppInfo/Application.php","line":62,"function":"getInstance","class":"OC\\RedisFactory","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":155,"function":"OCA\\NotifyPush\\AppInfo\\{closure}","class":"OCA\\NotifyPush\\AppInfo\\Application","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":118,"function":"OC\\AppFramework\\Utility\\{closure}","class":"OC\\AppFramework\\Utility\\SimpleContainer","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleCon [... cut ...] mmands","class":"OC\\Console\\Application","type":"->","args":[{"__class__":"Symfony\\Component\\Console\\Input\\ArgvInput"},{"__class__":"Symfony\\Component\\Console\\Output\\ConsoleOutput"}]},{"file":"/var/www/nextcloud/occ","line":11,"args":["/var/www/nextcloud/console.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/private/Memcache/Redis.php","Line":55,"CustomMessage":"Error during app loading: read error on connection"},"userAgent":"--","version":"21.0.4.1"}

```
</details>

nachoparker, 2021-09-18:
@AMstuff thanks, did the junk make its way back to the config.php after reboot?

What I don't understand is how that got there in the first place, maybe it's some old bug. I am going to try to enforce that redis is up and running before we set up the domain, see if that's the cause.

Right now I want to know if this is repeatable. That would be very helpful.

nachoparker, 2021-09-19:
I pushed something that might fix this. It would be great to get confirmation

Iolaum, 2021-09-19:
I haven't kept around a config that would cause the error :o 
But I think that if one gets the error they will have to manually edit the file in order to remove those weird error lines.

P.S. In my case I think connectivity may have played a role (at least it appeared to be when I was adding trusted proxies because I am trying to move to a reverse proxy setup).

AMstuff, 2021-09-19:
@nachoparker The junk didn't make it back into my config yet. I have just restarted my machine twice (still with ncp v1.39.1) and wasn't able to reproduce the error. So since i fixed the config manually once, it hasn't happened again, regardless of your recent changes. 
I also haven't kept a copy of the broken config around unfortunately, but I definitely didn't have any connectivity issues like Iolaum.

nachoparker, 2021-09-19:
ok, I'll close assuming that at some point in the past, who knows how your config was corrupted. We'll reopen if we get any more clues.

nachoparker, 2021-09-19:
Thanks all

closed 2021-09-19 by nachoparker

Iolaum, 2021-09-22:
I am still getting these errors, in a config that (I think) shouldn't otherwise get them. I am guessing they may be related to connectivity issues, is redis trying to ping these addresses?

Also, I m wondering if there is a programatic way to remove the error message splashing in the config file - or is that more likely an upstream bug?

nachoparker, 2021-09-22:
@lolaum are you saying that even if you fix up the config manually you can reproduce this issue by rebooting? If that's the case this is a perfect opportunity to dig in and fix it.

Iolaum, 2021-09-22:
Not reliably though. Whenever I fixed it I reboot ~3 times to see if the fix is stable.
At this time the config hasn't bugged out yet. I 'll keep an eye on it though.

I also may have an idea to re-produce it (by adding a something.lan address in trusted proxies that wouldn't resolve and ping back because it doesn't point to a host). If that works, what should I do to dig in? Creating an ncp-report and take it from there?

nachoparker, 2021-09-22:
I added some checks yesterday to try to avoid this situation, but ideally I would be able to understand it. The problem is that redis is not ready I think by the time we try to set up the domain every reboot, so this junk ends up in the config.

Waiting for the service to be up doesn't seem enough, so I added some extra checks yesterday. I hope I see no more reports of this, otherwise we'll have to take another angle, but being able to reproduce is a must.

nachoparker, 2021-09-22:
I also added a thing to the reboot where it detects the junk in the config and reverts it, for users already hit by this

Likqez, 2025-08-07:
Hi guys, i had a similar error on nextcloud `31.02.1`.
Just wanted to bump this so you know this can still be an issue.

config.php looked like this:
```
 'trusted_domains' =>
  array (
    0 => '#0',
    1 => '#1',
    2 => '#10',
    3 => '#11',
    4 => '#12',
    5 => '#13',
    6 => '#14',
    7 => '#15',
    8 => '#16',
    9 => '#17',
    10 => '#2',
    11 => '#3',
    12 => '#4',
    13 => '#5',
    14 => '#6',
    15 => '#7',
    16 => '#8',
    17 => '#9',
    18 => '(172.16.1.2),',
    19 => '/var/www/html/3rdparty/doctrine/dbal/src/Connection.php(323):',
    20 => '/var/www/html/3rdparty/doctrine/dbal/src/Connection.php(416):',
    21 => '/var/www/html/3rdparty/doctrine/dbal/src/Connection.php(458):',
    22 => '/var/www/html/console.php(28):',
    23 => '/var/www/html/lib/base.php(1156):',
    24 => '/var/www/html/lib/base.php(675):',
    25 => '/var/www/html/lib/private/App/AppManager.php(136):',
    26 => '/var/www/html/lib/private/App/AppManager.php(157):',
    27 => '/var/www/html/lib/private/AppConfig.php(1226):',
    28 => '/var/www/html/lib/private/AppConfig.php(1366):',
    29 => '/var/www/html/lib/private/AppConfig.php(243):',
    30 => '/var/www/html/lib/private/AppFramework/Bootstrap/Coordinator.php(48):',
    31 => '/var/www/html/lib/private/DB/Connection.php(903):',
    32 => '/var/www/html/lib/private/DB/Connection.php:237',
    33 => '/var/www/html/lib/private/DB/ConnectionAdapter.php(235):',
    34 => '/var/www/html/lib/private/DB/QueryBuilder/QueryBuilder.php(96):',
    35 => '/var/www/html/lib/private/legacy/OC_App.php(188):',
    36 => '/var/www/html/occ(33):',
    37 => '127.0.0.1',
    38 => '192.168.178.105',
    39 => '2)',
    40 => '5432',
    41 => 'An',
    42 => 'DoctrineDBALConnection->detectDatabasePlatform()',
    43 => 'DoctrineDBALConnection->getDatabasePlatform()',
    44 => 'DoctrineDBALConnection->getDatabasePlatformVersion()',
    45 => 'DoctrineDBALException:',
    46 => 'FATAL:',
    47 => 'Failed',
    48 => 'OC::init()',
    49 => 'OCAppAppManager->getInstalledApps()',
    50 => 'OCAppAppManager->getInstalledAppsValues()',
    51 => 'OCAppConfig->getValues(false,',
    52 => 'OCAppConfig->loadConfig(NULL,',
    53 => 'OCAppConfig->searchValues(enabled,',
    54 => 'OCAppFrameworkBootstrapCoordinator->runInitialRegistration()',
    55 => 'OCDBConnection->connect()',
    56 => 'OCDBConnection->getDatabaseProvider()',
    57 => 'OCDBConnectionAdapter->getDatabaseProvider()',
    58 => 'OCDBQueryBuilderQueryBuilder->expr()',
    59 => 'OC_App::getEnabledApps()',
    60 => 'SQLSTATE[08006]',
    61 => 'Stack',
    62 => '[7]',
    63 => 'at',
    64 => 'been',
    65 => 'connect',
    66 => 'connection',
    67 => 'database',
    68 => 'database:',
    69 => 'driver:',
    70 => 'enabled)',
    71 => 'exception',
    72 => 'failed:',
    73 => 'false)',
    74 => 'false,',
    75 => 'has',
    76 => 'in',
    77 => 'is',
    78 => 'localhost',
    79 => 'nextcloud',
    80 => 'occurred',
    81 => 'port',
    82 => 'postgres',
    83 => 'require_once(/var/www/html/c...)',
    84 => 'require_once(/var/www/html/l...)',
    85 => 'server',
    86 => 'starting',
    87 => 'system',
    88 => 'the',
    89 => 'thrown:',
    90 => 'to',
    91 => 'trace:',
    92 => 'unhandled',
    93 => 'up',
    94 => '{main}',
```