nextcloud/nextcloudpi #1333: No NEXTCLOUD_URL configured for notify_push

theCalcaholic, 2021-09-15:
Did we deliberately not set the `NEXTCLOUD_URL` config variable to `https://localhost` (as recommended in the notify_push repo)?
If we are using the public IP, as we currently do, I fear that the notify_push service will break whenever the IP is not a trusted proxy, which, for the majority of our users will be the case after aday (due to the usage of dynamic IPs).

nachoparker, 2021-09-15:
not sure I understand why we need `NEXTCLOUD_URL`, I think it captures it from `config.php` automatically

theCalcaholic, 2021-09-16:
Yeah, but it looks like it registers the proxy with the public IP of the instance instead of communicating locally. It *shouldn't* use the nextcloud url from config.php, because that way, communication is routed externally, which means, that the trusted-proxy will need to contain the public IP of the instance and thus, be updated every day in the case of a dynamic IP.

theCalcaholic, 2021-09-16:
To quote from the Troubleshooting instructions in the notify_push Readme:

> If your nextcloud hostname resolves do a dynamic ip you can try setting the NEXTCLOUD_URL to the internal ip of the server.

theCalcaholic, 2021-09-16:
FYI, we had that discussion before: https://github.com/nextcloud/nextcloudpi/pull/1298#discussion_r684585071

svenb1234, 2021-09-25:
I can confirm that I have to add my new dynamic IP to the trusted proxies list every day in order to activate HPB.

selfhost.eu is the DDNS provider.

gadric, 2021-09-25:
Izabela the same experience: #1346

nachoparker, 2021-10-04:
@svenb1234 do you not have a domain name?

svenb1234, 2021-10-04:
Not sure I understand the question. The answer would be yes I guess, bit that is too obvious.

Do you want to know the domain? If so I can send it to you. Or whether it is entered in some variable?

svenb1234, 2021-10-07:
'overwrite.cli.url' => 'https://mysubdomain.selfhost.eu/', and the HPB is still down after the IP changed.

nachoparker, 2021-10-12:
> I can add this like, in which section of the file should I add this line?

@gadric under the line that reads `[Service]`

thanks

gadric, 2021-10-12:
OK, 2 questions: 

1. which editor can I used in docker container to edit the file? Nano and vim seems not to exists. 
2. Do I need to remove the settings mentioned in #1346 like 

```
192.168.178.38 xxxxxxx.goip.de in /etc/hosts
```

svenb1234, 2021-10-14:
With 1.40.6 the HPB was down this morning again. Adding the current external IP to trusted proxies made it work again.

closed 2021-10-14 by nachoparker

gadric, 2021-10-16:
@svenb1234: at least my case this fixed the problem. Two days in a row with new Dyn IPs and HPB is up

Oberlurch3000, 2022-08-20:
I'm glad I found this thread. It took me so long to find out how the HPB will work on my instance. Setting ```NEXTCLOUDURL``` to localhost worked.

I recognized I have to manually start the HPB after every reboot ```sudo systemctl enable --now notify_push```. Is there a easy way to fix this (cronjob for example)?

theCalcaholic, 2022-08-20:
@Oberlurch3000 Could you please create a separate issue for this? notify_push shouldn't be run via systemd in docker

Oberlurch3000, 2022-08-21:
Ok, I'll open a new thread. My instance in no docker installation.